Koenkk/zigbee2mqtt #22803: Log section on z2m interface stay empty

devildant, 2024-05-28:
### What happened?

in the past when I went to the logs of the z2m interface, I saw the messages from all my devices, but since the latest versions, nothing if I change the level, nothing appears anymore

### What did you expect to happen?

See all log like the past

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20210708

### Adapter

Sonoff p

### Setup

Haos

piwi3910, 2024-05-28:
please post your config.

devildant, 2024-05-28:
> please post your config.

hi,
```
homeassistant: true
advanced:
  network_key:
    - 81
    - 105
    - 201
    - 63
    - 133
    - 64
    - 51
    - 243
    - 158
    - 103
    - 233
    - 105
    - 129
    - 66
    - 31
    - 143
  pan_id: 32473
  ext_pan_id:
    - 117
    - 115
    - 14
    - 170
    - 84
    - 215
    - 182
    - 252
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  transmit_power: 20
  log_level: debug
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 514
    protocol: udp4
    type: '5424'
  last_seen: ISO_8601_local
mqtt:
  server: mqtt://homeassistant.local:1883
  user: **********
  password: *********
serial:
  port: /dev/ttyUSB0
frontend:
  port: 8099
device_options:
  legacy: false
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
```

devildant, 2024-05-28:
[image]

Koenkk, 2024-05-28:
This will be fixed in the next z2m release (`MQTT publish`) will be logged as `info` again.

closed 2024-05-28 by Koenkk