Koenkk/zigbee2mqtt #22808: Colorock  CR-MNZ1  Zigbee version with energy monitor (UNREPORTABLE_ATTRIBUTE, consumption data is null)

nifelin123, 2024-05-28:
### What happened?

the data of the energy monitor is not pulled up. if you go to the developer console in the haElectricalMeasurement cluster and select the correct attributes there, the data can be read. There is also an error attached below (Status 'UNREPORTABLE_ATTRIBUTE') in the vine. Tried the converter from tuya.ts repository it didn't help. What else do you recommend to do?

[image]
[image]
[image]

### Zigbee2MQTT version

1.37.1 

### Adapter firmware version

20230507

### Adapter

CC2652P

### Setup

Intel NUC

### Debug log

Failed to configure '0x70b3d52b600b9d7a', attempt 1 (Error: ZCL command 0x70b3d52b600b9d7a/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:315:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:770:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:556:9) at Object.rmsVoltage (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:224:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2943:13) at Configure.configure (/app/lib/extension/configure.ts:121:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

alexps70, 2024-07-11:
I have the same device - no data on voltage/current/power

andregoncalvespires, 2024-07-16:
The same here. Any developments?

Koenkk, 2024-07-17:
In the reporting tab, do you get an `UNREPORTABLE_ATTRIBUTE` error for all these attributes?

- Cluster: `haElectricalMeasurement`, attributes `rmsVoltage`, `rmsCurrent`, `activePower`
- Cluster: `seMetering`, attributes: `currentSummDelivered`

If it gives an error for all we need to poll the attributes:

Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/436c0291d93741b3ac261455b52722aa): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

nifelin123, 2024-07-17:
> In the reporting tab, do you get an `UNREPORTABLE_ATTRIBUTE` error for all these attributes?
>
> * Cluster: `haElectricalMeasurement`, attributes `rmsVoltage`, `rmsCurrent`, `activePower`
> * Cluster: [...]

I did everything according to your instructions. The situation has not changed. All indicators are zero. only state switching is available. With a hub from Tuya, this relay works and transmits all indicators. I am attaching the error from the log again.

Error 2024-07-17 22:34:25z2m: Failed to configure 'Boiler', attempt 3 (Error: ZCL command 0x70b3d52b600b9d7a/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:316:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:771:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:557:9) at Object.rmsVoltage (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:221:5) at Object.configure (/app/data/extension/externally-loaded.js:34:9) at Configure.configure (/app/lib/extension/configure.ts:126:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

nifelin123, 2024-07-17:
> I did everything according to your instructions. The situation has not changed. All indicators are zero. only state switching is available. With a hub from Tuya, this relay works and transmits all [...]
[image]
[image]

[image]

nifelin123, 2024-07-17:
> In the reporting tab, do you get an `UNREPORTABLE_ATTRIBUTE` error for all these attributes?
>
> * Cluster: `haElectricalMeasurement`, attributes `rmsVoltage`, `rmsCurrent`, `activePower`
> * Cluster: [...]

The situation has changed, now only energy data is NULL
[image]

lpt2007, 2024-07-18:
Hi,

@nifelin123 solution works, but readings not correct?

I connected only input to my Colorock CR-MNZ1 Zigbee version and device shows me those readings?
[image]

Any idea?

andregoncalvespires, 2024-07-18:
We are close to a solution. The current reported by the device is in mA. Confirming the consumed energy reported in seMetering.currentSummDelivered

Redram49RUS, 2024-07-20:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/436c0291d93741b3ac261455b52722aa):

save this as file next to configuration.yaml as ext_converter.js
add it to configuration.yaml:
external_converters:
  - ext_converter.js
start z2m, check if issue is fixed

Have this error:
Integration error: external_converters - Integration 'external_converters' not found.

Koenkk, 2024-07-20:
@Redram49RUS you have to add it to your z2m configuration.yaml, not the home assistant ons

lpt2007, 2024-07-20:
Hi,

I try external converter. Now I get some values on W / A / V.
Current not show correctly, probably number is in mA not A as shown. 
Energy still Null? 

here i screenshot:
[image]

Redram49RUS, 2024-07-20:
> @Redram49RUS you have to add it to your z2m configuration.yaml, not the home assistant one

Thanks, it worked. The energy remained "Null".

Koenkk, 2024-07-21:
I've updated the [external converter](https://gist.github.com/Koenkk/436c0291d93741b3ac261455b52722aa), could you check it again after re-configuring the device by pressing the yellow refresh icon on the device page?

nifelin123, 2024-07-21:
> I've updated the [external converter](https://gist.github.com/Koenkk/436c0291d93741b3ac261455b52722aa), could you check it again after re-configuring the device by pressing the yellow refresh icon on the device page?

ok 10 min.

nifelin123, 2024-07-21:
> I've updated the [external converter](https://gist.github.com/Koenkk/436c0291d93741b3ac261455b52722aa), could you check it again after re-configuring the device by pressing the yellow refresh icon on the device page?

deleted the old converter, threw the new one and rebooted the system. Now all data has Null status. An error occurs when the yellow button is pressed.
ERROR

2024-07-21 13:07:08z2m: Failed to configure 'Boiler', attempt 5 (Error: ZCL command 0x70b3d52b600b9d7a/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:316:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:771:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:557:9) at Object.rmsVoltage (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:221:5) at Object.configure (/app/data/extension/externally-loaded.js:34:9) at Configure.configure (/app/lib/extension/configure.ts:126:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:56:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Redram49RUS, 2024-07-21:
> I've updated the [external converter](https://gist.github.com/Koenkk/436c0291d93741b3ac261455b52722aa), could you check it again after re-configuring the device by pressing the yellow refresh icon on the device page?

 2024-07-21 21:21:31Failed to configure 'Варочная панель', attempt 7 (Error: ZCL command 0x70b3d52b600b9be7/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:316:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:771:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:557:9) at Object.rmsVoltage (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:224:5) at Object.configure (/app/data/extension/externally-loaded.js:34:9) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Koenkk, 2024-07-21:
Are you sure you updated the external converter? Line 34 should be commented (which currently causes this `rmsVoltage` error)

nifelin123, 2024-07-21:
> Are you sure you updated the external converter? Line 34 should be commented (which currently causes this `rmsVoltage` error)

[image]

andregoncalvespires, 2024-07-21:
.js updated, z2m restarted.

[image]

[image]

andregoncalvespires, 2024-07-22:
The values ​​have been corrected, although the UNREPORTABLE_ATTRIBUTE error continues to appear. I have doubts whether the energy value is being reported correctly by the device

Error 2024-07-22 16:14:50z2m: Failed to configure 'Luz Mesa', attempt 1 (Error: ZCL command 0xa4c138e5bdf24678/1 genLevelCtrl.configReport([{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":1,"attribute":"currentLevel"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:316:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:771:26) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:557:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:107:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1006:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:190:21) at Configure.configure (/app/lib/extension/configure.ts:126:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

[image]

Redram49RUS, 2024-07-22:
I also updated ext.js, now current and power readings are not transmitted when a 2 kW load is on.

andregoncalvespires, 2024-07-23:
The energy measurement is really unreliable. The Tuya app already showed a large variation. Does anyone know if using calibration could help? Power and Current appear to be normal. Considering the variation between now and 14 hours ago, it shows a consumption of more than 200 kWh and it should be something around 10 kWh

[image]

Koenkk, 2024-07-24:
@andregoncalvespires if the energy measurement via the Tuya gateway is also incorrect, it's not something that can be fixed by z2m

andregoncalvespires, 2024-07-24:
It is clear @Koenkk . The question was considering the available calibration options. But I'm already discouraged by this device.
Thank you for the fabulous work you and everyone who contributes do.

Brucepbd, 2024-09-24:
The external converter seems to be working for me.
[image]

lucasnz, 2024-10-20:
This fixed the missing data for me :)

Brucepbd, 2024-10-21:
Although it is working.  I did get a random change in the energy level last week that through off my energy monitoring.  So I am relying on integrating the power to get more stable energy data.

closed 2024-12-29 by github-actions[bot]

Brucepbd, 2025-01-14:
> I have two Colorock CR-MNZ1 with power monitoring and have added the extension from this thread but it doesn't work and they only update values if I manually poll via the Dev console. I don't know if it makes any difference but I am running Zigbee2MQTT version 2.0.0 so maybe the extension isn't compatiable.
> 
> Other than the updating not working.
> 
> Power is in Watts which is fine Current says Amp but is mA Voltage is Volts which is fine Energy is hard to figure out with out reliable data on the other values but it isn't kWh, the best I can figure is it needs to be divided by 100 which may get it about what I think it should be.

Same here.  The last time my history shows a power and current measurement was January 3rd 2025.  They have both been zero since then.  I have not been using the extension though.  I thought it was working without the extension, so I removed it.  I'll setup the extension and report back to see if it changes.

Fograd, 2025-01-15:
@Koenkk with your external converter I managed to get `activePower` to poll (was able to read through Dev console)

But I cannot get it to automatically expose it through `Power` and the last value `Power` shows is the latest value that I polled from `activePower` manually. Am I doing something wrong?

Thanks in advance for your response. I have been banging my head against the wall for the past 4 hours trying to get it fixed...

Brucepbd, 2025-01-16:
I'm not sure how github works.  Since this issue is closed I'm not sure if anyone will pay attention to our posts.

Fograd, 2025-01-16:
> I'm not sure how github works. Since this issue is closed I'm not sure if anyone will pay attention to our posts.

@Brucepbd I managed to get help on Z2M discord channel that resolved my issue. Here is the response I got (shoutout to Rene):

If you are on z2m version 2.0, the location for external converters has changed
it now needs to be in `/homeassistant/zigbee2mqtt/external_converters` (essentially you need to create a sub-folder under `zigbee2mqtt` and put your `ext_converter.js` file above in there)

after you have replaced it:
1. Remove the following from your `/homeassistant/zigbee2mqtt/configuration.yaml` file as in the guide above:
```
external_converters:
  - ext_converter.js
``` 
2. restart z2m
3. open the most recent logdir in `/homeassistant/zigbee2mqtt/log`
4. open `log.log`
5. check if the converter is loaded or if loading gives an error. If no errors then force delete your device and re-pair it. (mine worked even without having to re-pair the device)

Let me know if you have any issues. Would be happy to help if I can!

Brucepbd, 2025-01-16:
> @Brucepbd I managed to get help on Z2M discord channel that resolved my issue. Here is the response I got (shoutout to Rene):
>
> If you are on z2m version 2.0, the location for external converters has [...]

Thank you!  It works now.  I remember seeing something about that in the release notes before I made the upgrade - but it looked like it was working after the upgrade was done (it was off and had zero for watts and current - which was correct).  I thought the converter was incorporated into the main z2m and not needed anymore.  So I forgot about the directory change and never followed through.  It wasn't until someone posted here that I took a closer look at the entity history and realized it was broken.

It will be great having all this information about this device in one place for others to get things working.  Thanks again for posting! :)

nblount, 2025-01-16:
This appears to have worked for one and not the other will have to try re-pairing tomorrow.

Brucepbd, 2025-01-18:
One thing that seems to be a continuing problem with this device is that the energy value makes large jumps during measurements.  As seen in the picture, for the same load it jumps from 53kWh to 107kWh.  It seems to happen during times when the load switches from off to on.  I don't know if this is a device problem or an external converter problem.  I had the same problem with a previous CR-MNZ1 and returned it.  I bought another one to replace it and it still has this problem.  So I can't use it for energy measurement.

[image]

cagarrinho, 2025-10-19:
I bought a Colorock 30A Zigbee Relay and added it to Zigbee2MQTT in HomeAssistant. As you can see in the images, I can't get data for: Power, Current, and Voltage. I can only receive Energy data. Does anyone have the same device and can help me? I've already tried an external converter I found online, but that didn't work either.
[image][image]