Koenkk/zigbee2mqtt #22809: UNSUPPORTABLE_ATTRIBUTE error for MCLH-07

almirus, 2024-05-28:
### What happened?

I periodically get an error for 2 devices MCLH-07

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20240315

### Adapter

egony cc2652p

### Setup

Add-on on Home Assistant OS x86

### Debug log

```
genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:315:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:770:26)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:556:9)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:76:13)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:112:17)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:21)
    at Configure.configure (/app/lib/extension/configure.ts:121:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
[2024-05-27 22:04:09] error: 	z2m: Failed to configure 'Датчик протечки под ванной', attempt 3 (Error: ZCL command 0x00158d0000b60d82/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:315:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:770:26)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:556:9)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:76:13)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:112:17)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:21)
    at Configure.configure (/app/lib/extension/configure.ts:121:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
[2024-05-27 22:08:45] error: 	z2m: Failed to configure 'Датчик протечки под мойкой на кухне', attempt 3 (Error: ZCL command 0x00158d0000b60e53/1 
```

Koenkk, 2024-07-02:
Is the battery % reported?

Koenkk, 2024-07-03:
Pushed a fix, can you check if it works?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Koenkk, 2024-07-04:
Perfect!

closed 2024-07-04 by Koenkk