Koenkk/zigbee2mqtt #22813: Smart Sprinkler Timer water metering

derekangl, 2024-05-28:
### Link

https://www.ebay.co.uk/itm/355700417511

### Database entry

{"id":21,"type":"EndDevice","ieeeAddr":"0xa4c13869455a3642","nwkAddr":26812,"manufId":4417,"manufName":"_TZE204_uab532m0","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"b��-fc��-\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1716919405582}

### Comments

hi, can you  hellp me sport new device.Please.
[image]
[image]

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_uab532m0',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},

### External definition

```shell
q
```

derekangl, 2024-05-31:
logs
Info 2024-05-31 20:31:21Succesfully interviewed '0xa4c13869455a3642'
Info 2024-05-31 20:31:21Successfully interviewed '0xa4c13869455a3642', device has successfully been paired
Info 2024-05-31 20:31:21Device '0xa4c13869455a3642' is supported, identified as: tuya water_valve here (TS0601_new)
Info 2024-05-31 20:31:22MQTT publish: topic 'z2mR22/bridge/event', payload '{"data":{"definition":{"description":"water_valve here","exposes":[{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601_new","options":[],"supports_ota":false,"vendor":"tuya"},"friendly_name":"0xa4c13869455a3642","ieee_address":"0xa4c13869455a3642","status":"successful","supported":true},"type":"device_interview"}'
Info 2024-05-31 20:31:22Configuring '0xa4c13869455a3642'
Info 2024-05-31 20:31:22MQTT publish: topic 'homeassistant/sensor/0xa4c13869455a3642/linkquality/config', payload '{"availability":[{"topic":"z2mR22/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"z2mR22/0xa4c13869455a3642/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c13869455a3642"],"manufacturer":"tuya","model":"water_valve here (TS0601_new)","name":"0xa4c13869455a3642","via_device":"zigbee2mqtt_bridge_0x04cd15fffee45691"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"z2mR22/0xa4c13869455a3642","name":"Linkquality","object_id":"0xa4c13869455a3642_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.37.1-dev","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"z2mR22/0xa4c13869455a3642","unique_id":"0xa4c13869455a3642_linkquality_z2mR22","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2024-05-31 20:31:22Successfully configured '0xa4c13869455a3642'

and

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_uab532m0',
        },
    ],
    model: 'TS0601_new',
    vendor: 'tuya',
    description: 'water_valve here',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
        ],
    },
    extend: [
        
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;

dominique6017, 2024-06-17:
I have the same valve and have found the tuya end point :
{"1":"Switch", boolean
"3":"Status",   "type": "Enum", "values": "{\"range\":[\"off\",\"auto\",\"disabled\"]}"
"5":"Countdown", "type": "Integer",    "values": "{\"unit\":\"min\",\"min\":1,\"max\":240,\"scale\":0,\"step\":1}"
"6":"Countdown Left",  "type": "Integer", "values": "{\"unit\":\"min\",\"min\":1,\"max\":240,\"scale\":0,\"step\":1}"
"9":"Water Current"          "type": "Integer", "values": "{\"unit\":\"gal/min\",\"min\":0,\"max\":1000000000,\"scale\":3,\"step\":1}"
"11":"Battery Percentage"    "type": "Integer", "values": "{\"unit\":\"%\",\"min\":0,\"max\":100,\"scale\":0,\"step\":1}"
"15":"Water Total"         "type": "Integer", "values": "{\"unit\":\"gal\",\"min\":0,\"max\":999999999,\"scale\":3,\"step\":1}"
"19":"Fault"
"37":"Weather Delay", type enum {\"range\":[\"24h\",\"48h\",\"72h\",\"cancel\"]}" 
"38":"Normal Timer",         "type": "String",    "values": "{\"maxlen\":255}"
"42":"Switch", boolean
"47":"Smart Irrigation"
"101":"Total flow reset switch",
"102":"Quantitative watering",
"103":"Flow switch"}
"104":"Child lock"
"105":"Surplus flow"
"106":"Single watering time"
"107":"Interface refresh",
"108":"Amount of watering per time"}

I don't know to write the converter and need help

xpavli44, 2024-06-21:
@dominique6017 are you familiar with https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-2-adding-converter-s-for-your-device ?

xpavli44, 2024-06-21:
this device https://www.zigbee2mqtt.io/devices/QT06_2.html seems to be pretty close? 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/giex.ts
I have no idea what I am doing, I just try to replicate the procedure depicted in this video https://www.youtube.com/watch?v=VJsXc54ZU3o

xpavli44, 2024-06-21:
ohhh .... it seems it may already exist? https://github.com/Koenkk/zigbee-herdsman-converters/blob/da65b1aeffd96527df02725b49de61e453fee059/src/devices/neo.ts#L130
just needs some tweaking?

xpavli44, 2024-06-21:
@dominique6017 and/or @derekangl  could you test if https://github.com/Koenkk/zigbee-herdsman-converters/pull/7670 helps with the issue? I will be able to do it in at least a week, I cannot get to my sensor earlier :(

mrespin, 2024-06-22:
Hiya,

I have this valve from [derekangl] and came up with this converter, based on the Koenkk/zigbee-herdsman-converters#7670 
NB zigbee-herdsman-converters update has not yet been mergeD. (latest v19.58.0).
When playing with On / Off (working with 3 seconds delay), I see errors:
Error 2024-06-22 02:52:20z2m: Exception while calling fromZigbee converter: Value '4' is not allowed, expected one of true,false}
Error 2024-06-22 02:52:26z2m: Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of true,false}
Error 2024-06-22 02:52:27z2m: No converter available for 'get' 'state' ()
Error 2024-06-22 02:52:32z2m: Exception while calling fromZigbee converter: Value '3' is not allowed, expected one of true,false}
Error 2024-06-22 02:53:02z2m: Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of true,false}
Error 2024-06-22 02:56:54z2m: No converter available for 'switch_enabled' ("ON")
Error 2024-06-22 02:56:55z2m: No converter available for 'switch_enabled' ("OFF")
Error 2024-06-22 02:57:00z2m: No converter available for 'total_flow_reset_switch' ("OFF")
Error 2024-06-22 02:57:01z2m: No converter available for 'total_flow_reset_switch' ("ON")
Error 2024-06-22 02:57:02z2m: No converter available for 'flow_switch' ("OFF")
Error 2024-06-22 02:57:04z2m: No converter available for 'flow_switch' ("ON")
Error 2024-06-22 02:56:45z2m: No converter available for 'get' 'state' ()
Error 2024-06-22 02:56:54z2m: No converter available for 'switch_enabled' ("ON")
Error 2024-06-22 02:56:55z2m: No converter available for 'switch_enabled' ("OFF")
Error 2024-06-22 02:57:00z2m: No converter available for 'total_flow_reset_switch' ("OFF")
Error 2024-06-22 02:57:01z2m: No converter available for 'total_flow_reset_switch' ("ON")
Error 2024-06-22 02:57:02z2m: No converter available for 'flow_switch' ("OFF")
Error 2024-06-22 02:57:04z2m: No converter available for 'flow_switch' ("ON")

Moreover, NULLS:
Water current 
Current water flow (L/min)
NullL/min
Battery percentage
Null%
Water total
Total water flow (L)
NullL
Fault
Fault status
Null
Smart irrigation 
Null
Surplus flow 
Null
Single watering duration 
Null
Single watering amount 
Null

External converter:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy')

const definition = 
{
    fingerprint: 
    tuya.fingerprint('TS0601', ['_TZE204_uab532m0']),
    zigbeeModel: ['NAS-WV03B'],
    model: 'NAS-WV03B_Jac',
    vendor: 'Neo',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    description: 'Smart sprinkler timer',
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
[... 24 lines cut ...]
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [3, 'status', tuya.valueConverter.onOff],
            [5, 'countdown', tuya.valueConverter.raw],
            [6, 'countdown_left', tuya.valueConverter.raw],
            [9, 'water_current', tuya.valueConverter.raw],
            [11, 'battery_percentage', tuya.valueConverter.batteryState],
            [15, 'water_total', tuya.valueConverter.raw],
            [19, 'fault', tuya.valueConverter.raw],
            [37, 'weather_delay', tuya.valueConverter.raw],
            [38, 'normal_timer', tuya.valueConverter.raw],
            [42, 'switch_enabled', tuya.valueConverter.onOff],
            [47, 'smart_irrigation', tuya.valueConverter.raw],
            [101, 'total_flow_reset_switch', tuya.valueConverter.onOff],
            [102, 'quantitative_watering', tuya.valueConverter.raw],
            [103, 'flow_switch', tuya.valueConverter.onOff],
            [104, 'child_lock', tuya.valueConverter.onOff],
            [105, 'surplus_flow', tuya.valueConverter.raw],
            [106, 'single_watering_duration', tuya.valueConverter.raw],
            [108, 'single_watering_amount', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

OgV1, 2024-06-24:
> ohhh .... it seems it may already exist? https://github.com/Koenkk/zigbee-herdsman-converters/blob/da65b1aeffd96527df02725b49de61e453fee059/src/devices/neo.ts#L130 just needs some tweaking?

Hi Guys,

I have created that converter based on this fingerprint **_TZE204_rzrrjkz2** in this [pull request](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7630)

And actually not sure if I did it right, because I still don't have a chance to test with my own device (Guess it because the fingerprint is now **_TZE204_uab532m0** and not the same as mine **_TZE204_rzrrjkz2**

Anyone have an idea, how to solve this? and why do we have different fingerprints?

> Hiya,
> 
> I have this valve from derekangl and came up with this converter, based on the [Koenkk/zigbee-herdsman-converters#7670](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7670) NB zigbee-herdsman-converters update has not yet been merged. (latest v19.58.0). When playing with On / Off (working with 3 seconds delay), I see errors: Error 2024-06-22 02:52:20z2m: Exception while calling fromZigbee converter: Value '4' is not allowed, expected one of true,false} Error 2024-06-22 02:52:26z2m: Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of true,false} Error 2024-06-22 02:52:27z2m: No converter available for 'get' 'state' () Error 2024-06-22 02:52:32z2m: Exception while calling fromZigbee converter: Value '3' is not allowed, expected one of true,false} Error 2024-06-22 02:53:02z2m: Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of true,false} Error 2024-06-22 02:56:54z2m: No converter available for 'switch_enabled' ("ON") Error 2024-06-22 02:56:55z2m: No converter available for 'switch_enabled' ("OFF") Error 2024-06-22 02:57:00z2m: No converter available for 'total_flow_reset_switch' ("OFF") Error 2024-06-22 02:57:01z2m: No converter available for 'total_flow_reset_switch' ("ON") Error 2024-06-22 02:57:02z2m: No converter available for 'flow_switch' ("OFF") Error 2024-06-22 02:57:04z2m: No converter available for 'flow_switch' ("ON") Error 2024-06-22 02:56:45z2m: No converter available for 'get' 'state' () Error 2024-06-22 02:56:54z2m: No converter available for 'switch_enabled' ("ON") Error 2024-06-22 02:56:55z2m: No converter available for 'switch_enabled' ("OFF") Error 2024-06-22 02:57:00z2m: No converter available for 'total_flow_reset_switch' ("OFF") Error 2024-06-22 02:57:01z2m: No converter available for 'total_flow_reset_switch' ("ON") Error 2024-06-22 02:57:02z2m: No converter available for 'flow_switch' ("OFF") Error 2024-06-22 02:57:04z2m: No converter available for 'flow_switch' ("ON") Error 2024-06-22 03:32:49z2m: No converter available for 'weather_delay' ("24h")
> 
> Moreover, NULLS: Status Null Water current Current water flow (L/min) NullL/min Battery percentage Null% Water total Total water flow (L) NullL Fault Fault status Null Smart irrigation Null Surplus flow Null Single watering duration Null Single watering amount Null
> 
> External converter:
> 
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee'); const tz = require('zigbee-herdsman-converters/converters/toZigbee'); const exposes = require('zigbee-herdsman-converters/lib/exposes'); const reporting = require('zigbee-herdsman-converters/lib/reporting'); const {} = require('zigbee-herdsman-converters/lib/modernExtend'); const e = exposes.presets; const ea = exposes.access; const tuya = require('zigbee-herdsman-converters/lib/tuya'); const legacy = require('zigbee-herdsman-converters/lib/legacy')
> 
> const definition = { fingerprint: tuya.fingerprint('TS0601', ['_TZE204_uab532m0']), zigbeeModel: ['NAS-WV03B'], model: 'NAS-WV03B_Jac', vendor: 'Neo', fromZigbee: [tuya.fz.datapoints], toZigbee: [tuya.tz.datapoints], description: 'Smart sprinkler timer', onEvent: tuya.onEventSetTime, configure: tuya.configureMagicPacket, exposes: [ e.switch(), e.enum('status', ea.STATE, ['off', 'auto', 'disabled']).withDescription('Status'), e.numeric('countdown', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown'), e.numeric('countdown_left', ea.STATE).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown left'), e.numeric('water_current', ea.STATE).withUnit('L/min').withValueMin(0).withValueMax(3785.41).withValueStep(0.001) .withDescription('Current water flow (L/min)'), e.numeric('battery_percentage', ea.STATE).withUnit('%').withValueMin(0).withValueMax(100).withDescription('Battery percentage'), e.numeric('water_total', ea.STATE).withUnit('L').withValueMin(0).withValueMax(378541.0).withValueStep(0.001) .withDescription('Total water flow (L)'), e.binary('fault', ea.STATE, 'DETECTED', 'NOT_DETECTED').withDescription('Fault status'), e.enum('weather_delay', ea.STATE_SET, ['24h', '48h', '72h', 'cancel']).withDescription('Weather delay'), e.text('normal_timer', ea.STATE_SET).withDescription('Normal timer'), e.binary('switch_enabled', ea.STATE_SET, 'ON', 'OFF').withDescription('Switch enabled'), e.numeric('smart_irrigation', ea.STATE).withDescription('Smart irrigation'), e.binary('total_flow_reset_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Total flow reset switch'), e.numeric('quantitative_watering', ea.STATE_SET).withUnit('L').withValueMin(0).withValueMax(10000) .withDescription('Quantitative watering'), e.binary('flow_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Flow switch'), e.binary('child_lock', ea.ST [... cut ...] ya.valueConverter.onOff], [47, 'smart_irrigation', tuya.valueConverter.raw], [101, 'total_flow_reset_switch', tuya.valueConverter.onOff], [102, 'quantitative_watering', tuya.valueConverter.raw], [103, 'flow_switch', tuya.valueConverter.onOff], [104, 'child_lock', tuya.valueConverter.onOff], [105, 'surplus_flow', tuya.valueConverter.raw], [106, 'single_watering_duration', tuya.valueConverter.raw], [108, 'single_watering_amount', tuya.valueConverter.raw], ], }, };
> 
> module.exports = definition;

xpavli44, 2024-06-24:
> And actually not sure if I did it right, because I still don't have a chance to test with my own device (Guess it because the fingerprint is now _TZE204_uab532m0 and not the same as mine _TZE204_rzrrjkz2?)

you should be able to add your fingerprint as well just change [this](https://github.com/Koenkk/zigbee-herdsman-converters/blob/41b26f2025b0627f984ccf0da65ca4051bd404f0/src/devices/neo.ts#L125) line to something like

`fingerprint: tuya.fingerprint('TS0601', ['_TZE204_uab532m0', ' _TZE204_rzrrjkz2']),`

having multiple fingerprints is quite common see [example](https://github.com/Koenkk/zigbee-herdsman-converters/blob/41b26f2025b0627f984ccf0da65ca4051bd404f0/src/devices/neo.ts#L39) a few lines above

if your device did not get recognized it was probably due to the lack of the fingerprint, as TuYa makes a mess in the identifications by reusing them incorrectly.

OgV1, 2024-06-24:
@xpavli44 Thanks man,

And about further investigation for [these errors](https://github.com/Koenkk/zigbee2mqtt/issues/22813#issuecomment-2183705606) do you have any idea where to start?

xpavli44, 2024-06-24:
@OgV1 unfortunately no clue :( 

best I can see is 

> Now set the Zigbee2MQTT log_level to debug and enable the external definition by adding the following to your Zigbee2MQTT configuration.yaml.

as per https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition

maybe it will provide a little more insight into the errors?

oliverwoodings, 2024-06-24:
I got one of these off AliExpress, currently the stuff in `neo.ts` isn't working particularly well, but I've got a few things working:
- I can turn it on and off remotely
- It reports current water flow

But nothing else is working properly. Some things I've noticed when debugging:
- The water flow is definitely not litres per minute - it's currently saying mine is pumping out 1483 l/min, someone call the firebrigade 😆 Perhaps it's litres per hour, but that still seems high...
- The 'status' (data point 3) is definitely not just a simple `onOff` - so far I've observed 4 different values. For now I have this: `[3, 'status', tuya.valueConverterBasic.lookup({'off': 0, 'one': 1, 'two': 2, 'three': 3, 'four': 4})]`. I've noticed that when it's off, the value is 0, when you turn it on remotely, the value is 3, and when you turn it on using the device on the button, the value is 4.
- The 'total water flow' (data point 15) is mad - at one point it was reporting massive negative numbers, and is now sitting at 15055228
- The battery status _never_ reports - it's always null
- Everything else is always null:

[image]

Looking at the logs, there is also definitely something wrong with `commandMcuSyncTime`. Even with the recommended `onEvent: tuya.onEventSetTime` , this comes up:
```
[2024-06-24 17:56:35] debug: 	z2m: Received Zigbee message from 'zigbee-garden-water-valve', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":116}' from endpoint 1 with groupID 0
[2024-06-24 17:56:35] debug: 	zh:controller:endpoint: ZCL command 0xa4c1386d144eb8bf/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[102,121,165,67,102,121,179,83]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-06-24 17:56:35] debug: 	z2m: No converter available for 'NAS-WV03B' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":116}'
```
Looking at the code in github, it looks like this might silently fail? https://github.com/Koenkk/zigbee-herdsman-converters/blob/0da0cbb88593b71f94385caeed4714d3dd0d1eb2/src/lib/tuya.ts#L192-L194

Any ideas @xpavli44 ?

OgV1, 2024-06-25:
Hi @oliverwoodings 

This might be a very dumb question, I can see the latest release [here](https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v19.59.2), but I'm not sure how I can get it to work with my device on my system? Seems like you guys can access it and test it already, I tried to research for this but cannot find anything, could you please suggest me how to get this work?

oliverwoodings, 2024-06-25:
@OgV1 by default, the version of herdsman used by z2m only updates when z2m gets a new release. I believe you can switch z2m  you can either switch zg2mqtt to use the 'dev' branch, which will make it keep herdsman up to date to the latest release: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

Personally, I've been testing using a custom external definition (using [these instructions](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_2-adding-your-device), using an adapted version of the code from `neo.ts` in herdsman.

OgV1, 2024-06-25:
Thanks @oliverwoodings, here's what I found according to your provided information, 

> * The water flow is definitely not litres per minute - it's currently saying mine is pumping out 1483 l/min, someone call the firebrigade 😆 Perhaps it's litres per hour, but that still seems high...
> * The 'total water flow' (data point 15) is mad - at one point it was reporting massive negative numbers, and is now sitting at 15055228

The water flow unit is normally Gallon, the unit conversion to liters might be messed up.

The current exposes and DP is below
```
        exposes: [
            e.numeric('water_current', ea.STATE).withUnit('L/min').withValueMin(0).withValueMax(3785.41).withValueStep(0.001)
                .withDescription('Current water flow (L/min)'),

        ],

        meta: {
            tuyaDatapoints: [
                [9, 'water_current', tuya.valueConverter.raw],
            ],
        },

```
> * The 'status' (data point 3) is definitely not just a simple `onOff` - so far I've observed 4 different values. For now I have this: `[3, 'status', tuya.valueConverterBasic.lookup({'off': 0, 'one': [...]

The sate should be these:
0 = Off
1 = _Currently unknown_
2 = Child lock activated
3 = Remotely On
4 = Physical button On

And yes the Child lock button is working

> * The battery status _never_ reports - it's always null
I tried to change the battery data point to something else, like 

The current exposes and DP is below
```
        exposes: [
            e.battery(),
        ],

        meta: {
            tuyaDatapoints: [
                [11, 'battery', tuya.valueConverter.raw],
            ],
        },

```

And I found another weird behavior. Looks like after we press the button, (doesn't matter remotely or physical) the device will switch off by itself after around 30 seconds, the reason is might because that there is no water through the valve, I have to test.

OgV1, 2024-06-25:
Okay, looks like I can get the Battery, Status and Current water flow (L/min) to work

but somehow these ones keep showing 0 
[image]

I can see from your image above that you somehow can make the "single_watering_amout" to show the value, would you mind sharing how? @oliverwoodings 
[image]

did I missed some DPs?

OgV1, 2024-06-28:
Hi @oliverwoodings, sorry for late reply,

Sure, here's the battery and L/min.

```
            e.numeric('water_current', ea.STATE).withUnit('L/min').withValueMin(0).withValueMax(3785.41).withValueStep(0.1)
                .withDescription('Current water flow (L/min)'), 
            e.battery(),
...
                [9, 'water_current', {
                    from: (v) => (v * GALLON_TO_LITER / 1000).toFixed(1),
                    to: (v) => Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000)
                }],
                [11, 'battery', tuya.valueConverter.raw],
...
```

eerison, 2024-07-01:
> @OgV1 by default, the version of herdsman used by z2m only updates when z2m gets a new release. I believe you can switch z2m you can either switch zg2mqtt to use the 'dev' branch, which will make it [...]

Hello guys,

Do you think this device will be available on the next release?

OgV1, 2024-07-03:
> Hello guys,
>
> Do you think this device will be available on the next release?

Hi @eerison,

Unfortunately I don't think we will, as I found the difficulties in findings the working exposes that can work with all of correct data-points, I will be happy if I could get some help from someone who are more expert than me.

eerison, 2024-07-03:
> Hi @eerison,
>
> Unfortunately I don't think we will, as I found the difficulties in findings the working exposes that can work with all of correct data-points, I will be happy if I could get some help [...]

Well I don't have any idea hehehe, But I just ordered this device and when it arrives I will try to help 🤞🏼

eerison, 2024-07-03:
hey guys

maybe you could open a PR with the current code state, then it could help others to test or help somehow.

xpavli44, 2024-07-03:
@eerison that's been done ... you can use https://github.com/Koenkk/zigbee-herdsman-converters/pull/7630/files to create external converter as per https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-2-adding-converter-s-for-your-device

eerison, 2024-07-03:
> @eerison that's been done ... you can use https://github.com/Koenkk/zigbee-herdsman-converters/pull/7630/files to create external converter as per https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-2-adding-converter-s-for-your-device

ahhh I just found it: https://www.zigbee2mqtt.io/devices/NAS-WV03B.html

the image looks a bit different of the original post 😄 

well if the integration is done, I guess this issue could be closed, couldn't it?

xpavli44, 2024-07-03:
@eerison I would not close it yet.

1. It is still quirky, throwing errors on button press etc. (read in above posts)
2. It also does not load the image in z2m so it needs some love as well

eerison, 2024-07-03:
> @eerison I would not close it yet.
> 
> 1. It is still quirky, it works but it is throwing errors on button press etc. (read in above posts)
> 2. It also does not load the image in z2m so it needs some love as well

thank you for clarify.

OgV1, 2024-07-03:
I just don't understand why my Water total and Single watering amount keeps showing 0 since the Water current is working, any ideas guys?

[image]
[image]

eerison, 2024-07-20:
Hello Guys.

my Sprinkler finally arrived :),

I am doing some tests and what works is just turn on and off using

```yaml
{"state": "ON"}
```

but no water flow then I tried to activate this `flow_switch`

using

```yaml
{"flow_switch": "ON"}
```

But I am getting this error
```
z2m: No converter available for 'flow_switch' ("ON")
```

I don't know if you faced the same issue ...

I didn't check yet how to debug it, I guess I can get this info here: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-2-adding-converter-s-for-your-device (as @xpavli44 mentioned), But in case you have some tips how I could debug it "easily", I would appreciate any help :)

eerison, 2024-07-20:
> @OgV1 by default, the version of herdsman used by z2m only updates when z2m gets a new release. I believe you can switch z2m you can either switch zg2mqtt to use the 'dev' branch, which will make it [...]

Hey @oliverwoodings 

could you give me some idea how to tests using [neo.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/da65b1aeffd96527df02725b49de61e453fee059/src/devices/neo.ts#L130)

when I click on `generate_external_definition` just appear this piece of code
[image]

But rI can't click on read/write button, am I missing something?

OgV1, 2024-07-22:
> > @OgV1 by default, the version of herdsman used by z2m only updates when z2m gets a new release. I believe you can switch z2m you can either switch zg2mqtt to use the 'dev' branch, which will make it keep herdsman up to date to the latest release: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html
> > Personally, I've been testing using a custom external definition (using [these instructions](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_2-adding-your-device), using an adapted version of the code from `neo.ts` in herdsman.
> 
> Hey @oliverwoodings
> 
> could you give me some idea how to tests using [neo.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/da65b1aeffd96527df02725b49de61e453fee059/src/devices/neo.ts#L130)
> 
> when I click on `generate_external_definition` just appear this piece of code ![Screenshot from 2024-07-20 18-22-40](https://private-user-images.githubusercontent.com/6358755/350706283-dea40dca-5a4a-426e-8093-0d764f67d466.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE2MjU0NDAsIm5iZiI6MTcyMTYyNTE0MCwicGF0aCI6Ii82MzU4NzU1LzM1MDcwNjI4My1kZWE0MGRjYS01YTRhLTQyNmUtODA5My0wZDc2NGY2N2Q0NjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjJUMDUxMjIwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yjc1MzQ5YTgzZDMzYjZmYjk3NTY2YmEwY2FiMGU4MGU3MjgyOWZiMTYyYzQzYzE1YTlkNDZjOTBmNWZhZGM3ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.IiZr1x9daLOMhBlrlVazoYi_75sRtXZA0ztep9esHIo)
> 
> But I can't click on read/write button, am I missing something?

@eerison My man, you definitely need to enable the debug for this

mine was first place the code in neo.ts and named to something like <name>.js (I'm using spk.js (for sprinkler, duh!) in **CONFIG/zigbee2mqtt**
[image]

then set your configuration.yaml as this instruction
[image]

mine was like this
[image]

after restart zb2mqtt the device will be ready to add (you may need to check if your device's fingerprint is matching these?
`fingerprint: tuya.fingerprint('TS0601', ['_TZE204_rzrrjkz2', '_TZE204_uab532m0']),`

eerison, 2024-07-22:
> > > @OgV1 by default, the version of herdsman used by z2m only updates when z2m gets a new release. I believe you can switch z2m you can either switch zg2mqtt to use the 'dev' branch, which will make it keep herdsman up to date to the latest release: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html
> > > Personally, I've been testing using a custom external definition (using [these instructions](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_2-adding-your-device), using an adapted version of the code from `neo.ts` in herdsman.
> > 
> > 
> > Hey @oliverwoodings
> > could you give me some idea how to tests using [neo.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/da65b1aeffd96527df02725b49de61e453fee059/src/devices/neo.ts#L130)
> > when I click on `generate_external_definition` just appear this piece of code ![Screenshot from 2024-07-20 18-22-40](https://private-user-images.githubusercontent.com/6358755/350706283-dea40dca-5a4a-426e-8093-0d764f67d466.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE2MjU0NDAsIm5iZiI6MTcyMTYyNTE0MCwicGF0aCI6Ii82MzU4NzU1LzM1MDcwNjI4My1kZWE0MGRjYS01YTRhLTQyNmUtODA5My0wZDc2NGY2N2Q0NjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjJUMDUxMjIwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yjc1MzQ5YTgzZDMzYjZmYjk3NTY2YmEwY2FiMGU4MGU3MjgyOWZiMTYyYzQzYzE1YTlkNDZjOTBmNWZhZGM3ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.IiZr1x9daLOMhBlrlVazoYi_75sRtXZA0ztep9esHIo)
> > But I can't click on read/write button, am I missing something?
> 
> @eerison My man, you definitely need to enable the debug for this
> 
> mine was first place the code in neo.ts and named to something like .js (I'm using spk.js (for sprinkler, duh!) in **CONFIG/zigbee2mqtt** ![image](https://private-user-images.githubusercontent.com/76679553/350844382-5bb20d12-df80-4937-a056-9ae66e0cf105.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE2MzMxMzYsIm5iZiI6MTcyMTYzMjgzNiwicGF0aCI6Ii83NjY3OTU1My8zNTA4NDQzODItNWJiMjBkMTItZGY4MC00OTM3LWEwNTYtOWFlNjZlMGNmMTA1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIyVDA3MjAzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM4YTk2YTc0NDA0ZGNkMGQ4ZDU2NDE5YjJjMzZkZTE2N2M0MDIyZTA0MTRmNjczNzM5MTJmNjUxYzBlNjE0YjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.m4vbT-0LptjV4_Net2mJfcd1T1cB9-154ZzEUGcvQ8w)
> 
> then set your configuration.yaml as this instruction ![image](https://private-user-images.githubusercontent.com/76679553/350845074-2d3357a5-cc46-453d-a5a8-50c510311e53.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE2MzMxMzYsIm5iZiI6MTcyMTYzMjgzNiwicGF0aCI6Ii83NjY3OTU1My8zNTA4NDUwNzQtMmQzMzU3YTUtY2M0Ni00NTNkLWE1YTgtNTBjNTEwMzExZTUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIyVDA3MjAzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYzYzU5Mjc3ZjM5MDRiNTU4OTc4Yjg2MjBmNzIyNTA4ZDljOWIxYjcyOTkxZTE1ZWRhNTIwNTgyOGZmNmZiOGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7nEdVHVCz3l8CWrbsnB6A4ai7gJ97Y8YdsAZHqFpVys)
> 
> mine was like this ![image](https://private-user-images.githubusercontent.com/76679553/350845371-4911929b-df08-4aa2-b5e0-75faa3d00b64.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE2MzMxMzYsIm5iZiI6MTcyMTYzMjgzNiwicGF0aCI6Ii83NjY3OTU1My8zNTA4NDUzNzEtNDkxMTkyOWItZGYwOC00YWEyLWI1ZTAtNzVmYWEzZDAwYjY0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIyVDA3MjAzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIzMTUyMGNiNTgxMDhlOTU0ZjhiNWE4NGM1NDM0MTVjMmU5NmIxNTE3YjFmODZkMmQxMzRjZDYyYTNjNGRhOTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.WX3BvE7tiTwu1AyGkLfVhLSosZPbTKsHTnxD9qlNiP0)
> 
> after restart zb2mqtt the device will be ready to add (you may need to check if your device's fingerprint is matching these? `fingerprint: tuya.fingerprint('TS0601', ['_TZE204_rzrrjkz2', '_TZE204_uab532m0']),`

Thank you @OgV1 <3, I will give a try today

eerison, 2024-07-22:
hey @OgV1 

Could you post your spk.js code here please, it will be easier to test

OgV1, 2024-07-22:
> hey @OgV1
> 
> Could you post your spk.js code here please, it will be easier to test

Yeah man feel free to try

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const GALLON_TO_LITER = 3.78541;

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_rzrrjkz2', '_TZE204_uab532m0']),
        zigbeeModel: ['NAS-WV03B'],
        model: 'NAS-WV03B',
        vendor: 'Neo',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        description: 'Smart sprinkler timer',
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the valve On or Off'),
            e.enum('status', ea.STATE, ['0', '1', '2', '3']).withDescription('Status'),
            e.numeric('countdown', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown'),
            e.numeric('countdown_left', ea.STATE).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown left'),
            e.numeric('water_current', ea.STATE).withUnit('L/min').withValueMin(0).withValueMax(3785.41).withValueStep(0.1)
                .withDescription('Current water flow (L/min)'), // Changed to liters
            e.battery(),
            e.numeric('water_total', ea.STATE).withUnit('L').withValueMin(0).withValueMax(378541.0).withValueStep(0.001)
                .withDescription('Total water flow (L)'), // Changed to liters
            e.binary('fault', ea.STATE, 'DETECTED', 'NOT_DETECTED').withDescription('Fault status'),
            e.enum('weather_delay', ea.STATE_SET, ['24h', '48h', '72h', 'cancel']).withDescription('Weather delay'),
            e.text('normal_timer', ea.STATE_SET).withDescription('Normal timer'),
            e.binary('switch_enabled', ea.STATE_SET, 'ON', 'OFF').withDescription('Switch enabled'),
            e.numeric('smart_irrigation', ea.STATE).withDescription('Smart irrigation'),
            e.binary('total_flow_reset_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Total flow reset switch'),
            e.numeric('quantitative_watering', ea.STATE_SET).withUnit('L').withValueMin(0).withValueMax(10000)
                .withDescription('Quantitative watering'), // Changed to liters
            e.binary('flow_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Flow switch'),
            e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withDescription('Child lock'),
            e.numeric('surplus_flow', ea.STATE).withDescription('Surplus flow'),
            e.numeric('single_watering_duration', ea.STATE).withDescription('Single watering duration'),
            e.numeric('single_watering_amount', ea.STATE).withDescription('Single watering amount'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff],
                [3, 'status', tuya.valueConverterBasic.lookup({'Off': 0, 'one': 1, 'Child lock activated': 2, 'On': 3, 'On_btn': 4})],
                [5, 'countdown', tuya.valueConverter.raw],
                [6, 'countdown_left', tuya.valueConverter.raw],
                [9, 'water_current', {
                    from: (v) => {
                        convertedValue = (v * GALLON_TO_LITER / 1000).toFixed(1); // Converts the raw value from gal/min to L/min with one decimal place
                        console.log(`Raw value for water_current: ${v}, Converted value: ${convertedValue}`);
                        return convertedValue;
                    },
                    to: (v) => {
                        const rawValue = Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000); // Converts L/min back to the raw value in gal/min
                        console.log(`Raw value for water_current: ${v}, Raw value: ${rawValue}`);
                        return rawValue;
                    },
                }],
                [11, 'battery', tuya.valueConverter.raw],
[... 10 lines cut ...]
                [102, 'quantitative_watering', {
                    from: (v) => (v * GALLON_TO_LITER / 1000).toFixed(1), // Converts the raw value from gallons to liters with one decimal place
                    to: (v) => Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000) // Converts liters back to the raw value in gallons
                }],
                [103, 'flow_switch', tuya.valueConverter.onOff],
                [104, 'child_lock', tuya.valueConverter.onOff],
                [105, 'surplus_flow', tuya.valueConverter.raw],
                [106, 'single_watering_duration', tuya.valueConverter.raw],
                [108, 'single_watering_amount', {
                    from: (v) => {
                        const convertedValue = (v * GALLON_TO_LITER / 1000).toFixed(1);
                        console.log(`Raw value for single_watering_amount: ${v}, Converted value: ${convertedValue}`);
                        return convertedValue;
                    },
                    to: (v) => {
                        const rawValue = Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000);
                        console.log(`Value to be sent for single_watering_amount: ${v}, Raw value: ${rawValue}`);
                        return rawValue;
                    }
                }],
            ],
        },
};

module.exports = definition;`

eerison, 2024-07-23:
just to be easy to copy and past in case someone else wants to test your code :)

```js
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const GALLON_TO_LITER = 3.78541;

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_rzrrjkz2', '_TZE204_uab532m0']),
        zigbeeModel: ['NAS-WV03B'],
        model: 'NAS-WV03B',
        vendor: 'Neo',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        description: 'Smart sprinkler timer',
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the valve On or Off'),
            e.enum('status', ea.STATE, ['0', '1', '2', '3']).withDescription('Status'),
            e.numeric('countdown', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown'),
            e.numeric('countdown_left', ea.STATE).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown left'),
            e.numeric('water_current', ea.STATE).withUnit('L/min').withValueMin(0).withValueMax(3785.41).withValueStep(0.1)
                .withDescription('Current water flow (L/min)'), // Changed to liters
            e.battery(),
            e.numeric('water_total', ea.STATE).withUnit('L').withValueMin(0).withValueMax(378541.0).withValueStep(0.001)
                .withDescription('Total water flow (L)'), // Changed to liters
            e.binary('fault', ea.STATE, 'DETECTED', 'NOT_DETECTED').withDescription('Fault status'),
            e.enum('weather_delay', ea.STATE_SET, ['24h', '48h', '72h', 'cancel']).withDescription('Weather delay'),
            e.text('normal_timer', ea.STATE_SET).withDescription('Normal timer'),
            e.binary('switch_enabled', ea.STATE_SET, 'ON', 'OFF').withDescription('Switch enabled'),
            e.numeric('smart_irrigation', ea.STATE).withDescription('Smart irrigation'),
            e.binary('total_flow_reset_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Total flow reset switch'),
            e.numeric('quantitative_watering', ea.STATE_SET).withUnit('L').withValueMin(0).withValueMax(10000)
                .withDescription('Quantitative watering'), // Changed to liters
            e.binary('flow_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Flow switch'),
            e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withDescription('Child lock'),
            e.numeric('surplus_flow', ea.STATE).withDescription('Surplus flow'),
            e.numeric('single_watering_duration', ea.STATE).withDescription('Single watering duration'),
            e.numeric('single_watering_amount', ea.STATE).withDescription('Single watering amount'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff],
                [3, 'status', tuya.valueConverterBasic.lookup({'Off': 0, 'one': 1, 'Child lock activated': 2, 'On': 3, 'On_btn': 4})],
                [5, 'countdown', tuya.valueConverter.raw],
                [6, 'countdown_left', tuya.valueConverter.raw],
                [9, 'water_current', {
                    from: (v) => {
                        convertedValue = (v * GALLON_TO_LITER / 1000).toFixed(1); // Converts the raw value from gal/min to L/min with one decimal place
                        console.log(`Raw value for water_current: ${v}, Converted value: ${convertedValue}`);
                        return convertedValue;
                    },
                    to: (v) => {
                        const rawValue = Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000); // Converts L/min back to the raw value in gal/min
                        console.log(`Raw value for water_current: ${v}, Raw value: ${rawValue}`);
                        return rawValue;
                    },
                }],
                [11, 'battery', tuya.valueConverter.raw],
                [15, 'water_total', {
                    from: (v) => (v * GALLON_TO_LITER / 1000).toFixed(1), // Converts the raw value from gallons to liters with one decimal place
                    to: (v) => Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000) // Converts liters back to the raw value in gallons
[... 8 lines cut ...]
                    from: (v) => (v * GALLON_TO_LITER / 1000).toFixed(1), // Converts the raw value from gallons to liters with one decimal place
                    to: (v) => Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000) // Converts liters back to the raw value in gallons
                }],
                [103, 'flow_switch', tuya.valueConverter.onOff],
                [104, 'child_lock', tuya.valueConverter.onOff],
                [105, 'surplus_flow', tuya.valueConverter.raw],
                [106, 'single_watering_duration', tuya.valueConverter.raw],
                [108, 'single_watering_amount', {
                    from: (v) => {
[... cut ...]
                        console.log(`Raw value for single_watering_amount: ${v}, Converted value: ${convertedValue}`);
                        return convertedValue;
                    },
                    to: (v) => {
                        const rawValue = Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000);
                        console.log(`Value to be sent for single_watering_amount: ${v}, Raw value: ${rawValue}`);
                        return rawValue;
                    }
                }],
            ],
        },
};

module.exports = definition;
```

eerison, 2024-07-30:
Hey @OgV1 

what did you enabled to see the `console.log` ? I have `log_level: debug`, But I don't see any console log :(

mpavlikWandera, 2024-07-30:
@eerison you need to go Settings -> System -> Logs -> pick Z2M addon in upper right corner

eerison, 2024-07-30:
well I could get some progress at lest 😄 

I tested console.debug on state and status field, and both I could see a log, But not with `flow_switch`, I belive it is happening because this error is happening before go into the `from and to` scope

```
[2024-07-30 10:57:12] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/sprinkler_01/set' with data '{"flow_switch":"ON"}'
[2024-07-30 10:57:12] error: 	z2m: No converter available for 'flow_switch' ("ON")
```

what this no converter available means? because there is a converter to `on/off` :/

eerison, 2024-07-30:
> I have the same valve and have found the tuya end point : {"1":"Switch", boolean "3":"Status", "type": "Enum", "values": "{"range":["off","auto","disabled"]}" "5":"Countdown", "type": "Integer", [...]

@dominique6017 are you sure that `Flow switch` doesn`t have any type?
because on your schema `Child lock`  doesn`t have any type, but it is a boolean, well it works hehehe

baggiowu, 2024-07-31:
> > ohhh .... it seems it may already exist? https://github.com/Koenkk/zigbee-herdsman-converters/blob/da65b1aeffd96527df02725b49de61e453fee059/src/devices/neo.ts#L130 just needs some tweaking?
> 
> Hi Guys,
> 
> I have created that converter based on this fingerprint **_TZE204_rzrrjkz2** in this [pull request](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7630)
> 
> And actually not sure if I did it right, because I still don't have a chance to test with my own device (Guess it because the fingerprint is now **_TZE204_uab532m0** and not the same as mine **_TZE204_rzrrjkz2**?)
> 
> Anyone have any idea how to solve [this](https://github.com/Koenkk/zigbee2mqtt/issues/22813#issuecomment-2183705606)? and why do we have different fingerprints on the same device?
> 
> edit: Somehow, mine was slightly different with no Neo logo on it, is there any chances that it not actually Neo's ? ![image](https://private-user-images.githubusercontent.com/76679553/342181863-38680fcb-693b-402b-ab2a-deb94c5b8803.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIzODY0MjYsIm5iZiI6MTcyMjM4NjEyNiwicGF0aCI6Ii83NjY3OTU1My8zNDIxODE4NjMtMzg2ODBmY2ItNjkzYi00MDJiLWFiMmEtZGViOTRjNWI4ODAzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDAwMzUyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ4OWZiNWM4ZTQ0YmUxNGRmYzQwNjVhZDA1YzY2OWY0ZTMwMjFhZDcxYjlhY2U3ODE3NzQzYTBlODk2OTlhMTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2A0K8NrK760Ky7YuchXhWRnpT37iKAjCzCarjCuhgBY)
> 
> FYI this is the info from my vendor: he seems to be sure that this one is Neo's with model number NAS-WV03B

**_TZE204_uab532m0**  its unit is Imperial gallons. 
**_TZE204_rzrrjkz2** its unit is US gallons.
Their model is 'NAS-WV05B2'.

eerison, 2024-07-31:
> **_TZE204_uab532m0** its unit is Imperial gallons. **_TZE204_rzrrjkz2** its unit is US gallons. Their model is 'NAS-WV05B2'.

Yep I also saw the model was wrong :/

I am looking forward to see your PR merged 🤞🏼

OgV1, 2024-08-02:
> @OgV1 how do you know those fields are correct?
> 
> ```
> {
>     "battery": 67,
>     "child_lock": "OFF",
>     "countdown": 4,
>     "countdown_left": 0,
>     "linkquality": 116,
>     "quantitative_watering": "0.0",
>     "single_watering_amount": "0.0",
>     "single_watering_duration": 0,
>     "state": "OFF",
>     "status": "Off",
>     "water_current": "0.0",
>     "water_total": "0.0",
>     "fault": null,
>     "flow_switch": null,
>     "normal_timer": null,
>     "smart_irrigation": null,
>     "surplus_flow": null,
>     "switch_enabled": null,
>     "total_flow_reset_switch": null,
>     "weather_delay": null
> }
> ```
> 
> did you use Tuya Iot Platform? https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

Hi @eerison 

Yes, I use the same method in that link (The Automatic method on step 8.)
[image]

eerison, 2024-08-02:
The PR from @baggiowu was closed :(

Could you send here the debug page, then I could see the fields and types.

like this one:
[image]

Note: this screenshot is from an wifi device that I have, as I don't have an getaway tuya I can't test the sprinkler :(

OgV1, 2024-08-03:
Yeah, I've tried update my zb2mqtt and that's the problem, seems I got the latest version of code from @baggiowu and I cannot access the debug anymore. :(

OgV1, 2024-08-03:
Try this

Standard Status Set
Code	Type	Values
switch	Boolean	
"{true,false}"
status	Enum	
{
  "range": [
    "off",
    "auto",
    "disabled"
  ]
}
countdown	Integer	
{
  "unit": "min",
  "min": 1,
  "max": 240,
  "scale": 0,
  "step": 1
}
countdown_left	Integer	
{
  "unit": "min",
  "min": 1,
  "max": 240,
  "scale": 0,
  "step": 1
}
water_current	Integer	
{
  "unit": "gal/min",
  "min": 0,
  "max": 1000000000,
  "scale": 3,
  "step": 1
}
battery_percentage	Integer	
{
  "unit": "%",
  "min": 0,
  "max": 100,
  "scale": 0,
  "step": 1
}
water_total	Integer	
{
  "unit": "gal",
  "min": 0,
  "max": 999999999,
  "scale": 3,
  "step": 1
}
weather_delay	Enum	
{
  "range": [
    "24h",
    "48h",
    "72h",
    "cancel"
  ]
}
normal_timer	String	
{
  "maxlen": 255
}
weather_switch	Boolean	
"{true,false}"
switch_enabled	Boolean	
"{true,false}"

OgV1, 2024-08-03:
After I tried with @baggiowu version released, there looks like I still get some errors

1. the water total is keeps showing 0
2. the flow switch and total flow switch will reported 'no converter' when I turned them ON of OFF
[image]

wifijt, 2024-08-06:
I have been playing with this code for the last few days - I can't get the battery to show at all. I have a Tuya account and went in and looked at the code - it says it's 11 and when connected to the Tuya cloud it reports as 100%.  I have gone back and forth with the e.battery() vs e.numeric(..) and the raw vs batterypercentage  to no affect. 

Is there a way in debug to see the report from the device? or does the topic have to be created with the correct data points for it to be published? 

My product name in Tuya is Zigbee 05浇灌开关 E版本 美制单位gal

Which translates to Zigbee 05 watering switch E version American unit gal

Here is my code just in case I've made a stupid mistake:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_rzrrjkz2', '_TZE204_uab532m0']),
        zigbeeModel: ['NAS-WV03B'],
        model: 'NAS-WV03B',
        vendor: 'Neo',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        description: 'Smart sprinkler timerzz',
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.switch(),
            e.enum('status', ea.STATE, ['off', 'auto', 'disabled', 'app_manual', 'key_control']).withDescription('Status'),
            e.numeric('countdown', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown'),
            e.numeric('countdown_left', ea.STATE).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown left'),
            e
                .numeric('water_current', ea.STATE)
                .withUnit('gal/min')
                .withValueMin(0)
                .withValueMax(3785.41)
                .withValueStep(0.001)
                .withDescription('Current water flow (L/min)'),
            e.numeric('battery_percentage', ea.STATE).withUnit('%').withValueMin(0).withValueMax(100).withDescription('Battery percentage'),
            e
                .numeric('water_total', ea.STATE)
                .withUnit('G')
                .withValueMin(0)
                .withValueMax(378541.0)
                .withValueStep(0.001)
                .withDescription('Total water flow (G)'),
            e.binary('fault', ea.STATE, 'DETECTED', 'NOT_DETECTED').withDescription('Fault status'),
            e.enum('weather_delay', ea.STATE_SET, ['24h', '48h', '72h', 'cancel']).withDescription('Weather delay'),
            e.text('normal_timer', ea.STATE_SET).withDescription('Normal timer'),
            e.binary('switch_enabled', ea.STATE_SET, 'ON', 'OFF').withDescription('Switch enabled'),
            e.numeric('smart_irrigation', ea.STATE).withDescription('Smart irrigation'),
            e.binary('total_flow_reset_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Total flow reset switch'),
            e
                .numeric('quantitative_watering', ea.STATE_SET)
                .withUnit('G')
                .withValueMin(0)
                .withValueMax(10000)
                .withDescription('Quantitative watering'),
            e.binary('flow_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Flow switch'),
            e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withDescription('Child lock'),
            e.numeric('surplus_flow', ea.STATE).withDescription('Surplus flow'),
            e.numeric('single_watering_duration', ea.STATE).withDescription('Single watering duration'),
            e.numeric('single_watering_amount', ea.STATE).withDescription('Single watering amount'),
        ],
        meta: {
           tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff],
                [
                    3,
                    'status',
                    tuya.valueConverterBasic.lookup({
                        off: tuya.enum(0),
                        auto: tuya.enum(1),
                        disabled: tuya.enum(2),
                        app_manual: tuya.enum(3),
                        key_control: tuya.enum(4),
                    }),
                ],
                [109, 'countdown', tuya.valueConverter.raw],
                [6, 'countdown_left', tuya.valueConverter.raw],
                [9, 'water_current', tuya.valueConverter.raw],
                [11, 'battery_percentage', tuya.valueConverter.raw],
                [15, 'water_total', tuya.valueConverter.raw],
                [19, 'fault', tuya.valueConverter.raw],
                [37, 'weather_delay', tuya.valueConverter.raw],
                [38, 'normal_timer', tuya.valueConverter.raw],
                [42, 'switch_enabled', tuya.valueConverter.onOff],
                [47, 'smart_irrigation', tuya.valueConverter.raw],
                [101, 'total_flow_reset_switch', tuya.valueConverter.onOff],
                [102, 'quantitative_watering', tuya.valueConverter.raw],
                [103, 'flow_switch', tuya.valueConverter.onOff],
                [104, 'child_lock', tuya.valueConverter.onOff],
                [105, 'surplus_flow', tuya.valueConverter.raw],
                [106, 'single_watering_duration', tuya.valueConverter.raw],
                [108, 'single_watering_amount', tuya.valueConverter.raw],
            ],
        },
    
};

module.exports = definition;
`

eerison, 2024-08-21:
Yeah I also tested @baggiowu implementation,  but it didn't work 😕

eerison, 2024-08-21:
Hey @OgV1 I saw that @baggiowu added different models, which model appeared for you?

NAS-WV03B?

eerison, 2024-08-21:
> **_TZE204_uab532m0** its unit is Imperial gallons. **_TZE204_rzrrjkz2** its unit is US gallons. Their model is 'NAS-WV05B2'.

I think the problem is , whe are still taking the old model, I guess it should be `NAS-WV05B2` as mentioned by @baggiowu.
But I don't know why it is still the old one `NAS-WV03B`

andrewcfitz, 2024-09-02:
> 30

My device keeps turning off after 30 seconds, any idea what is going on?

andrewcfitz, 2024-09-02:
> And I found another weird behavior. Looks like after we press the button, (doesn't matter remotely or physical) the device will switch off by itself after around 30 seconds, the reason is might because that there is no water through the valve, I have to test.
> Edit: Alright, I have tested, doesn't stop by itself, there is nothing to worry about this.

My deivce is turning off after 30 seconds, any idea what is going on?

wifijt, 2024-09-03:
The device will shut off if it doesn’t detect flow. Found the same thing on
mine.

Still no battery for me…

OgV1, 2024-09-05:
> Hey @OgV1 I saw that @baggiowu added different models, which model appeared for you?
> 
> NAS-WV03B?

Hi @eerison so sorry for taking so long, now I'm using NAS-WV05B2 like @baggiowu mentioned in

> **_TZE204_uab532m0** its unit is Imperial gallons. **_TZE204_rzrrjkz2** its unit is US gallons. Their model is 'NAS-WV05B2'.

bikingbadger, 2024-09-30:
I just picked one of these up and had to add _TZE204_z7a2jmyy to the fingerprint:

fingerprint: tuya.fingerprint('TS0601', ['_TZE204_rzrrjkz2', '_TZE204_uab532m0','_TZE204_z7a2jmyy']),

I found it by accident looking at the About tab:
[image]

Not sure if there is a way to add this too, my technical knowledge here is down to luck and copy and paste but sharing if someone else might come across this post

Koenkk, 2024-12-11:
Added `_TZE204_z7a2jmyy`!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-11 by Koenkk

eerison, 2025-01-24:
When I turn on and off I am getting this error message:
[image]

and for some reason the water flow is so low :(