Koenkk/zigbee2mqtt #22816: zh:ember: Delivery of BROADCAST failed for "65533" on fw 7.4.1, 7.4.2, and 7.43

lennon101, 2024-05-29:
### What happened?

- Journal of steps
	- upgraded zigbee2mqtt to 1.37
	- received the `zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462`
	- changed adapter: ezsp to adapter: ember and restarted
	- got an error that my coordinator was not on EZSP
	- upgraded my coordinator firmware to FW ncp-uart-hw-v7.4.2.0-zbdonglee-230400 using the steps outlined here:
		- https://github.com/Koenkk/zigbee2mqtt/issues/22398#issuecomment-2094674395
	- changed config in zigbee2mqtt to "adapter: ember" + "rtscts: false"
	- restarted zigbee2mqtt and zigbee network is working (mostly)
	- now at startup I get the message `zh:ember: Delivery of BROADCAST failed for "65533" `
	- pairing new entities does not work due to the same error
	- switching back to "adapter: ezsp" doesn't work either as I then get the error "zh:controller:greenpower: Received undefined command from '0'". another used already created a ticket for this.
	- Found this closed issue saying reporting exactly the same issue saying that 7.4.2 was the problem:
		- https://github.com/Koenkk/zigbee2mqtt/issues/22469
	- Downgraded zigbee coordinator fw to 7.4.1 using: `universal-silabs-flasher --device /dev/ttyACM0 flash --firmware ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl --force`
	- got the same error as before: `zh:ember: Delivery of BROADCAST failed for "65533" `
	- upgraded zigbee coordinator fw to 7.4.3 using: `universal-silabs-flasher --device /dev/ttyACM0 flash --firmware ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl --force`
	- got the same error as before: `zh:ember: Delivery of BROADCAST failed for "65533" `
	-

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

ncp-uart-hw-v7.4.3.0-zbdonglee-115200

### Adapter

zbdongle-e

### Setup

docker container

### Debug log

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15477318/log.log)

lennon101, 2024-05-29:
Update: 

- rolled back to zigbee2mqtt **server** version`1.36.1` and changed the settings in the configuation.yaml
	-
	  ```
	  serial:
	    port: /dev/ttyACM0
	    adapter: ezsp
	    baudrate: 115200
	    rtscts: false
	  ```
- Network returned to a completely working state (all devices have returned to network and all routers have also returned)
	- no errors in the log either
- ## Final state of working system is:
	- Zigbee2mqtt Server Version: `1.36.1` running docker container
	- Zigbee2mqtt coordinator version: `7.4.3`
	- configuration.yaml settings specific to fixing this issue:
		-
		  ```
		  serial:
		    adapter: ezsp
		  ```

Nerivec, 2024-06-01:
This is already being discussed in https://github.com/Koenkk/zigbee2mqtt/issues/22453
Unfortunately, there appears to be a half dozen or so different "fixes" from different people, all pointing to nowhere specific at the moment... And we have yet to be able to reproduce it in this manner, making it that much harder to dissect...

lennon101, 2024-06-03:
Apologies for the duplication! I did a search but didn't find that thread. I appreciate any/all efforts you are all doing to get this fixed. I'll endevour to keep looking it on my end and report back with any findings. Thanks!

kwetnico, 2024-06-15:
Hello
I have the same problem..
Sonoff with firmware 7.4.2

Nerivec, 2024-06-15:
Closing this thread in favor of the other one mentioned above to avoid feedback in multiple places.

closed 2024-06-15 by Nerivec

clertonraf, 2024-08-08:
Just to comment that I had the same configuration and the same problem. I went through https://github.com/Koenkk/zigbee2mqtt/issues/22453 and tried a bunch of things, but this proposed solution solved my issues.

lennon101, 2024-08-27:
> but this proposed solution solved my issues.

What proposed solution are you referring to?

Nerivec, 2024-08-28:
As mentioned in the other thread, issue should be solved in September release (current dev branch).

lennon101, 2024-08-28:
> As mentioned in the other thread, issue should be solved in September release (current dev branch).

Had a good dig around "the other thread" but couldn't find where anyone mentions the fix. Could you provide a link to a specific post? I'm glad to hear that the issue is getting fixed!

Nerivec, 2024-08-30:
https://github.com/Koenkk/zigbee2mqtt/issues/22453#issuecomment-2294108058
https://github.com/Koenkk/zigbee2mqtt/issues/22453#issuecomment-2295280123

jespervdw, 2024-12-21:
Had the same problem. Broadcast error every 10min. Reintalling Z2M and updating firmware didn't work.
For now, the error seems not to return since I don't allow new devices to join. 

`Zigbee2MQTT versie
[1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0) commit: [bc6eccf](https://github.com/Koenkk/zigbee2mqtt/commit/bc6eccf)
Coördinator type
EmberZNet
Coordinator revisie
7.4.5 [GA]
Coordinator IEEE adres
0xe0798dfffecfaeef
Versie frontend
0.7.6
zigbee_herdsman_converters_version
20.58.0
zigbee_herdsman_version
2.1.9`