Koenkk/zigbee2mqtt #22827: [New device support]: Luminea ZX-8588 (Looks like a whitelabel of TuYa QT-07S)

systemofapwne, 2024-05-29:
### Link

https://www.luminea.info/KAT158-Smarter-Zigbee-Sensor-zur-Feuchtigkeits-und-ZX-8588-919.shtml

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0xa4c138c4c50864a9","nwkAddr":26258,"manufId":4417,"manufName":"_TZE200_9cqcpkgb","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���-\u0011","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":72,"manufacturerName":"_TZE200_9cqcpkgb","zclVersion":3,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1717009458448}

### Comments

The ZX-8588 is a soil sensor, that has the same package as the TuYa QT-07S.
However, it looks like the reported datapoints have different datapoint IDs for temperature compared to the QT-07S.
I did not check, if it also supports illumination. When first powering it up, I only saw a few raw packages like this:

```2024-05-25T15:10:33.163226+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:10:33] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":10496}'
2024-05-25T15:10:33.692897+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:10:33] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":10496}'
2024-05-25T15:10:34.244057+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:10:34] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":10496}'
2024-05-25T15:10:34.813832+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:10:34] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":10496}'
2024-05-25T15:10:35.349340+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:10:35] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":10496}'
2024-05-25T15:10:36.456191+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:10:36] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3}],"seq":10752}'
2024-05-25T15:10:37.590425+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:10:37] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":11008}'
2024-05-25T15:10:43.329906+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:10:43] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":11264}'
2024-05-25T15:16:15.394877+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:16:15] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandMcuVersionResponse' and data '{"seq":8960,"version":81}'
2024-05-25T15:16:15.671124+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:16:15] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,25],"type":"Buffer"},"datatype":2,"dp":5}],"seq":12288}'
2024-05-25T15:16:15.800629+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:16:15] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3}],"seq":12544}'
2024-05-25T15:16:15.927452+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:16:15] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":12800}'
2024-05-25T15:16:16.058370+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:16:16] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":13056}'
2024-05-25T15:16:16.958372+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:16:16] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,25],"type":"Buffer"},"datatype":2,"dp":5}],"seq":13312}'
2024-05-25T15:16:17.084283+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:16:17] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3}],"seq":13568}'
2024-05-25T15:16:17.221157+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:16:17] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":13824}'
2024-05-25T15:16:17.359430+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:16:17] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":14080}'
2024-05-25T15:21:38.255286+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:21:38] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":14336}'
2024-05-25T15:21:38.829588+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:21:38] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":14336}'
2024-05-25T15:21:39.357249+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:21:39] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":14336}'
2024-05-25T15:21:39.908295+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:21:39] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":14336}'
2024-05-25T15:21:40.478860+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:21:40] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":14336}'
2024-05-25T15:21:41.016661+02:00 Debian Docker: iot_zigbee_1[854]: [2024-05-25 13:21:41] #033[34mdebug#033[39m: #011z2m: No converter available for 'ZX-8588' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq": [... cut ...]
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.battery(), e.temperature(), e.humidity()],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [3, 'humidity', tuya.valueConverter.raw], // <- mapped the datapoint
            [5, 'temperature', tuya.valueConverter.raw], // <- mapped the datapoint
            [15, 'battery', tuya.valueConverter.raw], // <- mapped the datapoint
        ],
    },
};

module.exports = definition;
```

Koenkk, 2024-05-30:
It's definitely different than the `QT-07S` which is a `TS0222`, this is a `TS0601`. Your external definition looks good, could you make a PR?

CrashHouse, 2024-06-02:
Hello, mine is sold as a QT-07S but as a TS0601. And OT isn’t recognised too by Z2M.

closed 2024-12-29 by github-actions[bot]