Koenkk/zigbee2mqtt #22833: [New device support]: eWeLink CK-BL702-MWS-01(7016)

Izya12, 2024-05-30:
### Link

https://aliexpress.ru/item/1005006811581649.html

### Database entry

{"id":33,"type":"Router","ieeeAddr":"0x7cb94c7700220000","nwkAddr":27405,"manufId":4742,"manufName":"eWeLink","powerSource":"Unknown","modelId":"CK-BL702-MWS-01(7016)","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,3,4,5,1030,4096,64512,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"hwVersion":0}},"msOccupancySensing":{"attributes":{"occupancy":0}}},"binds":[{"cluster":1030,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b82e8e","endpointID":1}],"configuredReportings":[{"cluster":1030,"attrId":0,"minRepIntval":10,"maxRepIntval":60,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20210610-BL","swBuildId":"1.0.2","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1717039025492}

### Comments

I can't seem to get the converter right.
I'm asking for help.

```
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,6,4,13,107,1,1,0,116,0,215,227,72,0,0,4,24,41,7,0,13,107,29,248]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,6,4,13,107,1,1,0,116,0,215,227,72,0,0,4,24,41,7,0,13,107,29,248]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,6,4,13,107,1,1,0,116,0,215,227,72,0,0,4,24,41,7,0,13,107,29] - 248
[2024-05-30 08:17:05] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":1030,"srcaddr":27405,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":116,"securityuse":0,"timestamp":4776919,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[24,41,7,0]}}
[2024-05-30 08:17:05] debug: 	zh:controller: Received payload: clusterID=1030, address=27405, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=116, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":41,"commandIdentifier":7},"payload":[{"status":0}],"command":{"ID":7,"name":"configReportRsp","parameters":[{"name":"status","type":32},{"name":"direction","type":32,"conditions":[{"type":"minimumRemainingBufferBytes","value":3}]},{"name":"attrId","type":33,"conditions":[{"type":"minimumRemainingBufferBytes","value":2}]}]}}
[2024-05-30 08:17:05] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/0x7cb94c7700220000', payload '{"last_seen":"2024-05-30T03:17:05.400Z","linkquality":116,"occupancy":null}'
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-30 08:17:05] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2024-05-30 08:17:05] debug: 	zh:controller:endpoint: ZCL command 0x7cb94c7700220000/1 msOccupancySensing.read(["occupancy"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-30 08:17:05] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x7cb94c7700220000:27405/1 (0,0,1)
[2024-05-30 08:17:05] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":27405,"destendpoint":1,"srcendpoint":1,"clusterid":1030,"transid":108,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,42,0,0,0]}}
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,13,107,1,1,6,4,108,0,30,5,16,42,0,0,0,3]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,6,4,13,107,1,1,0,116,0,50,236,72,0,0,7,24,3,10,0,0,24,0,13,107,29,45]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,6,4,13,107,1,1,0,116,0,50,236,72,0,0,7,24,3,10,0,0,24,0,13,107,29,45]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,6,4,13,107,1,1,0,116,0,50,236,72,0,0,7,24,3,10,0,0,24,0,13,107,29] - 45
[2024-05-30 08:17:05] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":1030,"srcaddr":27405,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":116,"securityuse":0,"timestamp":4779058,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,3,10,0,0,24,0]}}
[2024-05-30 08:17:05] debug: 	zh:controller: Received payload: clusterID=1030, address=27405, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=116, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":3,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":24,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-30 08:17:05] debug: 	z2m: Received Zigbee message from '0x7cb94c7700220000', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
[2024-05-30 08:17:05] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/0x7cb94c7700220000', payload '{"last_seen":"2024-05-30T03:17:05.435Z","linkquality":116,"occupancy":false}'
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-05-30 08:17:05] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,108,170]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,108,170]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,108] - 170
[2024-05-30 08:17:05] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":108}
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,6,4,13,107,1,1,0,116,0,5,250,72,0,0,8,24,42,1,0,0,0,24,0,13,107,29,38]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,6,4,13,107,1,1,0,116,0,5,250,72,0,0,8,24,42,1,0,0,0,24,0,13,107,29,38]
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,6,4,13,107,1,1,0,116,0,5,250,72,0,0,8,24,42,1,0,0,0,24,0,13,107,29] - 38
[2024-05-30 08:17:05] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":1030,"srcaddr":27405,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":116,"securityuse":0,"timestamp":4782597,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,42,1,0,0,0,24,0]}}
[2024-05-30 08:17:05] debug: 	zh:controller: Received payload: clusterID=1030, address=27405, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=116, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":42,"commandIdentifier":1},"payload":[{"attrId":0,"status":0,"dataType":24,"attrData":0}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2024-05-30 08:17:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-30 08:17:05] debug: 	z2m: Received Zigbee message from '0x7cb94c7700220000', type 'readResponse', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
[2024-05-30 08:17:05] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/0x7cb94c7700220000', payload '{"last_seen":"2024-05-30T03:17:05.492Z","linkquality":116,"occupancy":false}'
[2024-05-30 08:17:05] info: 	z2m: Successfully configured '0x7cb94c7700220000'
```

### External definition

```shell
const {identify, occupancy} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['CK-BL702-MWS-01(7016)'],
    model: 'CK-BL702-MWS-01(7016)',
    vendor: 'eWeLink',
    description: 'Automatically generated definition',
    extend: [identify(), occupancy()],
    meta: {},
};

module.exports = definition;
```

svara007, 2024-06-01:
I will support you. It doesn't work either. It would be nice to add support.

naziks, 2024-12-01:
@Izya12, do you have any updates on this sensor? I have recently bought the same one and now i'm trying to make it work :D

Izya12, 2024-12-04:
> @Izya12, do you have any updates on this sensor? I have recently bought the same one and now i'm trying to make it work :D

К сожалению новой информации нет, если у вас что-то получится, надеюсь что вы отпишитесь тут.
Unfortunately there is no new information, if you get something, I hope you'll post here.

Izya12, 2025-02-08:
У меня пока не получается подключить устройство.  
I'm not able to connect the device yet.

closed 2025-06-18 by github-actions[bot]