Koenkk/zigbee2mqtt #22844: [New device support]:

72longboard, 2024-05-31:
### Link

https://es.aliexpress.com/item/1005006923987304.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=6000095348167&aff_fcid=edc5a5efbe864160b08f0be54d56e580-1717115370159-07010-_onkx4EO&tt=MG&aff_fsk=_onkx4EO&aff_platform=default&sk=_onkx4EO&aff_trace_key=edc5a5efbe864160b08f0be54d56e580-1717115370159-07010-_onkx4EO&shareId=6000095348167&businessType=ProductDetail&platform=AE&terminal_id=4f343b0e05b24ec5a29e575a82deeb16&afSmartRedirect=y

### Database entry

{"id":18,"type":"EndDevice","ieeeAddr":"0x94b216fffef1b487","nwkAddr":47149,"manufId":4098,"manufName":"_TZ3210_5rta89nj","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,61184],"outClusterList":[10,25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1717113801806}

### Comments

Please make a new registration so that I can use this product

### External definition

```shell
const {identify, battery, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZ3210_5rta89nj',
    description: 'Automatically generated definition',
    extend: [identify(), battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;
```

jcob9320, 2024-05-31:
I found datapoints, but I'm not sure how to create an external converter.
datapoints is here,
`[4, 'battery_level'],
[102, 'device_control'],
[103, 'security_mode'],
[104, 'ratio_of_opening'],
[105, 'charge_state'],
[106, 'manual_mode'],
[107, 'fault'],
[108, 'motor_calibration'],
[109, 'installation_type'],
[110, 'slow_stop'],
[111, 'solar_energy_current'],
[112, 'fixed_window_sash'],
[113, 'motor_timeout'],
[114, 'window_state']
`

plplaaa2, 2024-06-16:
Although some progress has been made, the external converter can read data points but cannot be controlled.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: [
        '5rta89nj' // Tubular motors
    ],
    fingerprint: [
// Curtain motors:
        {modelID: 'TS0601', manufacturerName: '_TZ3210_5rta89nj'},
    ],
    model: 'TS0601',
    vendor: '_TZ3210_5rta89nj',
    description: 'window pusher',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(),
        e.cover_position('position', 'state', ea.STATE_SET,['OPEN', 'CLOSE', 'STOP']).withDescription('Window Control'),
        e.enum('alarm_mode', ea.STATE_SET, ['True', 'False']),
        e.binary('charge_state', ea.STATE, 'charging', 'discharging'),
        e.enum('manual_mode', ea.STATE_SET, ['True', 'False']),
        e.binary('error', ea.STATE, true, false).withDescription('Fault'),
        e.numeric('calibration', ea.STATE_SET).withValueMin(10).withValueMax(90).withDescription('Motor calibration').withUnit('s'),
        e.enum('motor_direction', ea.STATE_SET, ['left', 'right']).withDescription('Install Side'),
        e.enum('mode', ea.STATE_SET, ['True', 'False']).withDescription('Slow_stop'),
        e.numeric('countdown', ea.STATE_SET).withValueMin(10).withValueMax(90).withDescription('Motor Timeout').withUnit('s'),
        e.binary('window_detection', ea.STATE, true, false).withDescription('Window State'),
        
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [4, 'battery', tuya.valueConverter.raw],
            [102, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'CLOSE': tuya.enum(1), 'STOP': tuya.enum(2)})],
            [103, 'alarm_mode', tuya.valueConverterBasic.lookup({'True': tuya.enum(0), 'False': tuya.enum(1),})],
            [104, 'position', tuya.valueConverter.coverPosition],
            [104, 'position', tuya.valueConverter.raw],
            [105, 'charge_state', tuya.valueConverter.trueFalse1],
            [106, 'manual_mode', tuya.valueConverterBasic.lookup({'True': tuya.enum(0), 'False': tuya.enum(1),})],
            [107, 'error', tuya.valueConverter.trueFalse1],
            [108, 'calibration', tuya.valueConverter.calibration],
            [108, 'calibration', tuya.valueConverter.raw],
            [109, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            [110, 'mode', tuya.valueConverterBasic.lookup({'True': tuya.enum(0), 'False': tuya.enum(1)})],
            [113, 'countdown', tuya.valueConverter.countdown],
            [113, 'countdown', tuya.valueConverter.raw],
            [114, 'window_detection', tuya.valueConverter.trueFalse0],
            ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;

tribakzero, 2024-06-21:
I have the same device and the external converter works great to display some values like the battery, state, position, window_detection although it seems like the state is inverted (I open the window and it says it's closed).

Hope I could help actively but at least you have a beta tester here. 🙏

ezeghers, 2024-06-21:
> I have the same device and the external converter works great to display some values like the battery, state, position, window_detection although it seems like the state is inverted (I open the window and it says it's closed).
> 
> Hope I could help actively but at least you have a beta tester here. 🙏

Regarding the Open/closed state being reversed, it does the same in the the Moes or Smartlife app. Mine is mounted on the left side of the inside window and regardless of the settings in the app, it does not reverse. I sent it to the Moes support for them to investigate.

qkddn49, 2024-06-25:
> 나는 동일한 장치를 가지고 있으며 외부 변환기는 상태가 반전된 것처럼 보이지만 배터리, 상태, 위치, window_Detection과 같은 일부 값을 표시하는 데 훌륭하게 작동합니다(창을 열면 닫혀 있다고 표시됩니다).
> 
> 제가 적극적으로 도움을 드릴 수 있기를 바라지만 적어도 여기에는 베타 테스터가 있습니다. 🙏

.Can I get the source code?

tribakzero, 2024-06-26:
> > 나는 동일한 장치를 가지고 있으며 외부 변환기는 상태가 반전된 것처럼 보이지만 배터리, 상태, 위치, window_Detection과 같은 일부 값을 표시하는 데 훌륭하게 작동합니다(창을 열면 닫혀 있다고 표시됩니다).
> > 제가 적극적으로 도움을 드릴 수 있기를 바라지만 적어도 여기에는 베타 테스터가 있습니다. 🙏
> 
> .Can I get the source code?

https://github.com/Koenkk/zigbee2mqtt/issues/22844#issuecomment-2171450055

qkddn49, 2024-06-26:
It's explained that it can't be controlled
Is it controlled?

tribakzero, 2024-06-26:
> It's explained that it can't be controlled Is it controlled?

Current external converter allows to read values, no control enabled at the moment.
> [Although some progress has been made, the external converter can read data points but cannot be controlled.](https://github.com/Koenkk/zigbee2mqtt/issues/22844#issuecomment-2171450055)

qkddn49, 2024-06-26:
I got it.
Thank you.

acortelyou, 2024-07-23:
@kkossev Based on [your work](https://raw.githubusercontent.com/kkossev/Hubitat/development/Drivers/ZigBee%20Window%20Shade/Zigbee_Window_Shade_lib_included.groovy) I've got something partially working.

The missing link was `tuyaSendCommand: 0x04` as you discovered.

Below is the work in progress, once I've had some time to play around with it and figure out what is and isn't actually working I'll trying submitting the device.

For now at least I can open and close via automation so #closeenough.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZ3210_5rta89nj',
        },
    ],
    vendor: 'MOES',
    model: 'ZC-LP01',
    description: 'Smart Sliding Window Pusher',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(),
        tuya.exposes.batteryState(),
        e.cover_position('position', 'state', ea.STATE_SET, ['Open', 'Close', 'Stop']).withDescription('Window Control'),
        e.enum('alarm_mode', ea.STATE_SET, ['True', 'False']),
        e.binary('charge_state', ea.STATE, 'Charging', 'Discharging'),
        e.enum('manual_mode', ea.STATE_SET, ['True', 'False']),
        e.binary('fault', ea.STATE, ['Motor Fault', 'Clear']).withDescription('Fault'),
        e.numeric('calibration', ea.STATE_SET).withValueMin(10).withValueMax(90).withDescription('Motor Calibration').withUnit('s'),
        e.enum('motor_direction', ea.STATE_SET, ['Left', 'Right']).withDescription('Motor Direction'),
        e.enum('slow_stop', ea.STATE_SET, ['Enabled', 'Disabled']).withDescription('Slow Stop'),
        e.numeric('solar_energy_current', ea.STATE).withValueMin(0).withValueMax(99999).withDescription('Solar Energy Current'),
        e.enum('fixed_window_sash', ea.STATE_SET, ['True', 'False']),
        e.numeric('motor_timeout', ea.STATE_SET).withValueMin(10).withValueMax(90).withDescription('Motor Timeout').withUnit('s'),
        e.binary('window_detection', ea.STATE, ['True','False']).withDescription('Window Detection'),

    ],
    meta: {
        tuyaSendCommand: 0x04,
        tuyaDatapoints: [
            [4, 'battery', tuya.valueConverter.raw],
            [102, 'state', tuya.valueConverterBasic.lookup({open: 0, close: 1, stop: 2})],
            [103, 'alarm_mode', tuya.valueConverter.trueFalse0],
            [104, 'position', tuya.valueConverter.coverPosition],
            [104, 'position', tuya.valueConverter.raw],
            [105, 'charge_state', tuya.valueConverter.trueFalse1],
            [106, 'manual_mode', tuya.valueConverter.trueFalse0],
            [107, 'fault', tuya.valueConverter.trueFalse1],
            [108, 'calibration', tuya.valueConverter.raw],
            [109, 'motor_direction', tuya.valueConverterBasic.lookup({left: 0, right: 1})],
            [110, 'slow_stop', tuya.valueConverter.trueFalse0],
            [111, 'solar_energy_current', tuya.valueConverter.raw]
            [112, 'fixed_window_sash', tuya.valueConverter.trueFalse0],
            [113, 'motor_timeout', tuya.valueConverter.countdown],
            [113, 'motor_timeout', tuya.valueConverter.raw],
            [114, 'window_detection', tuya.valueConverter.trueFalse0],
        ],
    },
    extend: [
    ],
};

module.exports = definition;
```

72longboard, 2024-07-23:
> @kkossev Based on [your work](https://raw.githubusercontent.com/kkossev/Hubitat/development/Drivers/ZigBee%20Window%20Shade/Zigbee_Window_Shade_lib_included.groovy) I've got something partially [...]

It's Working!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Great Thx!!!!

lutterodt, 2024-07-27:
> @kkossev Based on [your work](https://raw.githubusercontent.com/kkossev/Hubitat/development/Drivers/ZigBee%20Window%20Shade/Zigbee_Window_Shade_lib_included.groovy) I've got something partially [...]

This actually works! Frontend control achieved! HA Community man! However, as indicated by a contributor up top, mounting the pusher on the left side of the "pushed window" is inverted on the controls. Looking to see if a DP for inversion control is available like it is on other zigbee devices.

ezeghers, 2024-07-27:
Gave this a try. Zigbee2mqtt responds to changes but I can't control it. If I manually open or close the window, it changes in Z3M but if I try to open or close the window in HA/Z2M the window does not respond. I checked that it is loading the updated config given above. Everything does work when I connect it through the Moes app. 

A few weeks ago, I reported the left side/right side reversed issue and they gave me a firmware update for the window pusher. That update fixed the reverse issue in the Moes app. I wonder if that firmware update is the cause of this issue.

tribakzero, 2024-07-27:
@ezeghers  mind sharing the firmware to test it?

ezeghers, 2024-07-27:
> @ezeghers mind sharing the firmware to test it?

I'm not sure how I can. They pushed it to me through the Moes app.

ezeghers, 2024-07-27:
Looks like I got it to work. I had to remove the pusher device in Z2M, saved my Z2M configuration, uninstalled Z2M from HA, reinstalled Z2M, restored the configuration and re-linked the device.

tribakzero, 2024-07-27:
@ezeghers  great news, so you have the switch correctly now and also able to trigger it from Z2M?

I'll ping moes for the firmware, thanks!

acortelyou, 2024-07-27:
The version I posted has bugs with the converters still, but the open/close commands do work.

I haven't had time to take a look but I suspect it will be a quick fix, just need to find docs on the "right" way to model the datapoints.

If someone wants to post the doc or fix it quicker feel free.

I suggest we align with the hubitat version.

ezeghers, 2024-07-27:
> @ezeghers great news, so you have the switch correctly now and also able to trigger it from Z2M?
> 
> I'll ping moes for the firmware, thanks!

It works and the open/closed state kind of works. Using the cover entity I can slide up to open the window and down to close, which is what I would expect. The status of the cover however always says Closed.
[image]
[image]

plplaaa2, 2024-07-28:
I made a nearly finished yaml. I hope everyone uses this code well.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: [
        '5rta89nj' // Tubular motors
    ],
    fingerprint: [
// Curtain motors:
        {modelID: 'TS0601', manufacturerName: '_TZ3210_5rta89nj'},
    ],
    model: 'ZC-LP01',
    vendor: 'Moes',
    description: 'window pusher',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(),
        e.cover_position('position', 'state', ea.STATE_SET,['OPEN', 'CLOSE', 'STOP']).withDescription('Window Control'),
        e.enum('alarm_mode', ea.STATE_SET, ['True', 'False']),
        e.binary('charge_state', ea.STATE, 'charging', 'discharging'),
        e.enum('manual_mode', ea.STATE_SET, ['True', 'False']),
        e.binary('fault', ea.STATE, ['Motor Fault', 'Clear']).withDescription('Fault'),
        e.numeric('countdown', ea.STATE_SET).withValueMin(10).withValueMax(90).withDescription('Mortor timeout').withUnit('s'),
        e.enum('motor_direction', ea.STATE_SET, ['left', 'right']).withDescription('Install Side Type'),
        e.enum('mode', ea.STATE_SET, ['True', 'False']).withDescription('Slow_stop'),
        e.numeric('current', ea.STATE).withValueMin(0).withValueMax(99999).withDescription('Solar Energy Current'),
        e.enum('motor_type', ea.STATE_SET, ['True', 'False']),
        e.enum('window_detection', ea.STATE, ['Opened', 'Closed', 'Pending']).withDescription('Window Detection'),
        
    ],
    meta: {
        // All datapoints go in here
        tuyaSendCommand: 0x04,
        tuyaDatapoints: [
            [4, 'battery', tuya.valueConverter.raw],
            [102, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'CLOSE': tuya.enum(1), 'STOP': tuya.enum(2)})],
            [103, 'alarm_mode', tuya.valueConverterBasic.lookup({'True': tuya.enum(0), 'False': tuya.enum(1),})],
            [104, 'position', tuya.valueConverter.coverPosition],
            [104, 'position', tuya.valueConverter.raw],
            [105, 'charge_state', tuya.valueConverter.trueFalse1],
            [106, 'manual_mode', tuya.valueConverterBasic.lookup({'True': tuya.enum(0), 'False': tuya.enum(1),})],
            [107, 'fault', tuya.valueConverter.trueFalse1],
            [108, 'countdown', tuya.valueConverter.raw],
            [109, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            [110, 'mode', tuya.valueConverterBasic.lookup({'True': tuya.enum(1), 'False': tuya.enum(0)})],
            [111, 'current', tuya.valueConverter.divideBy1000],
            [112, 'motor_type', tuya.valueConverterBasic.lookup({'True': tuya.enum(0), 'False': tuya.enum(1)})],
            [114, 'window_detection', tuya.valueConverterBasic.lookup({Open: 0, Close: 1, Stop: 2})],
            ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

acortelyou, 2024-07-29:
dp 102 seems to be "set state" while dp 114 seems to be "get state", haven't figured out how to expose that properly yet.

Maybe `composite()`? Maybe `commandsWindowCovering()`? Maybe action? 

Also there is `e.cover_mode()` which expected `reversed` aka `motor_direction`, `maintenance` aka `manual_mode`, and `calibration` aka `countdown` but i'm not sure how to map correctly either because this device seems to use enums and cover mode uses binary.

Zigbee is crazy @Koenkk I can't sort out how I'm "supposed" to do anything.

Koenkk, 2024-07-30:
If the converter from @plplaaa2 works well, could somebody make pull request to [`tuya.ts`](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts) for out-of-the-box support?

acortelyou, 2024-08-06:
I know it's missing some datapoints but, more importantly, the following is working 100% as expected without any errors for me.

I experimented with the other datapoints but they either didn't have an effect or didn't work in ways I understood.

If someone has an actual tuya gateway and wants to debug the extra settings go for it but this is good enough for me at the moment: full control and state of the basic functions.

```
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZ3210_5rta89nj']),
    model: 'ZC-LP01',
    vendor: 'Moes',
    description: 'Sliding Window Pusher',
    options: [exposes.options.invert_cover()],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
    ],
    meta: {
        tuyaSendCommand: 0x04,
        tuyaDatapoints: [
            [102, 'state', tuya.valueConverterBasic.lookup({CLOSE: tuya.enum(0), OPEN: tuya.enum(1), STOP: tuya.enum(2)})],
            [104, 'position', tuya.valueConverter.coverPosition],
            [114, 'state', tuya.valueConverterBasic.lookup({CLOSE: tuya.enum(0), OPEN: tuya.enum(1), STOP: tuya.enum(2)})],
        ],
    },
    extend: [
        tuya.modernExtend.dpBattery({dp: 4}),
    ],
};

module.exports = definition;
```

ezeghers, 2024-08-07:
> I know it's missing some datapoints but, more importantly, the following is working 100% as expected without any errors for me.
>
> I experimented with the other datapoints but they either didn't have [...]

Thanks for your work. Yes, this works with the basic functions, but for me, the Open and Close where switched around. As mentioned previously, I got a firmware update from Moes to correct the backward Open/Close status'.

acortelyou, 2024-08-07:
Perhaps some of my seemingly non-deterministic results were due to not having upgraded the firmware. I may give that a shot in the future.

I would suggest sticking to `modernExtend` for anyone looking to add the other datapoints as it is more explicit about the `from` and `to` datatypes which was giving me difficulty.

@ezeghers I am curious, is your `invert_cover` setting set to true or false? perhaps the `state` value converter needs to account for that in the same way `position` does.

DavidWAbrahams, 2024-08-08:
Working correctly here. While it initially had the direction reversed, it is now working as expected using the latest external_converter from @acortelyou and `invert_cover: true` for the device in my `config/zigbee2mqtt/config.yaml`

ezeghers, 2024-08-08:
> Perhaps some of my seemingly non-deterministic results were due to not having upgraded the firmware. I may give that a shot in the future.
>
> I would suggest sticking to `modernExtend` for anyone [...]

I don't have an invert cover setting so I assume that means false.

acortelyou, 2024-08-09:
> I don't have an invert cover setting so I assume that means false.

If you're using my version it's an "option" in the code which show up under "Settings (Specific)" in the web-ui.

But yeah if you let it alone that means false.

With different firmware versions and orientations out there having a separate invert option for both position and state is probably for the best.

I'll see if I can get that working and send the PR.

tribakzero, 2024-08-09:
To me it's working great after updating the invert_cover setting too.

razserv2010, 2024-08-14:
I would appreciate your help please.
I bought the MOES Tuya ZigBee Smart Sliding Window Pusher
And I try to add it with z2m but it is not supported.
How did you manage to add it?
[image]

tribakzero, 2024-08-14:
@razserv2010 follow the thread, you need to add a file and a config to your HA in order for it to work.

razserv2010, 2024-08-14:
> @razserv2010 follow the thread, you need to add a file and a config to your HA in order for it to work.

Hi, 
I followed and understood that some file needed to be added, but I did not understand which one and where to add it, if you could give me a guide for this I would be very happy.

tribakzero, 2024-08-14:
@razserv2010 
Copy this to a file: https://github.com/Koenkk/zigbee2mqtt/issues/22844#issuecomment-2270195989 named /homeassistant/zigbee2mqtt/ZC-LP01.js

Then inside /homeassistant/zigbee2mqtt/configuration.yaml add the following to the root and restart.

external_converters:
  - ZC-LP01.js

You may need to set your invert_cover following these steps: https://github.com/Koenkk/zigbee2mqtt/issues/22844#issuecomment-2278771683

That should be it.

razserv2010, 2024-08-14:
> @razserv2010 Copy this to a file: [#22844 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/22844#issuecomment-2270195989) named /homeassistant/zigbee2mqtt/ZC-LP01.js
> 
> Then inside /homeassistant/zigbee2mqtt/configuration.yaml add the following to the root and restart.
> 
> external_converters:
> 
> * ZC-LP01.js
> 
> You may need to set your invert_cover following these steps: [#22844 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/22844#issuecomment-2278771683)
> 
> That should be it.

It doesn't work for me, this is my configuration.yaml:

`homeassistant: true
advanced:
  network_key:
    - 242
    - 193
    - 230
    - 35
    - 134
    - 24
    - 56
    - 23
    - 157
    - 52
    - 153
    - 137
    - 75
    - 253
    - 193
    - 73
  pan_id: 58257
  ext_pan_id:
    - 31
    - 177
    - 51
    - 33
    - 25
    - 165
    - 247
    - 226
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: error
data_path: /share/zigbee2mqtt
external_converters:
  - ZC-LP01.js
mqtt:
  server: mqtt://***:1883
  user: ***
  password: ***
frontend:
  port: 8099
serial:
  adapter: zstack
  port: tcp://slzb-06p7.local:6638
  baudrate: 115200
device_options:
  legacy: false
devices:
  "0xa4c1380354fa5404":
    friendly_name: "0xa4c1380354fa5404"
  "0xa4c13880167d9af7":
    friendly_name: Fingerbot 1
  "0x94b216fffee6943d":
    friendly_name: ZC-LP01
    homeassistant:
      name: ZC-LP01
    optimistic: true`

moshiko2312, 2024-08-19:
Wow thanks

moshiko2312, 2024-08-19:
I fix the manual mode button, need look like this 

[106, 'manual_mode', tuya.valueConverterBasic.lookup({'Disable': tuya.enum(0), 'Enable': tuya.enum(1),})],

tribakzero, 2024-08-19:
@moshiko2312 you mean now you have an extra setting inside the "Settings (specific)" tab to enable/disable the physical buttons?
To me the buttons didn't worked, and by adding that line it didn't showed the setting anywhere but the buttons are enabled once more, so if it was it, then thanks.

moshiko2312, 2024-08-19:
@tribakzero 
[image]

moshiko2312, 2024-08-19:
> @moshiko2312 to me that's only state, position, battery and linkquality 😔

look up I put the code

tribakzero, 2024-08-19:
@moshiko2312 no idea why, but even copy/pasting and deleting and adding the device again still show me the same inputs.
At least the buttons work hehe.

effoc, 2024-08-19:
> @moshiko2312 no idea why, but even copy/pasting and deleting and adding the device again still show me the same inputs. At least the buttons work hehe.

I have to reboot my server and not only HA to make new changes to work

tribakzero, 2024-08-20:
Thanks @effoc that did it. Now with @moshiko2312 version I'm noticing that the feature to reverse the direction isn't working, if you add: `options: [exposes.options.invert_cover()]` to the root of the definition object it does what I suppose `Motor direction` intended.

Remember to reboot the whole thing for changes to take effect, for whatever reason that is.

Snowbars2000, 2024-09-08:
> @razserv2010 Copy this to a file: [#22844 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/22844#issuecomment-2270195989) named /homeassistant/zigbee2mqtt/ZC-LP01.js
>
> Then inside [...]

Thanks for the simple but very good advice. I just wanted to say that I needed to put a minus "-" before the file name and then everything worked.

[image]

ronaldopdc, 2024-10-03:
> },
>     extend: [
>         // A preferred new way of extending functionality.
>     ],
>     options: [exposes.options.invert_cover()],
> };
> 
> module.exports = definition;

@tribakzero, great work, but for, me the feature to reverse the direction isn't working. I put the options: [exposes.options.invert_cover()] as above and it isn't working. Can you help me please?

wajo666, 2024-11-24:
I'm getting this error 

[2024-11-24 13:29:24] error: 	z2m: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device

[2024-11-24 13:29:24] error: 	z2m: Failed to load external converter file 'ZC-LP01.js' (Invalid or unexpected token)

[2024-11-24 13:29:24] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version

[2024-11-24 13:29:24] error: 	zh:ezsp:uart: Error while parsing to NpiFrame 'Error: <-- CRC error: 964e24ab5593499c1ad773d41a9874f5de2b83fc7e1663e3c96c867e|6c86|77f1'

wajo666, 2024-11-24:
> @tribakzero
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     zigbeeModel: [
>         '5rta89nj' // Tubular motors
>     ],
>     fingerprint: [
> // Curtain motors:
>         {modelID: 'TS0601', manufacturerName: '_TZ3210_5rta89nj'},
>     ],
>     model: 'ZC-LP01',
>     vendor: 'Moes',
>     description: 'window pusher',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     onEvent: tuya.onEventSetTime,
>     configure: tuya.configureMagicPacket,
>     exposes: [
>         e.battery(),
>         e.cover_position('position', 'state', ea.STATE_SET,['OPEN', 'CLOSE', 'STOP']).withDescription('Window Control'),
>         e.binary('charge_state', ea.STATE, 'True', 'False'),
>         e.enum('manual_mode', ea.STATE_SET, ['Enable', 'Disable']),
>         e.binary('fault', ea.STATE, 'True', 'False').withDescription('Motor Fault'),
>         e.numeric('countdown', ea.STATE_SET).withValueMin(10).withValueMax(90).withDescription('Mortor timeout').withUnit('s'),
>         e.enum('motor_direction', ea.STATE_SET, ['Left Side', 'Right Side']).withDescription('Install Side Type'),
>         e.enum('mode', ea.STATE_SET, ['Enable', 'Disable']).withDescription('Slow_stop'),
>         e.enum('window_detection', ea.STATE, ['Opened', 'Closed', 'Pending']).withDescription('Window Detection'),
>         
>     ],
>     meta: {
>         // All datapoints go in here
>         tuyaSendCommand: 0x04,
>         tuyaDatapoints: [
>             [4, 'battery', tuya.valueConverter.raw],
>             [102, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'CLOSE': tuya.enum(1), 'STOP': tuya.enum(2)})],            
>             [104, 'position', tuya.valueConverter.coverPosition],
>             [104, 'position', tuya.valueConverter.raw],
>             [105, 'charge_state', tuya.valueConverter.trueFalse1],
>             [106, 'manual_mode', tuya.valueConverterBasic.lookup({'Disable': tuya.enum(0), 'Enable': tuya.enum(1),})],            
>             [108, 'countdown', tuya.valueConverter.raw],
>             [109, 'motor_direction', tuya.valueConverterBasic.lookup({'Left Side': tuya.enum(1), 'Right Side': tuya.enum(0)})],
>             [110, 'mode', tuya.valueConverterBasic.lookup({'Enable': tuya.enum(1), 'Disable': tuya.enum(0)})],
>             [114, 'window_detection', tuya.valueConverterBasic.lookup({Opened: 0, Closed: 1, Pending: 2})],
>             ],
>     },
>     extend: [
>         // A preferred new way of extending functionality.
>     ],
> };
> 
> module.exports = definition;
> ```

im getting this error 

[2024-11-24 13:29:24] error: 	z2m: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device

[2024-11-24 13:29:24] error: 	z2m: Failed to load external converter file 'ZC-LP01.js' (Invalid or unexpected token)

[2024-11-24 13:29:24] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version

[2024-11-24 13:29:24] error: 	zh:ezsp:uart: Error while parsing to NpiFrame 'Error: <-- CRC error: 964e24ab5593499c1ad773d41a9874f5de2b83fc7e1663e3c96c867e|6c86|77f1'

bersaneti, 2024-12-08:
> @tribakzero, great work, but for, me the feature to reverse the direction isn't working. I put the options: [exposes.options.invert_cover()] as above and it isn't working. Can you help me please?

I have the same problem

wajo666, 2024-12-14:
Have we had any stable convertor or still it under investigation?

Mrsash2020, 2024-12-27:
> I know it's missing some datapoints but, more importantly, the following is working 100% as expected without any errors for me.
>
> I experimented with the other datapoints but they either didn't have [...]

Ok for me this works perfectly. I have mine installed on the left. So close should move from left to right.

I do see the invert cover option in specific settings but ticking unticking does not change anything for me.

interestingly using moshiko2312's solution inverts the action so the close sends the window to the left which is actually open.

I think something needs tweaking.

From moshiko2312's config

Battery - Works
State - Works
Position - Not tested enough
State of charge - Unsure what this is - Assume charging via solar???
Manual mode - Not sure what this does
Motor Time Out - Not sure what this does. Motor slips a lot so is this a time out stop if runs for x number of seconds?
Install Side - Does nothing as far as I can tell.
Slow stop - What does this do exactly?
window detection (unfortunately not a binary sensor)

Think I need a little explanation of what these settings are if ok....
thanks

DavidWAbrahams, 2024-12-28:
I believe there are two device firmware versions and the configs above will
work correctly for one or the other, but not both.

I own 2 units bought months apart. Any config above that has the correct
orientation for one of my units will reverse the direction on the other.
Another commenter mentioned fixing this with a private firmware update
pushed through the app from the manufacturer (MOES), but I don't have the
OEM hub and app to do that.

I have tried all the variations of "invert_cover" and "position" without
success.

Mrsash2020, 2024-12-29:
Interesting. I brought mine three weeks ago. 
Using acortelyou's config for external converter the open and close go the correct way while using moshiko2312's config it is reversed by default in both cases.
It is to be noted that neither code options to invert/reverse work in reversing the direction be it the invert option in acortelyou's config in the setting(specific) screen nor moshiko2312's config with motor direction - left side/right side. So dont really think its a firmware thing. As noted for me its config specific acortelyou's vs moshiko2312's.......

This is as per my testing so not sure what everyone is using and how its working for them. Suggest everyone test if possible, state the side its installed and what they experienced. 

Personally I think both codes need to be merged and the non working stuff be removed also fixing the direction, but not skilled in it myself.
 
I contacted Moes to get issues sorted with slipping of the wheel I am experiencing and also to get the firmware being discussed(yet to hear from them). I think I installed mine at 40% height so almost in the middle. Noticed even thought its pushing fairly on the inner window it still slips. Placing my finger on the body of the motor when I run it makes it work much better(obviously). Think I need to install it on the bottom which would reduce the bowing of the moving pane(Noticed as you go towards the centre of the window Panel there is more bowing. Hopefully this fixes my issue.

wajo666, 2024-12-29:
Hi all,

I merged both convertors from top of topic :) still I have had some unsupported fields we can remove it and update convertors
My recom: in first it is needed take OTA from offical Tuya app. from factory is coming with version 1.0.10 and their send new version 1.0.13 with fixed correct state for motor_direction.

[image]
[image]

In picture you can see fields are still null we can remove it.

Notice:
fixed_window_sash -> swap button logic :D if you click left window go to right :D 
If someone has more time can cleanup code.

countdown it is for setup when motor will stop. in offical app you are able to change time or make calibration in this convertor you are able to make calibration when take slide to start from middle :D 

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: [
        '5rta89nj' // Tubular motors
    ],
    fingerprint: [
// Curtain motors:
        {modelID: 'TS0601', manufacturerName: '_TZ3210_5rta89nj'},
    ],
    model: 'ZC-LP01',
    vendor: 'Moes',
    description: 'window pusher',
    options: [exposes.options.invert_cover()],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(),
        e.cover_position('position', 'state', ea.STATE_SET,['OPEN', 'CLOSE', 'STOP']).withDescription('Window Control'),
        e.binary('charge_state', ea.STATE, 'True', 'False'),
        e.enum('manual_mode', ea.STATE_SET, ['Enable', 'Disable']),
        e.binary('fault', ea.STATE, 'True', 'False').withDescription('Motor Fault'),
        e.numeric('countdown', ea.STATE_SET).withValueMin(10).withValueMax(90).withDescription('Mortor timeout').withUnit('s'),
        e.enum('motor_direction', ea.STATE_SET, ['Left Side', 'Right Side']).withDescription('Install Side Type'),
        e.enum('mode', ea.STATE_SET, ['Enable', 'Disable']).withDescription('Slow_stop'),
        e.numeric('solar_energy_current', ea.STATE).withValueMin(0).withValueMax(99999).withDescription('Solar Energy Current'),
        e.enum('fixed_window_sash', ea.STATE_SET, ['True', 'False']),
//        e.enum('window_detection', ea.STATE, ['Opened', 'Closed', 'Pending']).withDescription('Window Detection'),

    ],
    meta: {
        // All datapoints go in here
        tuyaSendCommand: 0x04,
        tuyaDatapoints: [
            [4, 'battery', tuya.valueConverter.raw],
            [102, 'state', tuya.valueConverterBasic.lookup({CLOSE: tuya.enum(0), OPEN: tuya.enum(1), STOP: tuya.enum(2)})],
            [104, 'position', tuya.valueConverter.coverPosition],
            [104, 'position', tuya.valueConverter.raw],
            [105, 'charge_state', tuya.valueConverter.trueFalse1],
            [106, 'manual_mode', tuya.valueConverterBasic.lookup({'Enable': tuya.enum(0), 'Disable': tuya.enum(1),})],
            [108, 'countdown', tuya.valueConverter.raw],
            [109, 'motor_direction', tuya.valueConverterBasic.lookup({'Left Side': tuya.enum(1), 'Right Side': tuya.enum(0)})],
            [110, 'mode', tuya.valueConverterBasic.lookup({'Enable': tuya.enum(1), 'Disable': tuya.enum(0)})],
            [111, 'solar_energy_current', tuya.valueConverter.raw],
            [112, 'fixed_window_sash', tuya.valueConverterBasic.lookup({'True': tuya.enum(0), 'False': tuya.enum(1)})],
  //          [114, 'window_detection', tuya.valueConverterBasic.lookup({Opened: 1, Closed: 0, Pending: 2})],
            [114, 'state', tuya.valueConverterBasic.lookup({CLOSE: tuya.enum(0), OPEN: tuya.enum(1), STOP: tuya.enum(2)})],
            ],
    },
    extend: [
        // A preferred new way of extending functionality.
      tuya.modernExtend.dpBattery({dp: 4}),
    ],
};

module.exports = definition;`

Mrsash2020, 2024-12-29:
I dont have a moes or any other tuya gateway so not sure where to check in Home assistant which firmware I have installed. 
I can buy a cheap one if someone wants to recommend on Ali Ex. 
Looks like it would be a good investment if I have to request adding devices in the future. In Home assistant I am using SLZB-06.

Couple of questions.
What firmware are you on?
Does Motor Direction change actually reverses the direction it opens or closes?
Did not get the Fixed Sash thing at all?
Countdown - ok so it stops motor after specific seconds?

wajo666, 2024-12-29:
I have had 1.0.13 for my both sliders
I bought first cheapest  tuya gateway 
Yes countdown count time and stop it

fixed window ssh only changed behaviour for control button in tuya app I didn't find it

Mrsash2020, 2024-12-29:
sorry still dont get what that fixed sash is or supposed to be doing. 
Ok I am saying try and see if the motor direction left or right changes it at all and then change it back in tuya and home assistant for testing purposes please.

thanks

fratichelo, 2025-01-05:
Did anybody else have issues with the latest Z2M or HA Core updates?

ezeghers, 2025-01-05:
> Did anybody else have issues with the latest Z2M or HA Core updates?

Yes. All of the entities disappear with v2 of z2m.

jajaP42, 2025-01-05:
Yes, now my Support status is Not supported. Is there any way to fix it?

jajaP42, 2025-01-05:
@k7franklin
Oh thank you very much it's back!

moshiko2312, 2025-01-05:
I’m running all ok to me

[41 more replies]

closed 2025-05-19 by Koenkk