Koenkk/zigbee2mqtt #22845: Moes ZC-HM _TZE200_rjxqso4a - no data

matze20211, 2024-05-31:
### What happened?

The Device is 24 h online, but no Carbon monoxide data are shown (N/A), only Battery status and link

### What did you expect to happen?

CO discover
CO ppm
self test result

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

PI

matze20211, 2024-06-12:
after update to 1.38, same behaviour

all3kcis, 2024-09-10:
Same issue here, only N/A values reported

NickRemizov, 2024-11-08:
The same problem at z2m 1.41.0
zigbee-herdsman-converters version 20.38.0
On interview reports:
Device is only compliant to revision '21' of the ZigBee specification (current revision: 23).

maciejrozanski, 2024-11-25:
Same issue here

MqlBql, 2025-01-18:
Anyone found a fix for this ?

Chartreusito, 2025-01-18:
I had problems with this device not reporting anything but my underlying problem what that i needed to split my network as i reached 160+ devices. As soon as i removed two dozens of devices from my network, it started reporting correctly.

Chartreusito, 2025-01-19:
> Which version of z2m are you using? On 2.0.0 problem still exists (20 devices in the network)

Device is connected to a SLZB-06, here are my versions :

Zigbee2MQTT version
    [2.0.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator type
    zStack3x0

Coordinator revision
    20240710

Coordinator IEEE Address
    0x00124b0029e8a671

Frontend version
    0.9.4

zigbee-herdsman-converters version
    21.12.0

zigbee-herdsman version
    3.2.1

maciejrozanski, 2025-01-20:
After reset of the device (remove batteries and put them again), started working correctly. 

[image]

Upgrade z2m to 2.0.0 may be needed first.

closed 2025-03-30 by github-actions[bot]