Koenkk/zigbee2mqtt #22847: Inconsistent State and Position Reporting for Smartwings WM25L-Z Roller Shade

pruvi9019, 2024-05-31:
### What happened?

When controlling the Smartwings WM25L-Z roller shade manually via the device's remote, Zigbee2MQTT and Home Assistant report conflicting states and positions. Reversing the motor direction aligns the reports with the actual position but inversely affects the remote control commands, leading to operational inconsistencies.

### What did you expect to happen?

I expected the state and position in Zigbee2MQTT to consistently reflect the actual physical state of the roller shade, aligning correctly with manual controls and Home Assistant without needing to invert motor directions, which inversely affects the usability of the remote control.

### How to reproduce it (minimal and precise)

1. Without Reversing Motor Direction:

- Manually close the blinds using the remote; observe the reported state in Zigbee2MQTT as OPEN and position as 100%, while Home Assistant indicates OPEN at 0%.

- Manually open the blinds using the remote; observe the reported state in Zigbee2MQTT as CLOSE and position as 0%, while Home Assistant shows CLOSED at 100%.

2. With Reversed Motor Direction:

- Use the remote to reverse the motor direction.

- Repeat the above steps and notice that while the reported states now align with the physical state, the remote control commands work inversely.

### Zigbee2MQTT version

1.37.1 (commit: cda867a3)

### Adapter firmware version

20240316

### Adapter

SLZB-06p7

### Setup

Raspberry Pi 5

### Debug log

[log1.log](https://github.com/user-attachments/files/15510573/log1.log)
[log.log](https://github.com/user-attachments/files/15510574/log.log)

pruvi9019, 2024-06-01:
Below are the Datapoint I found for the device using Tuya

{"1":"Control",
"2":"Curtain position setting",
"3":"Current curtain position",
"4":"Mode",
"5":"Motor Direction",
"11":"Situation_set",
"12":"Fault",
"16":"Border",
"19":"Best Position",
"20":"Click Control"}

ThrashinVictim, 2024-06-13:
Same issue as I am having, but I also have the shades exposed to Alexa via HA and they are reversed as well. 

There is a way to reverse the motor for Alexa by programming the controller using key combinations, but that changes the reporting in HA. 

For example:
HA shows blinds closed, Alexa shows them as opened, and vice versa.

LordSushiPhoenix, 2024-09-05:
I just installed a Smartwings zigbee roller shade and had the same issue. With the motor in the default position (so the up button on the remote opens the shade), the shade would report CLOSED to Home Assistant. 

I found [this external converter](https://www.reddit.com/r/Zigbee2MQTT/comments/1d1b9kq/comment/l5xb6ds/) that corrected the issue so that the remote works in the correct direction and Home Assistant reports OPEN 100% when the shade is fully open.

note that the end of the file in the reddit comment has a syntax error at the end, here is the correct syntax:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const utils = require('zigbee-herdsman-converters/lib/utils');

backwards_cover_state = {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = {'open': 'downClose', 'close': 'upOpen', 'stop': 'stop', 'on': 'downClose', 'off': 'upOpen'};
            value = value.toLowerCase();
            utils.validateValue(value, Object.keys(lookup));
            await entity.command('closuresWindowCovering', lookup[value], {}, utils.getOptions(meta.mapped, entity));
        },
    };

module.exports = [
    {
        zigbeeModel: ['WM25/L-Z'],
        model: 'WM25L-Z',
        vendor: 'Smartwings',
        description: 'Roller shade',
        fromZigbee: [fz.cover_position_tilt, fz.battery],
        toZigbee: [backwards_cover_state, tz.cover_position_tilt],
        meta: {battery: {dontDividePercentage: true}, coverInverted: false},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.currentPositionLiftPercentage(endpoint);
        },
        exposes: [e.cover_position(), e.battery()],
    }
];
```

ThrashinVictim, 2024-09-07:
> I just installed a Smartwings zigbee roller shade and had the same issue. With the motor in the default position (so the up button on the remote opens the shade), the shade would report CLOSED to [...]

Tried that but didn't work for me.

LordSushiPhoenix, 2024-09-07:
What side of the shade is your charge port on? I'm wondering if that's related to the issue since the reported behaviour seems really inconsistent between owners. Mine is on the right.
[image]
[image]

ThrashinVictim, 2024-09-07:
> What side of the shade is your charge port on? I'm wondering if that's related to the issue since the reported behaviour seems really inconsistent between owners. Mine is on the right. [...]

Mine is also on right. Are you using Alexa as well?

LordSushiPhoenix, 2024-09-07:
No, I have it connected to Home Assistant through zigbee2mqtt only.

turbotiger74, 2024-11-10:
> I just installed a Smartwings zigbee roller shade and had the same issue. With the motor in the default position (so the up button on the remote opens the shade), the shade would report CLOSED to [...]

Thanks LordSushiPhoenix,
This ext_converter.js worked for me.  
There's one extra step involved though since I am using the zigbee2mqtt HA addon.
create a new folder called data in the zigbee2mqtt directory and create the ext_converter.js in there. 
`sudo /usr/share/hassio/homeassistant/zigbee2mqtt/data`
next, in 
zigbee2mqtt settings, click External converters and add the ext_converter.js there

turbotiger74, 2024-11-10:
> I just installed a Smartwings zigbee roller shade and had the same issue. With the motor in the default position (so the up button on the remote opens the shade), the shade would report CLOSED to [...]

Thanks LordSushiPhoenix,
This worked great for me.  I had to make sure I copied everything correctly, but once I did and added it, it worked great. I didn't have to invert the shades either.  I have an older smartwings roller shades bought in 12/2022, both left and right motor models.

ThrashinVictim, 2024-11-10:
That ==

> Thanks LordSushiPhoenix, This worked great for me. I had to make sure I copied everything correctly, but once I did and added it, it worked great. I didn't have to invert the shades either. I have [...]

Do you Alexa to control your shades?

turbotiger74, 2024-11-11:
No I only use HA to control them.

Korri, 2025-01-27:
Hey folks, I reached out to SmartWings and they did mention that:

> Sorry for the inconvenience. We’ve had similar feedback from other customers regarding the Zigbee status display issue on HA, and we’ve already resolved the issue on ZHA. However, due to platform compatibility, the issue has not been fully resolved on Z2M yet.

So it got me curious, and it seems like the main difference is that ZHA has [logic to invert the direction commands](https://github.com/zigpy/zha-device-handlers/blob/d3d43b17294bf9e2e783c3cfb299824d22ff92c2/zhaquirks/smartwings/wm25lz.py#L50-L53).

Looks like that's what the external converter is doing too, so it should probably be added to [the device code](https://github.com/Koenkk/zigbee-herdsman-converters/blob/46d86e8c7bb2c3f930039b3ae856fce7b6443830/src/devices/smartwings.ts#L13).

Korri, 2025-01-27:
I opened a pull request to merge this into the main code: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8681

closed 2025-01-27 by Koenkk

ThrashinVictim, 2025-02-02:
@Korri I am still having the issue I described above when using Alexa. Do you use Alexa with these blinds via HA?

> Hey folks, I reached out to SmartWings and they did mention that:
>
>
> So it got me curious, and it seems like the main difference is that ZHA has [logic to invert the direction [...]