Koenkk/zigbee2mqtt #22856: TuYa CK-BL702-AL-01 Does not permit joining as a router

stm2roi3, 2024-05-31:
### What happened?

I have installed 8 GU10 Tuya light bulbs (TuYa CK-BL702-AL-01). None of them is able to connect to each other when permitting the join of other devices via zigbee2mqtt. When I enable all Zigbee devices to permit joins those Tuya light bulbs connect to a IKEA Tradfri Driver 10W.

### What did you expect to happen?

When I click on “permit join” on the first connected Tuya Light bulb I want to a second Tuya light bulb to pair to the first one.

### How to reproduce it (minimal and precise)

- Pair first Tuya light bulb
- Click on “permit join” in order to bring first Tuya light bulb into pairing mode
- power second Tuya light bulb and bring it to pairing mode
- See no connection between first and second light bulb

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20210120

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Home Assistant 2024.5.5 on RPi4-64

janfuhlrott, 2024-06-19:
Same Problem here. 
HA OS on MacMini.

ccervigon, 2024-07-04:
Same problem here with home assistant. They are not able to connect to each other. Also they are not able to connect to another router only to sonoff.

rmxs, 2024-07-05:
Same issue

Koenkk, 2024-07-06:
This is probably a bug in the firmware of the bulb, not something that can be solved from z2m.

closed 2024-12-28 by github-actions[bot]