Koenkk/zigbee2mqtt #22863: NCP FATAL ERROR - reason ERROR_WRONG_DIRECTION

FredC27, 2024-06-01:
### What happened?

Hi,
I'm running Ember.

I noticed that with all my NOUS A1Z TS011F (_TZ3000_ksw8qtmt / _TZ3000_2putqrmw), i get those errors comming regurarly : 
`error 2024-06-01 07:40:24ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=58 Len=26].`
`error 2024-06-01 07:40:24!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!`
`error 2024-06-01 07:40:24ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=58 Len=26].`
`error 2024-06-01 07:40:24Default response to 0xa4c138f7e3ee084c failed`

Generally this does not affect operation, but Z2M can sometimes freeze, so as this is the only errors i get, i suspect it has something to do with it.

These messages only appear for this equipments

I've made the most recently firmware upgrade to 192 version for all.

I'm not an expert at all but i suspect that this has to do with GreenPower implementation in those plugs.

### What did you expect to happen?

Not to have those errors messages. What is the  "..;ATTEMPTING RESET..." for ? reset of what ? 

### Zigbee2MQTT version

    1.37.1 commit: cda867a

### Adapter firmware version

7.4.3 [GA]

### Adapter

SONOFF E v2 

### Setup

Add on - Debian Buster

Patricen1, 2024-06-01:
Same issue there with FW 7.4.2

Nerivec, 2024-06-01:
`ERROR_WRONG_DIRECTION` means a `command` was sent to Z2M instead of a `response`.
This can happen when interferences screw up a message in transit, but since you mention a specific device, it could be that device has a bad implementation in the firmware somewhere.
As with most frame errors, it results in a reset (reboot) of the adapter (NCP) to resume proper operation.

How often do you get these? That's pretty bad for the network since it's forced to restart every time.

FredC27, 2024-06-02:
Hi Nerivec,
Thanks for you answer. That's what i suspected about the restart....
I checked for the elapsed time between 2 ERRORS : 

error 2024-06-01 18:11:57!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 18:27:42!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 18:41:20!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 18:53:20!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 18:56:39!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 18:56:52!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 19:00:50!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 19:23:25!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 19:35:34!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 19:40:03!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 19:43:54!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 20:02:33!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 20:09:04!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 20:10:37!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 20:14:17!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 20:14:31!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 20:35:40!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 20:44:18!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-01 20:45:55!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-02 08:16:41!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-02 08:19:03!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-02 08:24:50!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
error 2024-06-02 08:24:57!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!

Pretty often, isn't it ? for each error above, the corresponding addresses was : 

error 2024-06-01 17:53:08Default response to 0xa4c138943a6da433 failed
error 2024-06-01 18:11:57Default response to 0xa4c138968479c7ed failed
error 2024-06-01 18:41:20Default response to 0xa4c138f7e3ee084c failed
error 2024-06-01 18:53:20Default response to 0xa4c138ab54473daf failed
error 2024-06-01 18:56:39Default response to 0xa4c138645dc17d38 failed
error 2024-06-01 18:56:52Default response to 0xa4c138468b4eb128 failed
error 2024-06-01 19:00:50Default response to 0xa4c138f7e3ee084c failed
error 2024-06-01 19:35:34Default response to 0xa4c138fe8b343f75 failed
error 2024-06-01 20:02:33Default response to 0xa4c138943a6da433 failed
error 2024-06-01 20:09:04Default response to 0xa4c138ab54473daf failed
error 2024-06-01 20:10:37Default response to 0xa4c138645dc17d38 failed
error 2024-06-01 20:14:17Default response to 0xa4c138f7e3ee084c failed
error 2024-06-01 20:14:31Default response to 0xa4c138943a6da433 failed
error 2024-06-01 20:35:40Default response to 0xa4c138943a6da433 failed
error 2024-06-01 20:45:55Default response to 0xa4c138ab54473daf failed
error 2024-06-02 08:16:41Default response to 0xa4c138f7e3ee084c failed
error 2024-06-02 08:19:03Default response to 0xa4c138ab54473daf failed
error 2024-06-02 08:24:50Default response to 0xa4c138468b4eb128 failed
error 2024-06-02 08:24:57Default response to 0xa4c138943a6da433 failed

I have 15 of those plugs. 
I have tried many things, unplug, reinsertion with erase, etc...
Not all of those cause a reset as we can read (7).
I checked the firmware version, all of the 15 are in v192
I checked if the link quality was related.
i checked the parameters between a plug not causing any trouble and the others.
i checked if those errors occurs when i make some actions on them.

thanks

FredC27, 2024-06-02:
What I'm noticing as I write this is that the errors stopped between 8.45pm and 8.16am. That's a very long time. It's as if a piece of equipment had stopped causing problems. I can't work out which.

Nerivec, 2024-06-02:
Check any of these:
- Plugs in question are no longer used during that "no error time"
- An automation no longer runs during that "no error time" (seems pretty clear something happens during day-time)
- A device that can potentially affect Zigbee 2.4GHz is off during that "no error time"
- Plugs in question are in the same general physical area

@Patricen1 Are you having the issue with the same device model? Any chance you can confirm that you also see periods without errors?

Patricen1, 2024-06-02:
[image]

Patricen1, 2024-06-02:
Hello, this is what appears when toggling the plug, as soon as it is not in the room where it had been paired but close to other router devices (less than 5m).

Error 2024-06-02 16:57:32Publish 'set' 'state' to 'smart power plug (ZB02)' failed: 'Error: ZCL command 0xa4c138e162960df8/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":225})'
Error 2024-06-02 16:57:36Publish 'set' 'state' to 'smart power plug (ZB02)' failed: 'Error: ZCL command 0xa4c138e162960df8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":226})'
Error 2024-06-02 16:57:41Publish 'set' 'state' to 'smart power plug (ZB02)' failed: 'Error: ZCL command 0xa4c138e162960df8/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":229})'
Error 2024-06-02 16:57:46Publish 'set' 'state' to 'smart power plug (ZB02)' failed: 'Error: ZCL command 0xa4c138e162960df8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":230})'
Error 2024-06-02 16:57:51Publish 'set' 'state' to 'smart power plug (ZB02)' failed: 'Error: ZCL command 0xa4c138e162960df8/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":233})'
Error 2024-06-02 16:57:56Publish 'set' 'state' to 'smart power plug (ZB02)' failed: 'Error: ZCL command 0xa4c138e162960df8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":235})'
Error 2024-06-02 16:58:01Publish 'set' 'state' to 'smart power plug (ZB02)' failed: 'Error: ZCL command 0xa4c138e162960df8/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":236})'

Patricen1, 2024-06-02:
Other issue : those plugs needs to be very close to coordinator to be paired, and as soon as the plug is not close to ZB dongle, it looks like the network does not rebuild routes to use routers.

FredC27, 2024-06-02:
> * n potentially affect Zigbee 2.4GHz is off during that "no error time"
> 
> * Plugs in question are in the same general physical area

FredC27, 2024-06-02:
Thanks Nerivec,

i think that i will disconnect 50% of the plugs, and check what happens and again and again. 

Those plugs are in v3.0 of zigbee.

Well i will not have the time for more than 10 days to try anything rigth now. 

If anyone has a recommandation for a perfect plug (small factor if possible), please tell me. It has to be available on the european / french market too... i'm just fed up, i just want them to work ! :)

Patricen1, 2024-06-06:
Some of the plugs we are discussing about are Ok, some others with low RF capability, looks like the production process is not stable. Also interested in getting perfect zigbee small form factor plugs as @FredC27 !

FredC27, 2024-06-06:
Hi Patricen1,
After all my tests it appears that nearly all my NOUS A1Z plugs behave the same maner. No too much people seems to be disturb like me. Perhaps it is due to my config. But i can't find anything path to resolve it. Recently i put back Z2M in EZSP, and thoses errors went away. I still a few other errors, from times to times this leads (1 per day) to a restart. I spend too much time to try to resolve....

Nerivec, 2024-06-06:
@FredC27 `ezsp` doesn't have checks in place for that kind of error (not implemented), so they would go unnoticed, and potentially (likely) lead to other issues.

Nerivec, 2024-06-17:
I guess there is a bad firmware out there, but since it seems they didn't change versions (you said they were all showing v192), impossible to tell which, or even confirm this theory...

bdeluca, 2024-08-17:
I have the same issue. And am happy to provide any details to you. I am planning to look at the wifi interference once some parts arrive.

FredC27, 2024-08-21:
hi, 
i'm not so sure but it seems that all my problems were linked to the CPU charge ! I changed it for a much faster one.
I'm in the process of testing. I'll have to be careful to confirm.

francois-le-ko4la, 2024-12-08:
Hello,

- Docker environment
- Zigbee2MQTT version: [1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0) commit: [861cba6](https://github.com/Koenkk/zigbee2mqtt/commit/861cba6)
- Coordinator type: EmberZNet / ZBDONGLE-E
- Coordinator revision : 7.4.4 [GA]

I do my update every Saturday @ 6am automatically with the stable branch.
I know... for home automation, enabling automatic updates is a bit risky, but it turns out I’ve had very few issues with Z2M and the latest ZBDONGLE-E firmware.

That said, I noticed yesterday the same error messages that I didn’t encounter with the previous version of Z2M.
I have the same NOUS plug and added a Tuya USB to extend my network on Friday, which successfully rebuilt it.
During this process, 5 devices moved to the NOUS plug, and I experienced the same type of error messages (ERROR_WRONG_DIRECTION/NCP....) when the update occurs on saturday.
I read this thread and re-paired the devices linked with NOUS plug with another router, and the errors disappeared.

My hypothesis is that the NOUS plug is good for extending a Zigbee network (according to my LQIs), but it should be avoided for direct connections with end devices. Unfortunately, I dont see a methodology to be sure that this plug will be avoided because, by design, our end device can move.

Thanks for this thread. It saved me a lot of time!

[image]

Nerivec, 2024-12-08:
@francois-le-ko4la If by Tuya USB you mean this device: https://www.zigbee2mqtt.io/devices/TS0207_repeater.html beware of the warning attached on the page. This device is known to break down networks because it was designed to handle only Tuya traffic, it basically discards everything else...
If you want to extend your network with a dedicated router, look into using an adapter flashed with a router firmware, it should be a lot safer.

closed 2024-12-10 by Nerivec

Barry40, 2025-01-06:
Same problem in here, wrong direction...
Same adapter, same firmware but with mous devices... Can we do something about it?

Nerivec, 2025-01-06:
This error no longer causes a reboot (disabled it some time ago), but it's still logged, since obviously it should not happen, and we really don't know what kind of other troubles these devices may be causing in the network.
Previous implementation (ezsp) did not check for it at all, hence, no message.