Koenkk/zigbee2mqtt #22870: ZG-204ZM - Wrong options to set

Koterak, 2024-06-01:
### What happened?

In 1.38.0 you introduced new options to set "PiR, PiR+Radar, Radar". But options are probably wrong: there is "Dadar" instead of "Radar" and there is error in zigbee2mqtt that it doesn't know this option

[image]

### What did you expect to happen?

Change "Dadar" to "Radar"

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20230922

### Adapter

EZSP-06

### Setup

Add-on on HASSIO

truongbber, 2024-06-01:
I also got the same error

truongbber, 2024-06-01:
> I also got the same error

[image]

mortanol, 2024-06-01:
Same error here. I was hoping that this option would help the radar to work. Currently, it's not. When I'm static, presence is set to none after fading time

idogrf, 2024-06-02:
Same issue

hfagelnour, 2024-06-03:
UP

Angus774, 2024-06-03:
Same error

stenly30, 2024-06-03:
Same error

ArkadiuszNiemiec, 2024-06-04:
This is already fixed in the dev branch. If you are using docker you can switch the tag to latest-dev. It's still getting stuck with presence detected, so it's unusable for me.

ArkadiuszNiemiec, 2024-06-04:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/7590

hfagelnour, 2024-06-05:
> This is already fixed in the dev branch. If you are using docker you can switch the tag to latest-dev. It's still getting stuck with presence detected, so it's unusable for me.

I have 6 of these, I am experiencing the same on 2 units of all 6 put in the same room in the same exact environment. the other 4 works flawlessly with the stable and now even better with the dev version (except for the cosmetic "Dadar" thing :) ). I guess what you are experiencing is a hardware issue.

Koenkk, 2024-06-05:
Fixed in the dev branch (including the typo)

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-06-05 by Koenkk