Koenkk/zigbee2mqtt #22871: Localize Icon - Breaks EMQX

senna1992, 2024-06-01:
### What happened?

Using the button "localize icon" unter "tools" generates way too long paths for your icons. 
This crashed my MQTT broker as the "bridge status" message containing these paths.

### What did you expect to happen?

Dont include paths in messages but rather in the logs or way shorter path lenght. 
example snippet of the path used inside of configuration.yaml
`    icon: >-
      data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAgAAAAIACAYAAAD0eNT6AAAABGdBTUEAALGPC/xhBQAAACBjSFJNAAB6JgAAgIQAAPoAAACA6AAAdTAAAOpgAAA6mAAAF3CculE8AAAABmJLR0QA/wD/AP+gvaeTAAAACXBIWXMAAAsSAAALEgHS3`

### How to reproduce it (minimal and precise)

open z2m, go to tools, press loclaize icon

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.3.2.0 build 212

### Adapter

EZSP

### Setup

HA yellow addon, with emqx as broker

pontuslandin89, 2024-06-13:
Same thing happened to me. The function removed many of the friendly names and sometimes did not load my device list at all. 

After going back to an old backup of my device list, everything works again. 

Also, the button "localize-icons is on my iphone 12 mini and chrome mostly not visible in UI. 

I was using my Sonoff-E dongle at the time.

KajbaM, 2024-07-30:
Well this thing apparently happened to me accidentally as I have pressed a button under tools that had no text (for some reason there is one button with no label in settings/tools tab). And I have no idea who thought this is a good idea or how this passed any serious quality control.

What am I talking about? Take another look at those "paths" for icons. Those are not paths, those are raw data of the icons in png base64 format - this is equivalent to opening png image in text editor and copy pasting the content into your yaml configuration file. This is just unacceptable at so many levels...

closed 2024-12-28 by github-actions[bot]