Koenkk/zigbee2mqtt #22872: Aqara ZNJLBL01LM - Position, State Closed and Opened isn't updated

H1ghSyst3m, 2024-06-01:
### What happened?

I newly installed 2 of my new Aqara Roller Driver E1. After Pairing I also updated the Firmware of the 2 Roller Drivers.

in Zigbee2MQTT, normally everything works good. 

Expect, the Position and the open/close state isn't being updated in Home Assistant

### What did you expect to happen?

It should automatically update the position and state after a change or action.

### How to reproduce it (minimal and precise)

Pair the Drivers and set them in Zigbee2MQTT up.

Go to a Dashboard and create a card. Now open or close the shutters. The shutters work, but the state and position stays the same

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus 

### Setup

Add-on on HomeAssistant OS direct

H1ghSyst3m, 2024-06-01:
Extra Information, the Position isn't being updated until I go to Zigbee2MQTT and refresh the position manually

samvel-galstyan, 2024-06-01:
same problem here. I downgraded the version of zigbee2mqtt to 1.35.1-1 and it solved my problem. Hope it will be fixed soon.

LaurentChardin, 2024-06-01:
I wonder if this is related to #22852  : do you have anything in your logs ? It looks like the integration is ok (control shutter works) but something is wrong for some events.

H1ghSyst3m, 2024-06-01:
I also upgraded the firmware of my sonoff usb now to see if this would change something, but still the same problem.

In the logs, I found this:
```
[2024-06-01 16:17:08] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"htsta-1"}'
[2024-06-01 16:17:08] info: 	z2m: Successfully changed options
[2024-06-01 16:17:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"htsta-1"}'
[2024-06-01 16:17:17] info: 	z2m: Successfully changed options
[2024-06-01 16:17:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"htsta-4"}'
[2024-06-01 16:22:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RolloLinks', payload '{"battery":100,"charging_status":false,"device_temperature":null,"linkquality":134,"motor_speed":"medium","motor_state":null,"position":100,"running":null,"state":"STOP","update":{"installed_version":3871,"latest_version":3870,"state":"idle"},"update_available":null}'
[2024-06-01 16:22:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RolloRechts', payload '{"battery":100,"charging_status":true,"device_temperature":null,"linkquality":145,"motor_speed":"medium","motor_state":null,"position":100,"running":null,"state":"STOP","update":{"installed_version":3871,"latest_version":3870,"state":"idle"},"update_available":null}'
```

The weird thing is, it says here too, that the position is 100% even though I stopped it in the middle

Xornop, 2024-06-05:
I'm seeing the same issue on my end. A while back there was a [similar topic](https://community.home-assistant.io/t/aqara-blind-motor-znjlbl01lm-not-update-position-unless-manually-update-position-in-z2m/488179) on the forums, where someone updated the firmware of their stick, but I can't reproduce this as a solution either.
I am seeing some error in the log though:
```
[2024-06-05 21:10:47] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom blinds', payload '{"battery":42,"charging_status":null,"device_temperature":null,"linkquality":232,"motor_speed":"high","motor_state":null,"position":90,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
[2024-06-05 21:10:49] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom blinds', payload '{"battery":42,"charging_status":null,"device_temperature":null,"linkquality":255,"motor_speed":"high","motor_state":null,"position":90,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
[2024-06-05 21:10:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Upstairs washer', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":296.14,"indicator_mode":"off","linkquality":96,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":232}'
[2024-06-05 21:10:51] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Upstairs washer', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":296.14,"indicator_mode":"off","linkquality":208,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":232}'
[2024-06-05 21:10:54] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Downstairs washer', payload '{"child_lock":"UNLOCK","countdown":null,"current":0,"energy":566.35,"indicator_mode":"off","linkquality":248,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":233}'
[2024-06-05 21:10:54] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Downstairs washer', payload '{"child_lock":"UNLOCK","countdown":null,"current":0,"energy":566.35,"indicator_mode":"off","linkquality":255,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":233}'
[2024-06-05 21:10:55] error: 	z2m: Publish 'set' 'position' to 'Bedroom blinds' failed: 'Error: ZCL command 0x54ef441000b58f2b/1 genAnalogOutput.write({"presentValue":90}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (92))'
[2024-06-05 21:10:55] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'position' to 'Bedroom blinds' failed: 'Error: ZCL command 0x54ef441000b58f2b/1 genAnalogOutput.write({\"presentValue\":90}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (92))'","meta":{"friendly_name":"Bedroom blinds"},"type":"zigbee_publish_error"}'
[2024-06-05 21:10:55] error: 	z2m: Publish 'set' 'position' to 'Bedroom blinds' failed: 'Error: ZCL command 0x54ef441000b58f2b/1 genAnalogOutput.write({"presentValue":90}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (91))'
[2024-06-05 21:10:55] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'position' to 'Bedroom blinds' failed: 'Error: ZCL command 0x54ef441000b58f2b/1 genAnalogOutput.write({\"presentValue\":90}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (91))'","meta":{"friendly_name":"Bedroom blinds"},"type":"zigbee_publish_error"}'
[2024-06-05 21:10:55] error: 	z2m: Publish 'set' 'position' to 'Bedroom blinds' failed: 'Error: ZCL command 0x54ef441000b58f2b/1 genAnalogOutput.write({"presentValue":90}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (90))'
[2024-06-05 21:10:55] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'position' to 'Bedroom blinds' failed: 'Error: ZCL command 0x54ef441000b58f2b/1 genAnalogOutput.write({\"presentValue\":90}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transaction [... cut ...]
[2024-06-05 21:11:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hall motion sensor', payload '{"battery":67.5,"battery_low":false,"linkquality":255,"occupancy":false,"tamper":false,"voltage":2800}'
[2024-06-05 21:11:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom blinds', payload '{"battery":42,"charging_status":null,"device_temperature":null,"linkquality":255,"motor_speed":"high","motor_state":null,"position":90,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
```

H1ghSyst3m, 2024-06-16:
I have 2 Shutters and now suddenly one of them seems working with position updates...
Don't know why and since when. The other one still doesn't work.
Both of them have still the same firmware

lskeet, 2024-10-16:
This is likely related to #23557 to which a fix has been integrated so give it a try

Xornop, 2024-11-14:
The blinds now do report opening/closing for me, and their position when stopped (either manually or after reaching their destination). Should the position not be live-updated?

H1ghSyst3m, 2024-11-14:
> The blinds now do report opening/closing for me, and their position when stopped (either manually or after reaching their destination). Should the position not be live-updated?

As I know it should only update after an action is started or stopped.

So a live view shouldn't be there

H1ghSyst3m, 2024-11-14:
Right now my roller drivers work fine now. And also all new Drivers I installed work properly work now.

So I'm closing the ticket

closed 2024-11-14 by H1ghSyst3m