Koenkk/zigbee2mqtt #22873: OTA Ember error

Mar1usW3, 2024-06-01:
### What happened?

If i try to update a device via OTA i get the following errors. I use a HA Yellow with Ember driver

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.4.2

### Adapter

HA Yellow - ember

### Setup

HA Yellow

### Debug log

```
[2024-06-01 13:49:50] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/Kinderzimmer Fensterbank', payload '{"brightness":62,"color_mode":"color_temp","color_temp":454,"color_temp_startup":65535,"last_seen":"2024-06-01T11:49:50.514Z","linkquality":192,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":16786948,"latest_version":16787456,"progress":57.22,"remaining":139,"state":"updating"},"update_available":null}'
[2024-06-01 13:49:50] debug: 	zhc:ota:common: Request offsets: fileOffset=155094 pageOffset=0                 dataSize=64
[2024-06-01 13:49:50] debug: 	zhc:ota:common: Payload offsets: start=155094 end=155144 dataSize=50
[2024-06-01 13:49:50] debug: 	zh:controller:endpoint: CommandResponse 0x0017880103f90135/11 genOta.imageBlockResponse({"status":0,"manufacturerCode":4107,"imageType":268,"fileVersion":16787456,"fileOffset":155094,"dataSize":50,"data":{"type":"Buffer","data":[222,12,226,192,48,239,128,155,175,150,31,208,183,109,56,202,225,3,1,51,189,128,161,233,78,107,70,60,72,213,187,237,68,3,205,181,226,84,154,224,56,19,170,157,84,81,154,76,186,64]}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-06-01 13:49:50] debug: 	zh:ember:queue: Status queue=0 priorityQueue=0.
[2024-06-01 13:49:50] debug: 	zh:ember: ~~~> [ZCL to=58395 apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":11,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"manufacturerCode":null,"transactionSequenceNumber":75,"commandIdentifier":5}]
[2024-06-01 13:49:50] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=56 Len=88]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=5]
[2024-06-01 13:49:50] debug: 	zh:controller:endpoint: CommandResponse 0x00178801011a6cbe/11 genOta.imageBlockResponse({"status":0,"manufacturerCode":4107,"imageType":267,"fileVersion":1124103171,"fileOffset":30606,"dataSize":50,"data":{"type":"Buffer","data":[82,129,101,66,6,164,230,155,250,132,223,205,75,25,202,128,102,79,213,134,81,122,30,126,61,51,33,83,174,109,100,86,147,185,118,156,40,249,96,192,3,44,175,147,160,27,242,15,4,112]}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-06-01 13:49:50] debug: 	zh:ember:queue: Status queue=1 priorityQueue=0.
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=5] Added to rxQueue
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6]
[2024-06-01 13:49:50] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=56 Len=7]
[2024-06-01 13:49:50] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=88 messageTag=10 status=SUCCESS]
[2024-06-01 13:49:50] debug: 	zh:ember: ~~~> [ZCL to=14442 apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":11,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"manufacturerCode":null,"transactionSequenceNumber":43,"commandIdentifier":5}]
[2024-06-01 13:49:50] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=57 Len=88]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=6]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=6] Added to rxQueue
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7]
[2024-06-01 13:49:50] debug: 	zh:ember:ezsp: Skipping async callback while waiting for response to command.
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=7] Added to rxQueue
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0]
[2024-06-01 13:49:50] debug: 	zh:ember:ezsp: Skipping async callback while waiting for response to command.
[2024-06-01 13:49:50] debug: 	zh:ember:ezsp: Skipping async callback while waiting for response to command.
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-01 13:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-01 13:49:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-01 13:49:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-01 13:49:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-01 13:49:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-01 13:49:53] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-01 13:49:53] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-01 13:49:53] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-01 13:49:53] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-01 13:49:56] debug: 	zh:ember:uart:ash: <--- [FRAME type=ERROR]
[2024-06-01 13:49:56] error: 	zh:ember:uart:ash: Received ERROR from NCP, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-06-01 13:49:56] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-01 13:49:56] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-06-01 13:49:56] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[... 32 lines cut ...]
[2024-06-01 13:49:57] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-06-01 13:49:57] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-06-01 13:49:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-06-01 13:49:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-06-01 13:49:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-06-01 13:49:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-06-01 13:49:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-06-01 13:49:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-06-01 13:49:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-06-01 13:49:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-06-01 13:49:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-06-01 13:49:58] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-06-01 13:49:58] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2024-06-01 13:49:58] debug: 	zh:ember:uart:ash: <--- [FRAME type=RSTACK]
[2024-06-01 13:49:58] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-06-01 13:49:58] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0]
[2024-06-01 13:49:58] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-06-01 13:49:58] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-06-01 13:49:58] debug: 	zh:ember:ezsp: ===> [FRAME: ID=0:"VERSION" Seq=0 Len=4]
[2024-06-01 13:49:58] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=0]
[2024-06-01 13:49:58] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-01 13:49:58] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-01 13:49:58] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=0] Added to rxQueue
[2024-06-01 13:49:58] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1]
```

Nerivec, 2024-06-01:
OTA has been thoroughly tested on very large ember networks (300+), so it shouldn't be a problem on smaller ones, except if something else is interfering...
Do you have any spammy devices on your network?
Last seen and availability can also interfere in some cases, try disabling either before starting the update, see if it improves.

PS: Z2M OTA is currently being reworked to address identified issues.

Mar1usW3, 2024-06-01:
no known spammy devices here. The reset repeated ~every minute till the OTA updated was done successfully.
Currently i don´t have any OTA pending to test but it was on every OTA update the same for about 8 devices.

Nerivec, 2024-06-01:
Do you have the list of models for those 8 devices?

Mar1usW3, 2024-06-01:
All of them are hue devices. Can share some more logs tomorrow.

Nerivec, 2024-06-10:
Any chance you can try latest dev branch, see how it behaves with the new [OTA refactoring](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7585)?

Nerivec, 2024-07-21:
Latest dev now has support for concurrency (should handle spam better), as well as several fixes from adding v8 support.
Can you confirm if this is still an issue with it?

Mar1usW3, 2024-07-29:
sorry for the late response...
I made the last days some OTA Tests with Yellow and the Skyconnect with the latest dev version and I was not able the reproduce the issue again.