Koenkk/zigbee2mqtt #22878: Several payload messages per second

BIGede, 2024-06-01:
### What happened?

After the update to 1.38.0 I have a socket from Nous that sends several payload messages per second. As a result, the log is flooded. The GUI is also much slower to use.

### Zigbee2MQTT version

1.38.1

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff Dongel-E

### Setup

Proxmox VM

### Debug log

[2024-06-01 17:30:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Steckdosen EG TV Beleuchtung', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":0.04,"indicator_mode":"off/on","last_seen":"2024-06-01T17:30:23+02:00","linkquality":92,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":234}'
[2024-06-01 17:30:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Steckdosen EG TV Beleuchtung', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":0.04,"indicator_mode":"off/on","last_seen":"2024-06-01T17:30:23+02:00","linkquality":92,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":234}'
[2024-06-01 17:30:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Steckdosen EG TV Beleuchtung', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":0.04,"indicator_mode":"off/on","last_seen":"2024-06-01T17:30:23+02:00","linkquality":88,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":234}'
[2024-06-01 17:30:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Steckdosen EG TV Beleuchtung', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":0.04,"indicator_mode":"off/on","last_seen":"2024-06-01T17:30:23+02:00","linkquality":92,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":234}'
[2024-06-01 17:30:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Steckdosen EG TV Beleuchtung', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":0.04,"indicator_mode":"off/on","last_seen":"2024-06-01T17:30:23+02:00","linkquality":92,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":234}'

Nerivec, 2024-06-01:
See if the new logging setting can help to reduce the load (assuming you don't care about MQTT publish lines):
https://www.zigbee2mqtt.io/guide/configuration/logging.html#reducing-mqtt-traffic

BIGede, 2024-06-01:
Ok, that helped. Thanks. 

Does that have any negative effects?

Nerivec, 2024-06-01:
You won't get `MQTT publish` (`info`) messages in the logs since it restricts `z2m:mqtt` namespace to `warning`.
It doesn't affect anything but the logged level for MQTT-related stuff.

_That kind of spammy device is definitely not good for your network though._

LaurentChardin, 2024-06-01:
I had similar problems in the past with devices with metering that were very very chatty: ideally it would be good to check their documentation to see if the reporting frequency can configured at the device level. I had to stop using some because they were spamming the zigbee network and creating congestion. (in the end, nothing Z2M is responsible of)

BIGede, 2024-06-01:
I meant that it started with the update. I have 18 Nous A1Z sockets in the network. Maybe that's the problem. I will take a closer look.

Nerivec, 2024-06-01:
1.37 had `MQTT Publish` lines in `debug`, but since we had many requests to put it back in `info`, it was raised again, with the setting I mentioned added to enable customization.

For devices that spam too much, sometimes adjusting the reporting intervals/values, and other settings can help reduce network congestion. See what's available for those plugs, and if they cooperate.

ghoz, 2024-06-03:
One other thing you can try beside changing the default reporting times is to add a small debounce to those devices (I set mine to 0.2s) to limit the number of mqtt publishes

BIGede, 2024-06-05:
How can I restore the configuration.yaml of Z2M to its original state? I tried yesterday and destroyed it in the process.  🙈 

I would like to see if the messages are back to normal. 

 homeassistant:
  legacy_entity_attributes: false
advanced:
  log_level: info
  log_namespaced_levels:
    z2m:mqtt: warning
  network_key:
    - 214
    - 105
    - 68
    - 176
    - 252
    - 16
    - 255
    - 117
    - 117
    - 83
    - 255
    - 140
    - 38
    - 82
    - 118
    - 12
  pan_id: 57154
  ext_pan_id:
    - 27
    - 208
    - 108
    - 218
    - 239
    - 239
    - 139
    - 189
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 514
    protocol: udp4
    type: '5424'
  last_seen: ISO_8601_local
mqtt:
  server: mqtt://core-mosquitto:1883
  user: mqttuser
  password: mqttuserpass
serial:
  port: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221130081631-if00
  adapter: ember
frontend:
  port: 8099
device_options:
  legacy: false
devices:
  '0x5c0272fffe70f4d2':
    friendly_name: Wohnzimmer Fenster Lampe
  '0x5c0272fffe87bc5d':
    friendly_name: Wohnzimmer Eck Lampe
  '0x5c0272fffe7e28b2':
    friendly_name: Esszimmer Fenster Lampe
  '0xa4c138e8a5a09ffa':
    friendly_name: Esszimmer Steckdose
  '0xa4c138faed8f1511':
    friendly_name: Anna Steckdose
  '0xa4c1389f2db67ed5':
    friendly_name: Schlafzimmer Steckdose
.
..
.
  '0xa4c138b6a70d687a':
    friendly_name: Steckdose EG Schlafzimmer Luftreiniger
  '0x0cae5ffffe4a4c56':
    friendly_name: EG Lampe Flur
 availability: true

Thank you.

BIGede, 2024-06-06:
> One other thing you can try beside changing the default reporting times is to add a small debounce to those devices (I set mine to 0.2s) to limit the number of mqtt publishes

What attitude have you changed? I have several for these sockets in Z2M. 

[image]

ghoz, 2024-06-08:
I changed the minimum interval and removed the Volt and Current reports as I don't use those.
Altough to be honest I didn't check if the firmware really honors these requests...

closed 2024-12-28 by github-actions[bot]

efilatov, 2025-01-02:
Guys, i can add that i got the same problem with Nous D4Z energy meter. First it raised my Domoticz CPU load to 100% due to amount of mqtt messages. I fixed this with debounce on zigbee2mqtt side, however after few hours i noticed my zigbee network is having issues even when i send commands from zigbee2mqtt UI itself (timeouts, delays). I raised this behavior to Nous support but i feel i will be returning the device back to the seller.