Koenkk/zigbee2mqtt #22879: host_fatal_error after changing firmware

teargas666, 2024-06-01:
### What happened?

hi. After upgrading firmware to the newest version I'm facing problems with my zigbee2mqqt. It's randomly stopping and when I'm trying to start the addon it's cannot start. I have to login to proxmox and change the settings in the VM. After that I have day or almost day of working addon... below you can find the details from the starting log: 

> [15:54:11] INFO: Preparing to start...
> [15:54:11] INFO: Socat enabled
> [15:54:12] INFO: Starting socat
> Socat startup parameters:
> Options:     -d -d
> Master:      pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
> Slave:       tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
> [15:54:12] INFO: Starting socat process ...
> 2024/06/01 15:54:12 socat[47] N PTY is /dev/pts/0
> 2024/06/01 15:54:12 socat[47] N listening on AF=2 0.0.0.0:8485
> [15:54:15] INFO: Starting Zigbee2MQTT...
> [2024-06-01 15:54:25] info: 	z2m: Logging to console, file (filename: log.log)
> [2024-06-01 15:54:25] info: 	z2m: Starting Zigbee2MQTT version 1.38.0 (commit #unknown)
> [2024-06-01 15:54:25] info: 	z2m: Starting zigbee-herdsman (0.49.2)
> [2024-06-01 15:54:26] info: 	zh:ember: Using default stack config.
> [2024-06-01 15:54:26] info: 	zh:ember: ======== Ember Adapter Starting ========
> [2024-06-01 15:54:26] info: 	zh:ember:ezsp: ======== EZSP starting ========
> [2024-06-01 15:54:26] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
> [2024-06-01 15:54:26] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
> [2024-06-01 15:54:26] info: 	zh:ember:uart:ash: Serial port opened
> [2024-06-01 15:54:26] info: 	zh:ember:uart:ash: ======== ASH starting ========
> [2024-06-01 15:54:29] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
> [2024-06-01 15:54:29] info: 	zh:ember:uart:ash: ======== ASH starting ========
> [2024-06-01 15:54:31] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
> [2024-06-01 15:54:31] info: 	zh:ember:uart:ash: ======== ASH starting ========
> [2024-06-01 15:54:34] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
> [2024-06-01 15:54:34] info: 	zh:ember:uart:ash: ======== ASH starting ========
> [2024-06-01 15:54:36] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
> [2024-06-01 15:54:36] info: 	zh:ember:uart:ash: ======== ASH starting ========
> [2024-06-01 15:54:39] error: 	z2m: Error while starting zigbee-herdsman
> [2024-06-01 15:54:39] error: 	z2m: Failed to start zigbee
> [2024-06-01 15:54:39] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
> [2024-06-01 15:54:39] error: 	z2m: Exiting...
> [2024-06-01 15:54:39] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
>     at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:850:19)
>     at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2683:24)
>     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
>     at Zigbee.start (/app/lib/zigbee.ts:62:27)
>     at Controller.start (/app/lib/controller.ts:109:27)
>     at start (/app/index.js:107:5)

### Zigbee2MQTT version

1.38.0-1

### Adapter firmware version

7.4.2 [GA]

### Adapter

ZBDongle-E

### Setup

Home Assistant on Proxmox VM. Z2M as an addon

Nerivec, 2024-06-01:
When it does this:
```logs
[2024-06-01 15:54:26] info: zh:ember:uart:ash: ======== ASH starting ========
[2024-06-01 15:54:29] info: zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-01 15:54:29] info: zh:ember:uart:ash: ======== ASH starting ========
[2024-06-01 15:54:31] info: zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-01 15:54:31] info: zh:ember:uart:ash: ======== ASH starting ========
[2024-06-01 15:54:34] info: zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-01 15:54:34] info: zh:ember:uart:ash: ======== ASH starting ========
[2024-06-01 15:54:36] info: zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-01 15:54:36] info: zh:ember:uart:ash: ======== ASH starting ========
```

that's usually when the baudrate of the flashed firmware, doesn't match the baudrate used by z2m. Make sure it matches, if it's not 115200 (default), you need to set it in the `configuration.yaml`. See [docs](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter).

If the baudrate is properly set, and it continues to do this, please post a log with `debug` level.

teargas666, 2024-06-01:
I've flashed it through the web flasher. I mean that standard settings were chosen. 
how can I debug?

Nerivec, 2024-06-01:
> I've flashed it through the web flasher. I mean that standard settings were chosen.

It depends on the firmware file you flashed. 
[ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl) => 115200
[ncp-uart-hw-v7.4.1.0-zbdonglee-230400.gbl](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.1.0-zbdonglee-230400.gbl) => 230400

If you are not sure, try flashing the [first one](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl) again. That's the default, Z2M should start without issue with that one (i.e. no need to explicitly set the baudrate in the `configuration.yaml`).

> how can I debug?

Just change the log level to `debug`.

teargas666, 2024-06-03:
ok, I've flashed it once again, but there is nothing new. During the night my addon stopped and cannot start... 

> [2024-06-02 23:05:37] info: 	z2m: Logging to console, file (filename: log.log)
> [2024-06-02 23:05:37] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
> [2024-06-02 23:05:37] info: 	z2m: Starting Zigbee2MQTT version 1.38.0 (commit #unknown)
> [2024-06-02 23:05:37] info: 	z2m: Starting zigbee-herdsman (0.49.2)
> [2024-06-02 23:05:37] debug: 	z2m: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[15],"extendedPanID":[96,101,71,10,251,167,157,133],"networkKey":"HIDDEN","panID":34434},"serialPort":{"adapter":"ember","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221202202613-if00"}}'
> [2024-06-02 23:05:38] debug: 	zh:adapter: Failed to validate path: 'Error: spawn udevadm ENOENT'
> [2024-06-02 23:05:38] info: 	zh:ember: Using default stack config.
> [2024-06-02 23:05:38] debug: 	zh:ember: Using delay=5.
> [2024-06-02 23:05:38] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[19,42,138,70,226,69,202,233,80,13,176,157,40,7,95,113],"panID":34434,"extendedPanID":[96,101,71,10,251,167,157,133],"channelList":[15]},"serialPort":{"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221202202613-if00","adapter":"ember"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
> [2024-06-02 23:05:38] info: 	zh:ember: ======== Ember Adapter Starting ========
> [2024-06-02 23:05:38] debug: 	zh:ember: Starting EZSP with stack configuration: "[object Object]".
> [2024-06-02 23:05:38] info: 	zh:ember:ezsp: ======== EZSP starting ========
> [2024-06-02 23:05:38] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
> [2024-06-02 23:05:38] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
> [2024-06-02 23:05:38] debug: 	zh:ember:uart:ash: Opening serial port with {"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221202202613-if00","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
> [2024-06-02 23:05:38] info: 	zh:ember:uart:ash: Serial port opened
> [2024-06-02 23:05:38] info: 	zh:ember:uart:ash: ======== ASH starting ========
> [2024-06-02 23:05:38] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
> [2024-06-02 23:05:38] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
> [2024-06-02 23:05:38] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
> [2024-06-02 23:05:38] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
> [2024-06-02 23:05:38] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
> [2024-06-02 23:05:38] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
> [2024-06-02 23:05:38] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
> [2024-06-02 23:05:39] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
> [2024-06-02 23:05:39] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
> [2024-06-02 23:05:39] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
> [2024-06-02 23:05:39] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
> [2024-06-02 23:05:39] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
> [2024-06-02 23:05:39] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
> [2024-06-02 23:05:39] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
> [2024-06-02 23:05:39] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
> [2024-06-02 23:05:39] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
> [2024-06-02 23:05:40] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | NCP status: NO_ERROR
> [2024-06-02 23:05:40] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
> [2024-06-02 23:05:40] info: 	zh:ember:uart:ash: ======== ASH starting ========
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
> [2024-06-02 23:05:40] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
> [2024-06-02 23:05:41] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
> [2024-06-02 23:05:41] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
> [2024-06-02 23:05:41] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[... cut ...]
> [2024-06-02 23:05:51] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
>     at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:850:19)
>     at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2683:24)
>     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
>     at Zigbee.start (/app/lib/zigbee.ts:62:27)
>     at Controller.start (/app/lib/controller.ts:109:27)
>     at start (/app/index.js:107:5)

Nerivec, 2024-06-03:
So it started properly but then stopped for some reason, and then couldn't start again.
Can you check the machine's logs to see if something is going wrong with the USB?
Do you have the logs from before it crashed and restarted?

It's behaving like the port it is trying to open after the restart is no longer the Sonoff Dongle-E, so it can open the port, but is unable to start the EmberZNet stack, since anything not Dongle-E would not reply to stack reset request. Something with proxmox and the passing of the port not happening as expected? I'm not familiar with proxmox, so, not sure how that happens or how it could go wrong...

teargas666, 2024-06-03:
Can it be caused as my Zigbee dongle is on the extension cord based on the LAN wire?

Skaronator, 2024-08-03:
I had the same issue with the "Waiting for RSTACK". I downgraded my [SLZB-06](https://www.zigbee2mqtt.io/devices/SLZB-06.html#smlight-slzb-06) core version from v2.3.6 back to v2.0.18 and this fixed my issue.

At least for this device it could be related that I'm unable to upgrade the zigbee module from 20240408 to 20240510 due to other issues.

closed 2024-12-28 by github-actions[bot]