Koenkk/zigbee2mqtt #22881: total freeze of pi3 on starting

jonesMeUp, 2024-06-01:
### What happened?

In 9/10 tries zigbee2mqtt freezes my pi3 right after  ... "RTS/CTS config is off, enabling flow control."
If it starts (via _npm start_), then i can use it and can pair my 2 Ikea Waterleak sensors. Stopping and starting z2m again is no problem.
But when i reboot the raspi, _npm start_ freeze it again.

- The serial port and the whole z2m folders are writeable by user pi.
- I tried several USB cords and all USB-Ports of the raspi.
- I tried several positions of the dongle, near at the raspi and far away
- No log file is written on a freeze.
- On all tries only the dongle and the power cord is connected to the pi.
- It happens with ember and ezsp.
- My raspi pico with a 433Mhz sender connected to serial port (echo "sendRf(674647)" > /dev/ttyACM0) has no problems with my raspi3 running on bookworm. 
- I tried /dev/ttyACM0 and /dev/serial/by-id/........

I have used z2m about 2 month before i updated the raspi to the latest raspian with all updates.
The first reboots made no problems, but now its frustrating.
I updated the dongle after the first freezes, but it didn't change anything. 

_npm test_ shows no problems even before a freezing "npm start"

What can I do to squeeze an error message out of z2m instead of a freeze?
Thanks for your time
jones

### What did you expect to happen?

An error message instead of a full crashed computer

### How to reproduce it (minimal and precise)

Reboot raspi on a working config

### Zigbee2MQTT version

1.37.0 - 1.38.0 (ember) 1.37.0 - 1.37.1 (ezsp)

### Adapter firmware version

7.4.2 [GA] (for ember) and for ezsp the last version shown on https://darkxst.github.io/silabs-firmware-builder/

### Adapter

ZBDongle-E

### Setup

plain z2m on Raspi3b with fhem

### Debug log

**no log on freezed raspi, here is a log of a running z2m**

[2024-05-31 21:51:07] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-31 21:51:07] info: 	z2m: Starting Zigbee2MQTT version 1.37.1 (commit #cda867a)
[2024-05-31 21:51:07] info: 	z2m: Starting zigbee-herdsman (0.46.6)
[2024-05-31 21:51:08] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-31 21:51:08] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-31 21:51:08] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-31 21:51:08] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-05-31 21:51:08] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-31 21:51:08] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-31 21:51:09] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-05-31 21:51:09] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-05-31 21:51:09] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-05-31 21:51:10] info: 	zh:ember: [STACK STATUS] Network up.
[2024-05-31 21:51:10] info: 	zh:ember: [INIT TC] NCP network matches config.
[2024-05-31 21:51:10] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-05-31 21:51:10] info: 	zh:controller: Wrote coordinator backup to '/opt/zigbee2mqtt/data/coordinator_backup.json'
[2024-05-31 21:51:10] info: 	z2m: zigbee-herdsman started (resumed)
[2024-05-31 21:51:10] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":2,"revision":"7.4.2 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2024-05-31 21:51:10] info: 	z2m: Currently 0 devices are joined:
[2024-05-31 21:51:10] warning: 	z2m: `permit_join` set to  `true` in configuration.yaml.
[2024-05-31 21:51:10] warning: 	z2m: Allowing new devices to join.
[2024-05-31 21:51:10] warning: 	z2m: Set `permit_join` to `false` once you joined all devices.
[2024-05-31 21:51:10] info: 	z2m: Zigbee: allowing new devices to join.
[2024-05-31 21:51:10] info: 	zh:ember: [STACK STATUS] Network opened.
[2024-05-31 21:51:10] info: 	z2m: Connecting to MQTT server at mqtt://localhost
[2024-05-31 21:51:10] info: 	z2m: Connected to MQTT server
[2024-05-31 21:51:10] info: 	z2m: Zigbee2MQTT started!
[2024-05-31 21:53:21] info: 	z2m: Disconnecting from MQTT server
[2024-05-31 21:53:21] info: 	z2m: Stopping zigbee-herdsman...
[2024-05-31 21:53:21] info: 	zh:controller: Wrote coordinator backup to '/opt/zigbee2mqtt/data/coordinator_backup.json'
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Total frames: RX=76, TX=143
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   DATA frames : RX=75, TX=67
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   DATA bytes  : RX=794, TX=731
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=75
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-05-31 21:53:21] info: 	zh:ember: [STACK STATUS] Network closed.
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash: Port closed. Error? no
[2024-05-31 21:53:21] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-05-31 21:53:21] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-05-31 21:53:21] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-05-31 21:53:21] info: 	z2m: Stopped zigbee-herdsman
[2024-05-31 21:53:21] info: 	z2m: Stopped Zigbee2MQTT

Nerivec, 2024-06-01:
From the logs, seems like it's starting properly _when it starts_. So I guess there must be something messing with it outside of Z2M.
Did you try plugging the adapter in a powered USB hub (a hub with an external power adapter to avoid drawing from the PI)?

jonesMeUp, 2024-06-01:
thx for your answer.
no, i don't have a powered USB hub.
But why it was running about 2 month without any problems?

Nerivec, 2024-06-01:
The update could have change internal power settings. It could also be the power adapter of the PI showing signs of trouble.
I'm no PI expert though, could be something else entirely, but power is often a source of weird issues on the PI...

jonesMeUp, 2024-06-01:
Ok, i will buy a powered USB and keep you informed.
Thx, again

nono20004, 2024-06-02:
Hello I have the same problem with zwave2mqtt have update your raspberry firmware ? If you have update, downgrade your raspberry pi 3 b model and your problème solved.
Sorry for my bad English

jonesMeUp, 2024-06-02:
@nono20004 thx, yes i have the latest firmware on my pi. i can`t wait to test your hint (when i am back home in the night or tomorrow).
@Nerivec the powered usb-switch will arrive in about 3 days, then i will test it with both firmawares.

nono20004, 2024-06-02:
Ok your are welcome
I wait your final result

jonesMeUp, 2024-06-02:
So... you are both right:

1. its not a z2m problem
2. raspi is the enemy

I tested a bunch of firmware verisons and thats the result:
[image]

**All** firmwares above 6.1.77 didnt't work.
I leave this issue open until i checked the last rpi firmware with an aktive USB-Hub.
This should help desperate raspi users to fix the problem. 

Without your help i didn't know where to search, because the dongle says max 100mA and Raspi3b says a sum of 1,2A is given to USB, so i took this as a fact and searched for days on my z2m settings, USB-Cords, dongle position, etc and also killed a 32**GB** SD-Card which was degraded to a 30**MB** Card, after cutting power on each try, while green LED says disc activity forever.

nono20004, 2024-06-03:
Hello thank you for your response.
I wish that USB hub solve the problem but for me with zwave aotec gen 5 I didn't
Good luck

JacekKac, 2024-06-04:
@jonesMeUp how did you downgrade pi firmware?

JacekKac, 2024-06-04:
ok i did a rpi-update to the newst firmware and it's working ok, in the meantime i disabled bt to save energy (not sure if this is necessary

jonesMeUp, 2024-06-04:
For all that need an info how to update:
On the picture above, you see the correct file (6.1.77) on the right you can copy the hash.
then sudo rpi-update _hash_ example for 6.1.77
`sudo rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f`

@JacekKac  all firmwares > 6.1.77 didn't work for me. make 1 or 2 reboot and look if it still works

jonesMeUp, 2024-06-04:
With my new active USB-Hub i have tested 2 of the newer firmwares and all worked well.
As the `max_usb_current=1` is meanwhile ignored by the raspi, i can see 2 solutions:

1. downgrade the raspi tp 6.1.77
2. use an aktive USB-Hub

jonesMeUp, 2024-06-08:
update:
the problem seems to be solved in the upcoming 6.7.y kernel
[patch info](https://github.com/raspberrypi/linux/pull/5775)

closed 2024-06-08 by jonesMeUp

Fougues, 2024-12-17:
Hi,
I have the same problem as you. where do you go to do the update: rpi-update ?
or do you have to type this the rpi-update to update?

Thanks a lot

Fougues, 2024-12-17:
my raspberry runs under HAOS

clauso, 2024-12-18:
You have to type in the command on the command line. if there is no rpi-update, then you need to install it first `sudo apt install rpi-update`.
Disclaimer: I have never used HAOS, so I cannot tell you, how it works there.

Fougues, 2024-12-18:
Thanks 
 But there is no command line on HAOS

Fougues, 2025-01-07:
do you think that version 2 of Zigbee2mqtt corrected the problem?

has anyone tried with the new version

jonesMeUp, 2025-01-07:
again, it's not a Zigbee2mqtt problem, it's a linux kernel problem (broken usb-power delivery).
don't know anything about haos, but you have to ask there for help to switch to a working kernel version.

uowis, 2025-01-08:
Why same kernel working on ZHA ?
Haos don't permit to change kernel so we only have the solution to wait and Believe Haos switch quick in working kernel version.

Fougues, 2025-01-08:
Helo,

HAOS developers should be informed who are in principle very responsive.  Do you know how to inform them?
THANKS

bennydiamond, 2025-02-19:
Thank you I was going crazy on this. `rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f` command did the trick.

Here's my setup
Raspberry Pi 3B
Powered from a 3A 5V switching regulator through Pi's GPIO header. Very low ripple validated with a scope.
OS: DietPi 9.10.0, kernel 5.15.0. Latest release as of 2025-02-19.
ZBDongle-E attached through a 30cm USB2.0 extension cable
Wired Ethernet, Wifi/BT disabled

Zigbee2MQTT was initially installed as a Docker container but as I was having freeze issues, I tried a bare metal install. Installing it natively on host would not solve the freeze issue.
Adding a 30 seconds pre-start delay to zigbee2mqtt systemd unit wouldn't solve freeze issues.
Currently rocking a bare metal install.

I also had the same issue with Raspberry Pi OS 64-Bit previously. I thought it might be an OS issue and that's why I decided to go for DietPi. Both OSes gave me the same issues.

I'm very happy to have stumbled on this issue.

Here's what fixed it. As other mentioned running the following command to downgrade your firmware works.

```rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f```

I selected yes on both warning prompts. The one about boot partition size (irrelevant as it concerns Pi4 and Pi5 only) as well as the one telling you it's a bad idea to invoke `rpi-update` without being told to do so.

Once finished, I rebooted and since then, no more freezes.

jorgedeand, 2025-03-14:
> Thanks But there is no command line on HAOS

I'm also running on HAOS. I found that you can downgrade to a HAOS version to a one that has a lower linux kernel version than the working one. For mi in RPi3B, HAOS 12.0 works like a charm. Then wait to update HAOS when finally includes kernel 6.7+ for your system.

In order to downgrade HAOS version I did the following:
In the addon File editor: in Settings/Execute shell command
`ha os update --version 12.0`

JacekKac, 2025-08-27:
can sombody confirm if is it fixed? i run into a hangup once again after upgrading to newest kernel not sure if this was the reason but i can boot without the dongle connected.

Bummvoll, 2025-09-02:
I would also second the question - running into similar issues now on HA OS 16.1

Is this related to the same old cause?

pbasista, 2025-09-22:
For what it is worth, I am also experiencing freezing of the entire Raspberry Pi 3 Model B Rev 1.2 since recently. It is reproducible every time while starting zigbee2mqtt with a Sonoff USB dongle adapter (not sure which version exactly) and a known good configuration that used to work on the same hardware with the same adapter before. It seems like it started happening after a software update.

I am running Arch Linux ARM with Linux kernel 6.12.47-1-rpi on armv7l architecture.

I am wondering why this issue keeps reoccurring with the newer Linux kernels. Perhaps it is indeed caused by some USB power delivery issue which was not actually fixed properly the last time.

pbasista, 2025-10-20:
> I don't know what it do

I would suggest to use the mainline Linux kernel. From my experience, it works correctly there.

The disadvantage is that not every advertised Raspberry Pi 3 feature works out of the box with the current mainline Linux kernels. A notable example is hardware accelerated video decoding and encoding. But as long as you do not need that and plan on using the Raspberry Pi e.g. as a Home Assistant server, it should probably be fine.

For instance the Arch Linux ARM offers a Raspberry Pi 3 image with the mainline Linux kernel. It uses the U-Boot bootloader and therefore things need to be configured slightly differently than what is common with images provided by the Raspberry Pi Foundation. For instance, editing `/boot/config.txt` has no effect and the U-Boot configuration needs to be edited instead to provide correct boot parameters to the Linux kernel. As long as you are okay with using that, it should be a fair option to consider, in my opinion.

jonesMeUp, 2025-10-20:
I used the older kernel version since i installed a new trixie with no problems, so i went never in problems again and can't say anything about haos.
but did you try this:
`Freeze ab Kernel 6.5 → behoben ab Kernel 6.7, Workaround per dtoverlay=dwc2,dr_mode=host oder durch Deaktivieren von hciuart in älteren Kerneln.`

hah-caa, 2026-10-17:
In case others are still having this issue and end up here as I did: I ran into the same issue with a fresh install of Raspberry Pi OS Trixie on a raspberry Pi 3B and using the Sonoff Zigbee Dongle E. The workaround from @CreatureFear was a hint to try resetting the device without unplugging and rebooting. It seem that once used, the device is left in a state that will trigger a system freeze upon restarting Z2M, but I found the following to work.

First: determine the vendor/product codes for the device (IDs will differ, depending on setup):
```shell
$ lsusb
Bus 001 Device 004: ID 1a86:55d4 QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2
``` 

Using this information, reset the device before starting, or between restarts of, Z2M:
```shell
$ sudo usb_modeswitch -v 0x1a86 -p 0x55d4 --reset-usb
``` 

To avoid manual interventions, the following directive can be adding to the `[Service]` block in  zigbee2mqtt.service:
`ExecStartPre=+/usr/sbin/usb_modeswitch -v 0x1a86 -p 0x55d4 --reset-usb`

One caveat is that `systemctl restart zigbee2mqtt` still results in a system freeze. Maybe there is a timing issue with that. Executing 
```shell
$ sudo systemctl stop zigbee2mqtt
$ sudo systemctl start zigbee2mqtt
``` 
as well as rebooting seems to work reliably (so far).