Koenkk/zigbee2mqtt #22882: No devices can be added

klio-klio, 2024-06-01:
### What happened?

I have 2 Zigbee networks in operation. 1 USB dongle with CC2538 hardware works completely. The second is SONOFF Zigbee 3.0 USB Dongle Plus, CC2652P. I have the problem that I can't teach any devices at once with the network with the Sonof Stick. I haven't changed the config files for at least 2 years

### Zigbee2MQTT version

1.38.0 

### Adapter firmware version

Coordinator-Typ zStack3x0 Coordinator-Version 20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus, CC2652P

### Setup

Add-on Homeassistant Intel 

### Debug log

[configuration Sonof.zip](https://github.com/user-attachments/files/15522992/configuration.Sonof.zip)

Tonio16, 2024-06-02:
Hello

`. I have the problem that I can't teach any devices at once with the network with the Sonof Stick.`
What do you mean? A device can be only paired to a single coordinator.

Antoine

klio-klio, 2024-06-03:
> Hallo
> 
> `. I have the problem that I can't teach any devices at once with the network with the Sonof Stick.` Was meinst du? Ein Gerät kann nur mit einem einzigen Koordinator gekoppelt werden.
> 
> Antoine

Yes. Pairing with cc2538 works. With Sonof no more.

closed 2024-12-21 by github-actions[bot]