Koenkk/zigbee2mqtt #22887: [New device support]: 	MG-AUSW02 is the same as LXN56-TS27LX1.2

jamesahendry, 2024-06-02:
### Link

https://www.amazon.com.au/Approved-Smart-Outlet-Automation-Control/dp/B08KD5S3H8/ref=sr_1_6?crid=32YCRIKA6N5ZN&dib=eyJ2IjoiMSJ9.ZX4hhGTegGWkk6-3RJpPp47NDk7rqa4PcL3XvtIxQTD-XZ8enCyu1ZrFwhikQZd10dXXCEj0kevOvVq3MIxhzOit0pPsfZkiQwF2fd0ZdfjSxyMZbJNPsRDPo6aXUwMeojTu0ZoOfdQFjZPhAHEUS7_u0nrTiVD8ULZMhLFqns0xJ6EWh-uFUo5R4GFmhCYOxOyCdvl1dmbRyiiAUZr3Ygzhrh2L49iJQ5U_JeJIiiKC6FpkiI7IMHFB-akrG8XDQt0sI6dtBTZk9atGzojyAZTR_-d7bV8I3rVz8_Zkj_8.ybCdREK2JnkJmSSrmjIx89gfGvSidPIH7ZS5A3mhP_4&dib_tag=se&keywords=zigbee+gpo&qid=1717322153&sprefix=zigbee+g%2Caps%2C262&sr=8-6

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x00158d000476f6ad","nwkAddr":40622,"manufId":4478,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN56-TS27LX1.2","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,3,6,5,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LXN56-TS27LX1.2","manufacturerName":"3A Smart Home DE","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"2019.11.11","swBuildId":"1000-0001"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0030cc05a8","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0030cc05a8","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,4,3,6,5],"outClusterList":[25],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0030cc05a8","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0030cc05a8","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"2019.11.11","swBuildId":"1000-0001","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1717320971532}

### Comments

I bought what I thought was MG-AUSW02.  It looks exactly the same as 

https://www.zigbee2mqtt.io/devices/MG-AUWS01.html

apart from the physical relay buttons being physically spread.  Link below to what I bought

https://www.amazon.com.au/Approved-Smart-Outlet-Automation-Control/dp/B08KD5S3H8/ref=sr_1_6?crid=32YCRIKA6N5ZN&dib=eyJ2IjoiMSJ9.ZX4hhGTegGWkk6-3RJpPp47NDk7rqa4PcL3XvtIxQTD-XZ8enCyu1ZrFwhikQZd10dXXCEj0kevOvVq3MIxhzOit0pPsfZkiQwF2fd0ZdfjSxyMZbJNPsRDPo6aXUwMeojTu0ZoOfdQFjZPhAHEUS7_u0nrTiVD8ULZMhLFqns0xJ6EWh-uFUo5R4GFmhCYOxOyCdvl1dmbRyiiAUZr3Ygzhrh2L49iJQ5U_JeJIiiKC6FpkiI7IMHFB-akrG8XDQt0sI6dtBTZk9atGzojyAZTR_-d7bV8I3rVz8_Zkj_8.ybCdREK2JnkJmSSrmjIx89gfGvSidPIH7ZS5A3mhP_4&dib_tag=se&keywords=zigbee+gpo&qid=1717322153&sprefix=zigbee+g%2Caps%2C262&sr=8-6

I have referenced, 

https://github.com/Koenkk/zigbee2mqtt/issues/21330

However when I removed my extenal converters on 1.38.0 the device was not detected

with extenal converters
[image]

without
[image]

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LXN56-TS27LX1.2'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'MG-AUSW02', // Vendor model number, look on the device for a model number
    vendor: 'Nue / 3A', // Vendor of the device (only used for documentation and startup logging)
    description: 'Smart Double GPO', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off], // We will add this later // fz.battery, fz.temperature, fz.humidity
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right') ],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {left: 1, right: 2};
    },
};

module.exports = definition;
```

Koenkk, 2024-06-02:
Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition) instead?

jamesahendry, 2024-06-03:
I am a complete novice at this - but I'll give it a go!

jamesahendry, 2024-06-03:
@Koenkk - Sorry for the simple questions but I'm not a software developer by trade and have never used Github to contribute to code!  I hope I've got it right after following the instructions

kamaldeepdhiman, 2024-06-08:
thanks mate i use above mentioned external convertor https://github.com/Koenkk/zigbee2mqtt/issues/21330. my Double GPO working fine now. thanks and appreciate your help.

Koenkk, 2024-06-08:
@kamaldeepdhiman integrated your changes, thanks! Device will be supported in the next z2m release.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-06-08 by Koenkk