Koenkk/zigbee2mqtt #22889: SONOFF SNZB-02P is missing from OTA tab on 1.38.0

satter, 2024-06-02:
### What happened?

After upgrade from 1.37.1 to 1.38.0 my SONOFF SNZB-02P is missing from OTA tab in the web UI:
[image]
Devices tab:
[image]

### What did you expect to happen?

The device should be displayed in OTA tab as it supports updates:
```
{
    "battery": 100,
    "humidity": 42.2,
    "last_seen": "2024-06-02T14:01:21+03:00",
    "linkquality": 120,
    "temperature": 29.1,
    "update": {
        "installed_version": 8448,
        "latest_version": 8448,
        "state": "idle"
    }
}
```

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

Official docker image on Ubuntu host

Koenkk, 2024-06-02:
This seems to be caused by https://github.com/Koenkk/zigbee-herdsman-converters/pull/7572 which removes `ota()`. @burmistrzak was this a mistake?

satter, 2024-06-02:
I think it would be better to keep OTA enabled even if there is no image available right now. IIUC it will be just a repo update as z2m could fetch new OTA file from repo with no z2m/zhc version upgrade required

'Images currently unavailable for device'  warnings should be lowered to info I think. There is no need to display those in UI popups

Koenkk, 2024-06-03:
@burmistrzak what was the exact log message?

satter, 2024-06-04:
Log message was like this:
```
warning: 	zhc:ota:common: Images currently unavailable for device 'SNZB-01P', hardwareVersion '0', manufacturerName eWeLink, {"fieldControl":0,"manufacturerCode":4742,"imageType":2054,"fileVersion":8704}'
```
Web UI popup looked like this:
[image]

satter, 2024-06-04:
I found this initiative from two years ago:
https://github.com/zigpy/zigpy/issues/987
https://github.com/Koenkk/zigbee-herdsman-converters/issues/4610

So I submitted a support ticket to Sonoff asking for OTA files

Koenkk, 2024-06-04:
`Images currently unavailable for device` will be logged to `debug` now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

satter, 2024-06-22:
I got this reply form Sonoff support:

> Please refer to the tutorial to configure Zigbee OTA in Home Assistant.
> https://sonoff.tech/product-review/local-configure-zigbee-ota-in-home-assistant-tutorial/#3

> There is no new version of firmware for SNZB-01P and SNZB-02P now, It will be uploaded when there is an update. Please follow the website to know the update, your patience is highly appreciated!

scottjim, 2024-12-17:
@satter A new firmware version has just been released: v2.2.0

https://forum.ewelink.cc/t/zigbee-snzb-02d-update-firmware/102657[](url)

satter, 2024-12-21:
@scottjim I don't have the SNZB-02D sensor model, only the SNZB-02P.
But the D model looks nice; I think I'll buy one.

Micha-Btz, 2025-01-14:
I was in hope for a new firmware, because my lost connection all the time.
Do you think there is hope for a new version?

closed 2025-03-24 by github-actions[bot]