Koenkk/zigbee2mqtt #22890: Ember failing to set options

Marlor, 2024-06-02:
### What happened?

Using the new ember driver these warnings show up on startup:

```
warning: 	zh:ember: [EzspConfigId] Failed to SET "ADDRESS_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
warning: 	zh:ember: [EzspConfigId] Failed to SET "NEIGHBOR_TABLE_SIZE" TO "26" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
warning: 	zh:ember: [EzspConfigId] Failed to SET "SOURCE_ROUTE_TABLE_SIZE" TO "200" with status=ERROR_INVALID_VALUE. Firmware value will be used instead.
warning: 	zh:ember: [EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
```

I compered the values with the Firmware values from
https://github.com/NabuCasa/silabs-firmware/wiki/Zigbee-EmberZNet-NCP-firmware-configuration
```
CONFIG_ADDRESS_TABLE_SIZE=16
CONFIG_APS_UNICAST_MESSAGE_COUNT=20
CONFIG_NEIGHBOR_TABLE_SIZE=26
CONFIG_SOURCE_ROUTE_TABLE_SIZE=200
CONFIG_MULTICAST_TABLE_SIZE=16
```

I feel like there is something wrong in Firmware or the way zigbee2mqtt is setting the options if,

Setting CONFIG_ADDRESS_TABLE_SIZE from 16 to 16 gets an ERROR_OUT_OF_MEMORY error etc
and especially if

setting SOURCE_ROUTE_TABLE_SIZE from 200 to 200 returns ERROR_INVALID_VALUE

### What did you expect to happen?

only warning
`warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.`
feels valid, trying to set 32 if 20 is set in firmware

### How to reproduce it (minimal and precise)

Fresh install of Homeassistant,Hassos and zigbee2mqtt
Latest SkyConnect Firmware flashed via
https://skyconnect.home-assistant.io/firmware-update/

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.4.2 [GA]

### Adapter

SkyConnect

### Setup

Add-on in Hassos on PI4

francoisp59, 2024-06-02:
I see these 3 messages:
zh:ember: [EzspConfigId] Failed to SET "ADDRESS_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
zh:ember: [EzspConfigId] Failed to SET "SOURCE_ROUTE_TABLE_SIZE" TO "200" with status=ERROR_INVALID_VALUE. Firmware value will be used instead.
zh:ember: [EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.

In case it matters, my Sonoff Dongle-E adapter is updated to 7.4.3.

DuncanRae, 2024-06-03:
I have an issue where I can't change any setting such as Debounce or Optimistic. It when I change the setting, it immediately goes back to what it was before. Not sure if it's related to this or not. 

Also using Ember with Sonoff E Dongle 7.4.2 firmware. Was working fine previously. Issue seems to have arisen with latest 1.38 Z2M release.

Nerivec, 2024-06-04:
Technically, it shouldn't be a problem if it fails to set a value in normal circumstances (if the adapter indeed lacks sufficient memory for it).
But here the firmware has to be in some kind of weird state to give these on skyconnect/dongle-e, it should go through fine.
Did you try unplugging the adapter for a minute or so, then see if it still gives the error when starting z2m again?
Can you link what firmware you are both using?

PS: About `ERROR_INVALID_VALUE`, I'm guessing it's a firmware bug that gives the wrong message, and is actually just another `ERROR_OUT_OF_MEMORY`.

Marlor, 2024-06-04:
Unplugging didn't help and i reflashed just to make sure.

Flashed via:
https://skyconnect.home-assistant.io/firmware-update/
Firmware used:
https://github.com/home-assistant/skyconnect.home-assistant.io/blob/main/public/firmwares/skyconnect_ncp-uart-hw_7.4.2.0.gbl
Build via:
https://github.com/NabuCasa/silabs-firmware-builder

I did also try firmware
https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/skyconnect/ncp-uart-hw-v7.4.2.0-skyconnect-115200.gbl

on this Firmware there is only one warning
`warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.`

acseven, 2024-06-16:
Also getting those warning messages on skyconnect (ZBT-1), after updating firmware via the [web flasher utility ](https://connectzbt1.home-assistant.io/firmware-update/#from-skyconnect) and using the suggested firmware update:

```
[11:33:55] INFO: Starting Zigbee2MQTT...
[2024-06-16 11:33:59] warning: 	zh:ember: [EzspConfigId] Failed to SET "ADDRESS_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-06-16 11:33:59] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-06-16 11:33:59] warning: 	zh:ember: [EzspConfigId] Failed to SET "NEIGHBOR_TABLE_SIZE" TO "26" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-06-16 11:33:59] warning: 	zh:ember: [EzspConfigId] Failed to SET "SOURCE_ROUTE_TABLE_SIZE" TO "200" with status=ERROR_INVALID_VALUE. Firmware value will be used instead.
[2024-06-16 11:33:59] warning: 	zh:ember: [EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
```

```
Zigbee2MQTT version: 1.38.0
Coordinator type: EmberZNet
Coordinator revision: 7.4.2 [GA]
Frontend version:
0.7.1
19.47.1
0.49.2
```

Nerivec, 2024-07-21:
Next release (current dev), will no longer try to override some in-firmware values to avoid issues from recent restrictions from Silabs. Z2M will rely on values built into the firmware instead.

@acseven Any update on the crash? Logs?

acseven, 2024-07-28:
> Next release (current dev), will no longer try to override some in-firmware values to avoid issues from recent restrictions from Silabs. Z2M will rely on values built into the firmware instead.
> 
> @acseven Any update on the crash? Logs?

@Nerivec Sorry, i've moved back  to ezsp firmware given that ember was too unstable. I might give it a go again in the near future.

Nerivec, 2024-08-20:
Should no longer be an issue since these values are no longer written during startup from 1.39.1.
If you do encounter any seemingly related troubles, please open a new issue (logs and behaviors here are no longer relevant since most of the startup was refactored).

closed 2024-08-20 by Nerivec

acseven, 2024-09-15:
> Next release (current dev), will no longer try to override some in-firmware values to avoid issues from recent restrictions from Silabs. Z2M will rely on values built into the firmware instead.
> 
> @acseven Any update on the crash? Logs?

Apologies on the late reply; I've updated my Skyconnect dongle with the latest firmware, and I've been using Z2M with 'ember' for a week now with no issues. It seems to have improved the network stability and device connectivity altogether.