Koenkk/zigbee2mqtt #22892: Bosch thermostat BTH-RA only Heating is shown as HVAC Mode in Home Assistant

adidashi79, 2024-06-02:
### What happened?

After Update to 1.38.0-1 only one System Mode is shown with the Bosch Thermostats, only heat. Bevor the update, the Bosch-specific operating mode did override system mode and the three operating modes where shown (schedule/manual/paus as automatic/heat/off)and could be used in Home Assistant. In zigbee2mqtt everything does show up.

### What did you expect to happen?

See above, so have the three modes heat, automatic and off in Home Assistant.

### How to reproduce it (minimal and precise)

Open climate control and try to switch Mode.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20210708

### Adapter

Sonor P

### Setup

Add-on Home Assistant OS

### Debug log

[2024-06-02 15:48:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":134,"color":{"x":0.15260547798886093,"y":0.055405508506904705},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 2', payload '{"brightness":134,"color":{"x":0.15260547798886093,"y":0.055405508506904705},"color_mode":"xy","color_temp":500,"linkquality":141,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":147,"color":{"x":0.1527122911421378,"y":0.06001373311970703},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 1', payload '{"brightness":147,"color":{"x":0.1527122911421378,"y":0.06001373311970703},"color_mode":"xy","color_temp":500,"linkquality":120,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":134,"color":{"x":0.15260547798886093,"y":0.055725947966735336},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 2', payload '{"brightness":134,"color":{"x":0.15260547798886093,"y":0.055725947966735336},"color_mode":"xy","color_temp":500,"linkquality":141,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":147,"color":{"x":0.1527275501640345,"y":0.060318913557640956},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 1', payload '{"brightness":147,"color":{"x":0.1527275501640345,"y":0.060318913557640956},"color_mode":"xy","color_temp":500,"linkquality":120,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":147,"color":{"x":0.1527122911421378,"y":0.059693293659876405},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 1', payload '{"brightness":147,"color":{"x":0.1527122911421378,"y":0.059693293659876405},"color_mode":"xy","color_temp":500,"linkquality":120,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":134,"color":{"x":0.15260547798886093,"y":0.055725947966735336},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 2', payload '{"brightness":134,"color":{"x":0.15260547798886093,"y":0.055725947966735336},"color_mode":"xy","color_temp":500,"linkquality":141,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":146,"color":{"x":0.1527122911421378,"y":0.059693293659876405},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 1', payload '{"brightness":146,"color":{"x":0.1527122911421378,"y":0.059693293659876405},"color_mode":"xy","color_temp":500,"linkquality":123,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":134,"color":{"x":0.15260547798886093,"y":0.055405508506904705},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 2', payload '{"brightness":134,"color":{"x":0.15260547798886093,"y":0.055405508506904705},"color_mode":"xy","color_temp":500,"linkquality":141,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":145,"color":{"x":0.1527122911421378,"y":0.05952544441901274},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 1', payload '{"brightness":145,"color":{"x":0.1527122911421378,"y":0.05952544441901274},"color_mode":"xy","color_temp":500,"linkquality":123,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":145,"color":{"x":0.1526970321202411,"y":0.05922026398107881},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 1', payload '{"brightness":145,"color":{"x":0.1526970321202411,"y":0.05922026398107881},"color_mode":"xy","color_temp":500,"linkquality":120,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":135,"color":{"x":0.15260547798886093,"y":0.055405508506904705},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 2', payload '{"brightness":135,"color":{"x":0.15260547798886093,"y":0.055405508506904705},"color_mode":"xy","color_temp":500,"linkquality":141,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":145,"color":{"x":0.1527122911421378,"y":0.05952544441901274},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 1', payload '{"brightness":145,"color":{"x":0.1527122911421378,"y":0.05952544441901274},"color_mode":"xy","color_temp":500,"linkquality":120,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:48:10] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue PlayG', payload '{"brightness":145,"color":{"x":0.1526970321202411,"y":0.05922026398107881},"color_mode":"xy","color_temp":500,"state":"ON"}'
[2024-06-02 15:48:10] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 1', payload '{"brightness":145,"color":{"x":0.1526970321202411,"y":0.05922026398107881},"color_mode":"xy","color_temp":500,"linkquality":120,"power_on_behavior":"off","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[... cut ...]
[2024-06-02 15:48:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Play 2', payload '{"brightness":39,"color":{"x":0.2635690852216373,"y":0.26877241168841076},"color_mode":"xy","color_temp":153,"linkquality":141,"power_on_behavior":"off","state":"OFF","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":null}'
[2024-06-02 15:49:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Luftsensor Küche', payload '{"battery":100,"humidity":64.4,"linkquality":54,"temperature":20.4}'

TheMagnetar, 2024-06-02:
Maybe this helps?
- https://github.com/Koenkk/zigbee-herdsman-converters/pull/7498#issuecomment-2101401201
- https://github.com/Koenkk/zigbee-herdsman-converters/pull/7498#issuecomment-2105424011

dierochade, 2024-06-08:
@adidashi79 
Did you succeed in adressing the issue? 

I found that the [code snippet provided to do the mapping in home assistant](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7498#issuecomment-2126213758) had an error, so I did update it. This should work now...

gpayer, 2024-06-11:
I have the same problem and I'm really not sure what to do now. Should I remove the entities from Z2M and readd them? Would they then show up correctly in HA?
The other solution seems to add an entry in configuration.yaml, did I understand this correctly?

dierochade, 2024-06-11:
> I have the same problem and I'm really not sure what to do now. Should I remove the entities from Z2M and readd them? 

No

>Would they then show up correctly in HA? 

No

> The other solution seems to add an entry in configuration.yaml, did I understand this correctly?

Unfortunately no. 

The solution is to alter the way z2m presents the device to home assistant in auto discovery. As I had some problems to figure out the situation myself, a brief summary:

z2m publishes a for each TRV (and similar for any other device) a message under the topic homeassistant/climate/ieee_of_the_device/climate/config (example for my setup: homeassistant/climate/0x18fc26000003b5e4/climate/config). 

This message is retained and allows home assistant to identify, read and control the device.

This message looks similar to this (put in any [json formatter ](https://jsonformatter.org/) to make it readable): 

`
{
  "action_template": "{% set values = {None:None,'idle':'idle','heat':'heating','cool':'cooling','fan_only':'fan'} %}{{ values[value_json.running_state] }}",
  "action_topic": "zigbee2mqtt/Thermostat Bosch 5 Kinderzimmer",
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state",
      "value_template": "{{ value_json.state }}"
    },
    {
      "topic": "zigbee2mqtt/Thermostat Bosch 5 Kinderzimmer/availability",
      "value_template": "{{ value_json.state }}"
    }
  ],
  "availability_mode": "all",
  "current_temperature_template": "{{ value_json.local_temperature }}",
  "current_temperature_topic": "zigbee2mqtt/Thermostat Bosch 5 Kinderzimmer",
  "device": {
    "identifiers": [
      "zigbee2mqtt_0x18fc26000003b5e9"
    ],
    "manufacturer": "Bosch",
    "model": "Radiator thermostat II (BTH-RA)",
    "name": "Thermostat Bosch 5 Kinderzimmer",
    "sw_version": "3.05.09",
    "via_device": "zigbee2mqtt_bridge_0xe0798dfffe0f683d"
  },
  "max_temp": "30",
  "min_temp": "5",
  "mode_command_topic": "zigbee2mqtt/Thermostat Bosch 5 Kinderzimmer/set",
  "mode_state_template": "{% set values = {'schedule':'auto','manual':'heat','pause':'off'} %}{% set value = value_json.operating_mode %}{{ values[value] if value in values.keys() else 'off' }}",
  "mode_state_topic": "zigbee2mqtt/Thermostat Bosch 5 Kinderzimmer",
  "modes": [
    "off",
    "heat",
    "auto"
  ],
  "name": null,
  "object_id": "thermostat_bosch_5_kinderzimmer",
  "origin": {
    "name": "Zigbee2MQTT",
    "sw": "1.38.0",
    "url": "https://www.zigbee2mqtt.io"
  },
  "temp_step": 0.5,
  "temperature_command_topic": "zigbee2mqtt/Thermostat Bosch 5 Kinderzimmer/set/occupied_heating_setpoint",
  "temperature_state_template": "{{ value_json.occupied_heating_setpoint }}",
  "temperature_state_topic": "zigbee2mqtt/Thermostat Bosch 5 Kinderzimmer",
  "temperature_unit": "C",
  "unique_id": "0x18fc26000003b5e9_climate_zigbee2mqtt",
  "mode_command_template": "{% set values = { 'auto':'schedule','heat':'manual','off':'pause'} %} \t{\"operating_mode\": \"{{ values[value] if value in values.keys() else 'pause' }}\"}"
}
`
you must replace the following properties with these values (already done in the example):

> ```
> modes: 
> 	["off", "heat", "auto"]
> mode_state_template: 
> 	{% set values = {'schedule':'auto','manual':'heat','pause':'off'} %}
> 	{% set value = value_json.operating_mode %}
> 	{{ values[value] if value in values.keys() else 'off' }}
> mode_command_topic: 
> 	zigbee2mqtt/<REPLACE with your friendly name>/set
> mode_command_template: 
> 	{% set values = { 'auto':'schedule','heat':'manual','off':'pause'} %}
> 	{"operating_mode": "{{ values[value] if value in values.keys() else 'pause' }}"}
> ```

In essence this reroutes the command for the operation mode to the right attribute and maps the command values 

This must be done every time, z2m sends the autodiscovery message (no other messages are published for these topics)
So, for instance, it must be done on every restart of z2m. 

This can be done with an import of the premade template for node red (addon available) as linked [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7498#issuecomment-2126213758). This should work more or less out of the box, only replacement of ieee and friendly name is required. Maybe you need to restart z2m (if node red is not ready fast enough) on bootup .

It also can done with an automation by using the mqtt.publish service (see [documentation](https://www.home-assistant.io/integrations/mqtt/#publish--dump-services)). you can use a template to modify the payload or just catch the message for each device, make the changes manually, save it and publish it on an appropriate trigger (restart and new message in the topic should do it). 
As I do almost all automations with node red, i cannot provide a blueprint for this. 

Hope this helps.

gpayer, 2024-06-12:
@dierochade thank you!

I already pieced together a solution by reading your other comments in the linked threads. Or I might just go back to 1.37 ;-)

To whoever will take care of this regression, the necessary changes for the auto detection template are already there, they just have to integrated into `homeassistant.js`. Did I understand the code correctly that it might be enough to add the necessary changes into the `legacyMapping` array?

Primevia, 2024-06-14:
I see all comments and issues relate to the Bosch thermostat which I also have. But I want to point out that the new released version of z2m has some more malfunctions. As example my entire power plugs and temp sensor randomly do not work. Multiple automation I have work randomly but I never had issues since over the last 3 years. Even sometimes when I open the z2m webui I do not see any devices inside. I am not a supporter of rollback but looking the result the new version has brought, it might be the best idea. Currently I have on hold the entire nodered climate on hold because it’s not working anymore. Likely it is summer and heaters are not needed till September I reflect on rollback to previous version.

Primevia, 2024-06-14:
> > I see all comments and issues relate to the Bosch thermostat which I also have.
> 
> @Primevia Can you please be a bit more specific? What issues _exactly_ are you having with the Bosch TRV? 🤔
@burmistrzak , the issue is the one documented here, I only have left the heat function in HA and cannot power it off anymore. I have three of the Bosch thermostat ll. inside nodered I have setup a small climate function to power on power off or set different temperature based on different conditions and triggers. In nodered I can also not power off the TRV because the havc mode off is not supported anymore. 
> 
> > But I want to point out that the new released version of z2m has some more malfunctions. As example my entire power plugs and temp sensor randomly do not work.
> 
> Which devices are unreliable?
@burmistrzak , Nous A1Z. here I have some automation to power on power of at specific conditions and triggers. Even if all is met they remain on or off and there is no error message in HA. I only see the automation was correctly executed. Before 1.38.0.1 never had this issue. 
> 
> > Multiple automation I have work randomly but I never had issues since over the last 3 years. Even sometimes when I open the z2m webui I do not see any devices inside.
> 
> Seems unrelated and cannot confirm myself. Still, what's your setup (size, versions, OS, etc.) like?
@burmistrzak , hassio no docker, z2m and nodered app on same rpi. More than 60 services, 1200 entities , 25 z2m devices, 3 tasmota, 6 Zwave, 10 wifi, wallbox, solar System. Always updated to last version immediately as soon  available. Last firmware on all devices and last version of all apps and integration. 
> 
> > I am not a supporter of rollback but looking the result the new version has brought, it might be the best idea. Currently I have on hold the entire nodered climate on hold because it’s not working anymore. Likely it is summer and heaters are not needed till September I reflect on rollback to previous version.
> 
> More details would be appreciated.
I hope I could deliver all information.

gpayer, 2024-06-14:
So I did some work on this.

I was able to create a unit test by adding the appropriate Bosch RBSH-TRV0-ZB-EU definition to the zigbeeHerdsman,js stub.

Initially without futher changes it produced the discovery payload that is currently actually produced, so that's good. I was able to adapt the code in homeassistant.js, quite crudely for testing purposes, so that a working discovery payload with appropriate templates was produced. And the payload check in my new unit test passes atm.

But from an architectural point of view more work should be invested to make this nice and extendable for future unknown use cases.

So my idea, I don't now if that is already possible, is that at the end of `exposeToConfig` for every discovery payload an extension hook is triggered, which might change the payload further. Like adding mappings for a device with poor protocol conformance.

I don't now what's possible with Z2M's extension system, but even if this case is not possible at the moment, it's probably not too complicated to add another possible hook.

This way both backwards compatibility can be managed and future use highly specific use cases can be handled. And if I understood the Z2M frontend correctly such extension could also be added at runtime to a running system.

@burmistrzak  what do you think?

gpayer, 2024-06-15:
@burmistrzak  thanks for your input, I solved it pretty similar. However there are some important changes:

* operating_mode is not in firstExpose, I implemented it this way:
  ```typescript
  const hasOperatingMode = definition && Array.isArray(definition.exposes) && definition.exposes.find((expose) => expose.name === 'operating_mode');
  ```
* `discoveryEntry.discovery_payload.mode_command_topic` can't stay true, because then `set/system_mode` is used as topic suffix. I actually changed code later on to allow for non-boolean values of `mode_command_topic`.
* But most importantly, if I understood this correctly, `operating_mode` is some custom Bosch invention. Its values are probably not standardized in any way as well. So there *must* be a check that this is a Bosch RBSH-TRV0-ZB-EU.

My suggestion, now without extension mechanics, would be to send all discovery payloads at the end of `exposeToConfig` together with device details etc. to a special function `overridePayloadForNonConformingDevices` or so. If the current device is one of the offending ones, then the custom payload values are set. Otherwise nothing is changed. This would be kind of similar to the `legacyMapping` definition, but more flexible.

I'll try to implement this tonight, just check my fork and branch for details: https://github.com/gpayer/zigbee2mqtt/tree/ha-ext-support-operating-mode

gpayer, 2024-06-16:
@burmistrzak thanks for your code review and the support in the PR! :grinning: 

Digging into this task was thankfully not too difficult, the code base is really good quality. I'm looking forward to being actually able to test this change, testing in production is sometimes necessary :wink:

adidashi79, 2024-06-16:
> @adidashi79 Did you succeed in adressing the issue?
> 
> I found that the [code snippet provided to do the mapping in home assistant](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7498#issuecomment-2126213758) had an error, so I did update it. This should work now...

I am not that keen in all that programming and scripting. Made a solution in NodeRed to set off via setting 5 degree and save the old temperature to turn on with that. So its fine and I understand what's your point with Bosch not sticking to Zigbee.
Thank you for your great work, that even such a dummy as me can do great things with Zigbee2mqtt.

Mar1usW3, 2024-06-17:
> @adidashi79 We already worked something out for HA users.
> 
> Once https://github.com/Koenkk/zigbee2mqtt/pull/2307 is merged, everything should _just work_, no manual configuration required. 😅

i think you linked the wrong pr

gpayer, 2024-06-17:
@Mar1usW3 this is the correct PR: #23075

gpayer, 2024-07-08:
The fix for this bug will come with release 1.39.1, however I don't know when it will be released.

That's why I created an addon to fix this problem locally in HA: https://github.com/gpayer/bosch-bth-ra-fixer

I'm running this on my live HA instance, it's so nice to finally have everything working again without manual intervention. Of course this addon will hopefully be obsolete soon. :wink:

tsloms, 2024-07-12:
After reading all this (and also other issues correlating with the Bosch Thermostat) I want to say **thank you** to everyone for contributing! 
I'm no expert at all in terms of programming and I'm glad that there will be a fix sometime soon for me, as I'll just be waiting for 1.39.1 until I upgrade and will stay on 1.37. for the time being.

gpayer, 2024-12-21:
@adidashi79 This issue has been fixed since the version mentioned above was released. So you could close this issue if you want to.

Or has anyone else the necessary rights to do that?

adidashi79, 2024-12-22:
Thank you for that good work.

PR #23075 (2024-07-03): Add device-specific config overrides in HomeAssistant extension
This fixes #22892.

On the one hand the Home Assistant MQTT extension expects a standardized interface to interact with devices, on the other hand Z2M provides a quite direct access to the exposed endpoints of a device.

Unfortunately Bosch BTH-RA implemented a custom expose `operating_mode`, which no other device uses.

This PR introduces a self-contained method `overridePayloadForNonConformingDevices` where special adaptions to the produced config can be done. At the moment this is only done for Bosch BTH-RA.

Unfortunately I wa [... cut]
closed 2024-12-22 by adidashi79