Koenkk/zigbee2mqtt #22894: zigbee2mqtt crashing

ar1nd4m, 2024-06-02:
### What happened?

While trying to add 3 Tradfri bulbs, the system keeps crashing saying DB entry already exists.

### What did you expect to happen?

To be paired

### How to reproduce it (minimal and precise)

Not sure, this seems to work usually.

### Zigbee2MQTT version

1.37.0 commit: 46f34c8

### Adapter firmware version

20240315

### Adapter

zStack3x0

### Setup

containerized on docker

### Debug log

[2024-06-02 16:34:47] debug: 	zh:controller: Device '0x0cae5ffffe0b127a' accepted by handler
[2024-06-02 16:34:47] debug: 	zh:controller: New device '0x0cae5ffffe0b127a' joined
[2024-06-02 16:34:47] debug: 	zh:controller: Creating device '0x0cae5ffffe0b127a'
[2024-06-02 16:34:47] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-06-02 16:34:47] info: 	zh:controller: Interview for '0x0cae5ffffe0b127a' started
[2024-06-02 16:34:47] debug: 	zh:controller:device: Interview - start device '0x0cae5ffffe0b127a'
[2024-06-02 16:34:47] debug: 	zh:zstack:znp: SREQ: --> ZDO - nodeDescReq - {"dstaddr":39703,"nwkaddrofinterest":39703}
[2024-06-02 16:34:47] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,2,23,155,23,155,35]
[2024-06-02 16:34:47] debug: 	zh:controller: Device '0x90ab96fffe879051' accepted by handler
[2024-06-02 16:34:47] debug: 	zh:controller: Deleted device '0x90ab96fffe879051' joined, undeleting
[2024-06-02 16:34:47] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-06-02 16:34:47] debug: 	zh:controller: Device '0x90ab96fffe879051' is already in database with different network address, updating network address
[2024-06-02 16:34:47] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-06-02 16:34:47] info: 	zh:controller: Interview for '0x90ab96fffe879051' started
[2024-06-02 16:34:47] debug: 	zh:controller:device: Interview - start device '0x90ab96fffe879051'
[2024-06-02 16:34:47] debug: 	zh:controller:device: Interview - skip node descriptor request for '0x90ab96fffe879051', already got it
[2024-06-02 16:34:47] debug: 	zh:controller: Device '0x90ab96fffe40a580' accepted by handler
[2024-06-02 16:34:47] debug: 	zh:controller: Deleted device '0x90ab96fffe40a580' joined, undeleting
[2024-06-02 16:34:47] info: 	z2m: Device '0x90ab96fffe879051' joined
[2024-06-02 16:34:47] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt_middle/bridge/event', payload '{"data":{"friendly_name":"0x90ab96fffe879051","ieee_address":"0x90ab96fffe879051"},"type":"device_joined"}'
[2024-06-02 16:34:47] info: 	z2m: Starting interview of '0x90ab96fffe879051'
[2024-06-02 16:34:48] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt_middle/bridge/event', payload '{"data":{"friendly_name":"0x90ab96fffe879051","ieee_address":"0x90ab96fffe879051","status":"started"},"type":"device_interview"}'
[2024-06-02 16:34:48] info: 	z2m: Device '0x0cae5ffffe0b127a' joined
[2024-06-02 16:34:48] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt_middle/bridge/event', payload '{"data":{"friendly_name":"0x0cae5ffffe0b127a","ieee_address":"0x0cae5ffffe0b127a"},"type":"device_joined"}'
[2024-06-02 16:34:48] info: 	z2m: Starting interview of '0x0cae5ffffe0b127a'
[2024-06-02 16:34:48] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt_middle/bridge/event', payload '{"data":{"friendly_name":"0x0cae5ffffe0b127a","ieee_address":"0x0cae5ffffe0b127a","status":"started"},"type":"device_interview"}'
Error: DatabaseEntry with ID '25' already exists
    at Database.insert (/app/node_modules/zigbee-herdsman/src/controller/database.ts:38:19)
    at Device.undelete (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:439:25)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:573:20)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)

LaurentChardin, 2024-06-02:
Hi,

Did you basically added the 3 of them at the same time, aka powering on the 3 bulbs at once, and clicking on Permit join ?
Sounds like a race condition. Maybe try to add them one by one ?

Best,

Koenkk, 2024-06-02:
Should have been fixed in the 1.38.0 release

closed 2024-06-02 by Koenkk