Koenkk/zigbee2mqtt #22897: PTM 215Z enOcean slow respons randomly

samuele2723, 2024-06-02:
### What happened?

Hello, i write here to have some support for my home switches because my wife is crazu with me!!

I have a large setup with Sonoff CC2531 coordinator, my network is composed by 190 devices across 4 levels (garage, ground, 1st, 2nd).
Overall the network works good, means i don't see delays of any kind when i use from home assistant app or from Z2M interface the lights. 

Only random delays i have, are when using my switches around the house, that are all PTM 215Z enOcean. And my wife is mainly using these physical buttons to control the lights.

[https://zigbee2mqtt.io/devices/PTM_215Z.html#enocean-ptm%2520215z]

Sometimes, when i click, the light is switch on\off instantly, (i could not even say there is a delay).
Some other times it takes 1 to 3 seconds. This happens also in the same switch in the same moment, means if i keep push on\off the delays changes time by time.

I can't say it's an hardware issue, because i have 25 of them and all are affected (especially ones far from coordinator that's at garage level) and also because i've used these with Philips Hue Bridge and they seemed to be more responsive in that setup

### What did you expect to happen?

I expect that there is no delay or minimum and repetitible always similar when i use switches. They should be almost instant in their responsiveness, I don't also want to see difference from ground floor or second floor, i would like to understand how to prioritize message responsiveness from these switches

### How to reproduce it (minimal and precise)

I can reproduce it every time, and is not related to what light i turn on, or how (home assistant or MQTT) seems related to the delay of the button transmission of PTM 215Z enOcean itself.

It can also be noticed by watching the state change of click actions of PTM 215Z enOcean in the dashboard, you can perceive the message is received in delay from the switch.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20210708

### Adapter

CC2531 zStack3x0

### Setup

Home assistant OS with MQTT and Z2m Addon

samuele2723, 2024-06-02:
Hello there, i didnt know about that. I paired all of them with "enable join" active by all devices i guess, because i leave always open the network when z2m start.

So should i re pair them selecting the closer router for each? Could you explain more ?

Koenkk, 2024-06-04:
No nothing has changed regarding GP devices

LaurentChardin, 2024-06-04:
GP devices are tricky because they don't really connect to the Zigbee network : they need to be *translated* to Zigbee using the Green Power Proxy, which is a capability that all routers have if they are Zigbee 3.0 (if i am not mistaken). Therefore there are really not regular devices, and the way the routers behaves to translate the GP messages into Zigbee messages can vary across manufacturers. 

As @burmistrzak states, you want to check your mesh quality too (printing the map and checking the LQI of your routers), and also be sure you have routers with good GP proxy support (difficult to know actually, but in the end, the philips ones have very good support for GP)

LaurentChardin, 2024-06-07:
I always thought that Input and Output GP clusters were part of the Zigbee 3.0 specifications, and needed for the certification.

Hedda, 2024-06-07:
> > I always thought that Input and Output GP clusters were part of the Zigbee 3.0 specifications, and needed for the certification.
> 
> @LaurentChardin Apparently not, because the E27 Hue bulbs are definitely Zigbee 3.0 certified. It really seems to be a policy choice by Signify to only include a Green Power Sink (server) with their hardwired fixtures.

As I understand it, all Zigbee Router (ZR) devices need to at a minimum include support for ZGPB/GPB (Zigbee Green Proxy Basic) to be compliant with Zigbee 3.0 certification requirements

chris-1243, 2024-06-08:
@burmistrzak 

> One good indicator I've found is the presence of an input cluster on the Green Power endpoint.
So far all GP proxies with an input cluster on that endpoint, will use Groupcast and not Unicast to pair ZGP devices.

Your sentence is a bit confusing for me. Shall I use a router with an input/output cluster on endpoint 242 to get unicast ?

chris-1243, 2024-06-09:
Now I do understand better your explanation. 
I have 7 PTM215Z and none of them are paired to a Hue router even if I have some in my network.

- 2 are paired via an Ikea LED2101G4 with input/output cluster on endpoint 242
- 2 other via an Ikea ICPSH24-10EU-IL-2 with input/output cluster on endpoint 242
- 2 via a Sonoff Dongle-E router with output cluster on endpoint 242 only
- 1 via an Ubisys S2 with input/output cluster on endpoint 242.

I have a PTM 216Z which is paired via a Hue 4034031P7 with output cluster on endpoint 242 only and it seems to work fine.

All my PTM 215Z work fine without any delays. My network is quite small compred to yours as I have only 35 devices.

chris-1243, 2024-06-09:
I just had a look on how to sniff traffic and it seems possible to use an ezsp adapter for this (I have a dongle-e laying for testing). You will have to give me some time to get everything running under windows... (not se easy🫣) I will try to send you some data.

Anything special I will have to look for?

chris-1243, 2024-06-09:
It is running in more than 10 minutes....

Now, you will have to help me a bit what I should check and extract to help you. There are so many different information that I am lost. I can see this
[image]
[image]

chris-1243, 2024-06-10:
> Which GP Proxy exactly did you use? IKEA or Ubisys?

The PTM used in the screenshots is paired via an IKEA router (LED2101G4).

chris-1243, 2024-06-10:
So, I used the same IKEA router (LED2101G4), and here are the results of a re-pairing of a PTM215Z.

[image]
[image]
[image]

If you need more or other fields to look at, just let me know

chris-1243, 2024-06-10:
Here we go

[image]
[image]
[image]

chris-1243, 2024-06-10:
@burmistrzak Tanks for your effort. Much appreciated.

I am aware it is not the best topic to use, I have a question about PTM216Z. Would you think there is any possiblity to better support them in term of router able to translate the frame sent by this device ? 
It is quite hard to find a Hue device able to do the translation. I have 4 different Hue lamps and only one is able to do the job...shame.
I may open a new issue if required😉

chris-1243, 2024-06-10:
@burmistrzak 

With no success I tried those models:
  - [8718696548738](https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738)
  - [9290022166](https://www.zigbee2mqtt.io/devices/9290022166.html#philips-9290022166)
  - [5047130P9](https://www.zigbee2mqtt.io/devices/5047130P9.html#philips-5047130p9)
 
I successfully paired it via this model:
  - [4034031P7](https://www.zigbee2mqtt.io/devices/4034031P7.html#philips-4034031p7)

chris-1243, 2024-06-10:
They all have a Green Power on endpoint 242 (at least the output cluster).
[8718696548738](https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738) has even an input cluster... I guess they might be too old.

chris-1243, 2024-06-11:
@burmistrzak PTM215Z are able to pair without any issues or maybe the one you just found. 

I will try to make a packet capture regarding PTM216Z. [8718696548738](https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738) just had an firmware update and it seems to accept to pair a PTM216Z...

chris-1243, 2024-06-12:
I hope I provided the right sequences. First, when the pairing is not working

[image]
[image]

Then when it successfully paired via [8718696548738](https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738)

[image]
[image]
[image]

And I have two *.pcapng backup for each try if more easier for you to analyse

chris-1243, 2024-06-13:
@burmistrzak To share my *.pcapng directly via github, it dosen't sound a good idea. I'm going to share my whole network on the internet and maybe more other info.

I did not apply any specific configuratioj to my PTM216Z. I have only one for testing. I do prefer the older 215Z.

I will do an other try. I suspect I did not reset correctly the PTM216Z.

chris-1243, 2024-06-13:
I used two differents bulbs

 - https://www.zigbee2mqtt.io/devices/9290022166.html#philips-9290022166 not successfull

- https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738 successfull pairing

chris-1243, 2024-06-13:
I have one based on an EZSP adapter running `ember` driver. The main problem is the PTM is able to pair directly to the coordinator on `ember`.

I will find a way to create a test network on `zstack` and remove two devices of my main network...

chris-1243, 2024-06-13:
@burmistrzak I just remember I have a RF-Star dongle laying somewhere for a backup of my main network😁.

Let's use it and share the result on github or via the link provided. It will be "safer" like this

chris-1243, 2024-06-13:
Finally, I created a test network and you get below two links with the successful and unsuccessful pairing. I did not have more time to try to set a command-based device model via NFC. I need to dig a bit more.

[failed](https://onetimedrop.app/#n2Sz0LGdkGx74zxKlvPzN985thaDB5giwk6rmkC9bLDivHwmZtmweeeuyzXTw7ECeqNLYdYrDjKESSPztvcyeI1)

[successful](https://onetimedrop.app/#D18nrpBYZk5fGI9vHwDX4PSi4tYYLzSvOXcuKpa9FIbuQkVm9SpFb7UNJGnach0rvB5rtghQvCE0QAFnwCx3Ku1)

I hope it may help to find out what is going on with this model.

chris-1243, 2024-06-13:
The link again...my computer turned off and everything has been closed......

[failed](https://onetimedrop.app/#5pnqa5YMVmT0p2ACmov3Y0B4sT9CzE6OgFdQDvvtfs41cHdCCeQISG9VpqH3Yg7BeiwmtAsLGgx1yzLHalaVj81)

[successful](https://onetimedrop.app/#l65F8Q1qT9OJcz2O7iOwYsMfZXmGCP3bcOpBFgISXzII9mX57oRCzB7rrJ2jiDVhkN9K4H1IhsaUJM3w8RbqRO1)

my apologies....

chris-1243, 2024-06-13:
Here we go

[PTM216Z](https://onetimedrop.app/#PVIpNyrtLLeluvBU8BABRkFDCF3qSRpTKzlbM5VnwWXWGaSNd6dSbj8lq8DOYn6mMZVmTwmOWvqPYosIfe52gO1)

chris-1243, 2024-06-13:
@burmistrzak yes I did. For me, the transfer is completed.

A new link?

chris-1243, 2024-06-13:
onetimedrop.app/#DxOMFwTK4sKtiZ1VtxklKf9PrqtKZ32U7rcu64NhKQiFzBJKCL5elyQSFVERwW6R2oeqC0LJGippwFg7Upz08Y1

Let's try this

chris-1243, 2024-06-13:
Good I am.....
I will make an other try tomorrow. The light used is in a kid room and it is not the the best moment to turn on/off a light.

I just had a look how to configure via NFC. It seems straight forward. At least, the Enocean app is easy to use. The Z2M documentation needs to be update as it doesn't point to the latest user manual.

chris-1243, 2024-06-13:
> Please share your experience! Quite interested in how well the NFC thing actually works.

I used an IOS smartphone and NFC works nicely. I could change the channel, commisioning mode and send the learn telegram. Unfortunately, I could not set any other configuration. The app did not provide other parameters to be set.

chris-1243, 2024-06-14:
> Quite impressive what EnOcean is able to accomplish with such little energy.

Yes and I do use NFC for pairing now (PTM216Z). No need anymore to click more than 7s on a button and to confirm the channel. Really great option. 

> Really like these things, even though they're a bit louder than regular switches in some cases. I'll have to someday look into acoustic dampening, to perfect them.

PTM216Z seems a bit louder than PTM215Z for me... and I don't like the click feeling while using the PTM216Z. As I have only one and for testing purpose only. I'm fine. They might be a bit louder, especially during the night when it is really quiet.

Below, my try of the day. I disabled all the filter in Wireshark and I did multiple tries. One interesting point, I did received errors in Z2M.
```
zigbee2mqtt  | [2024-06-14 13:12:14] error:     zh:controller:greenpower: Error: Data request failed with error: 'No network route' (205)
``` 
The link:

onetimedrop.app/#MnGceqV1g5h5m2fI6EaN45VCujugdGSot8p1XBaRefy50SRjGMPvhrZTF4PahKbsgKOc1dwaMSHxw2C8CMhiU71

chris-1243, 2024-06-14:
onetimedrop.app/#ixsNdIwBk2Yee6VFSTisl2UaSNUHYMnvAF83kNRzS4irWepa9G3DikNYHSi1EHFBK7qzq3hKt0wOnCysa7Qh411

I guess it is just a matter of feeling. PTM216Z seems a bit harder to press (it is new compared to my PTM215Z). It is more the sounds which, in my opinion, is more present while clicking.

What I don't like compared to the PTM215Z is how the commands are sent, especially the `release`. It is always the same `release` command sent. While using the PTM215Z/ZE, you normally have `press_x` then `release_x`. PTM216Z, it is `press_x` then just `release`. More complex when you create automation in HA or any other tool.

chris-1243, 2024-06-14:
[Pairing_PTM216Z_failed_2024.06.14.zip](https://github.com/user-attachments/files/15840605/Pairing_PTM216Z_failed_2024.06.14.zip)

> Are your EnOcean switches replacing traditional light switches, i.e. is there a hole in wall behind?
I'll definitely have to experiment with different types of fire-rated acoustic foam. When these switches are mounted on solid concrete or brick, they're surprisingly quiet in comparison.

Both way. On solid wall or to replace traditional light switches. It may depend on your wall maybe and the acoustic foam or material you have behind. For me, the click sound is almost similar. 

> Can't this be changed by switching to a command-based mode?

I need to read again the user manual. The description above reflects the generic command-based mode. Let's have a closer look.

chris-1243, 2024-06-14:
Now, I have one... 

[log.txt](https://github.com/user-attachments/files/15841325/log.txt)

chris-1243, 2024-06-15:
@burmistrzak 

> Packet capture
   Debug logs

[Pairing_PTM216Z_failed_2024.06.15.zip](https://github.com/user-attachments/files/15848353/Pairing_PTM216Z_failed_2024.06.15.zip)

> Devices involved (model, firmware)

Philips [9290022166](https://www.zigbee2mqtt.io/devices/9290022166.html#philips-9290022166), firmware 1.116.3

> Procedure (when, what, how)

I tried to pair the PTM216Z via NFC and via the standard procedure (old the channel button for more than 7s and wait in Z2M if the device is recognized). Both attempt failed with a no network route error (205).

> Also would be interesting to see, if anything changes when you switch the PTM216Z into a command-based mode (e.g. GreenPower_2).

Unfortunately and via the IOS app, I am not able to make those changes. I may only choose the channel, commissioning mode and send the pairing frame.

chris-1243, 2024-06-17:
> Can you provide another set of packets + logs, but use a PTM215Z this time around?
Everything else should stay the same (Hue light used, etc.), if possible.

[Pairing_PTM215Z_2024.06.17.zip](https://github.com/user-attachments/files/15863589/Pairing_PTM215Z_2024.06.17.zip)

There it is... Everything remained the same (same RPI4, coordinator, Hue router, Z2M version...) except the PTM used as requested.

> It should be possible to configure the PTM216Z device mode using a generic NFC tool, but we don't know what commands to send... Can you try contacting EnOcean for help with this?

The email is on its way. Let's see the answer from EnOcean. I downloaded a generic NFC app for iOS. I need to be sure which commands I should send. Otherwise, my PTM216Z might be broken...

chris-1243, 2024-06-18:
As I was trying to add some routers in your [table](https://github.com/Koenkk/zigbee2mqtt.io/pull/2811) I found **IKEA LED2103G5** is able to translate ZGP frame. So, as I do not use it for the moment, I picked this one to do some tries. I do not have to remove a device from my main network.

This IKEA device as both input/output cluster on endpoint 242. Pairing a PTM216Z just failed but the PTM215Z paired correctly in `unicast` mode.

> Also, which of the Hue lamps you've tested have a Green Power input and output cluster?

- Pairing successful:
  - [4034031P7](https://www.zigbee2mqtt.io/devices/4034031P7.html#philips-4034031p7), input/output cluster
  - [8718696548738](https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738), input/output cluster

- Pairing failed:
  - [5047130P9](https://www.zigbee2mqtt.io/devices/5047130P9.html#philips-5047130p9), only output cluster
  - [9290022166](https://www.zigbee2mqtt.io/devices/9290022166.html#philips-9290022166), only output cluster

[Pairing_PTM21xZ_ikea_led2103G5_2024.06.18.zip](https://github.com/user-attachments/files/15883111/Pairing_PTM21xZ_ikea_led2103G5_2024.06.18.zip)

You will find in this*.zip file a debug log and two captures. One regarding the PTM215Z pairing and the second regarding the failed attempt of the PTM216Z. Both tries were done via **IKEA LED2103G5**.

chris-1243, 2024-06-18:
> Also, and please correct me if I'm wrong here, but it seems to be impossible to pair the PTM216Z as unicast? If you get it paired, then only in broadcast/groupcast mode, right?

This is completely correct. Each time I have been able to pair the PTM216Z, it was only in `broadcast/groupcast` mode. I tried several different routers and except the two mentioned above from Philips, none have been able to achieve a successful pairing.

I would be really curious to see how a PTM215ZE is pairing to a network. It should be a `greenpower_2` and I would assume is acting like the PTM215Z. This an assumption only...

chris-1243, 2024-06-18:
I just had a look in the PTM215ZE and the default `ZGP Device ID` used is 0x02 (ZGP ON/OFF Switch). While pairing the good old PTM215Z, it is sending the exact same `Device ID`.

[image]

On the other hand, the PTM216Z is using `ZGP Device ID` 0x07 (Generic). I guess, as you already pointed, there might be something to dig in.  

[image]

chris-1243, 2024-06-18:
> PTM 215ZE uses by default the command-based device model “ON / OFF Switch” identified
by Device ID 0x02. PTM 215ZE can be configured via NFC to use one of the other command-
based device models or the Generic Switch device model via the NFC PC configuration tool
as described in chapter 4.2.

I need to find an NFC reader/writer in order to be able to change the Device ID....

it is not written the same in the PTM216Z user manual.

> By default, PTM 216Z uses the Generic Switch model identified by Zigbee Green Power Device
ID 0x07. It is possible to select a different (command-based) model via NFC as described in
chapter 4.2.

chris-1243, 2024-06-18:
> Maybe give this app a try: https://apps.apple.com/us/app/nfc-tools/id1252962749

I have it as well but the main problem is what kind of NFC commands should I write not to brick my PTM216Z and to change the ZGP device ID

chris-1243, 2024-06-19:
I just had a look there https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/converters/fromZigbee.ts and in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/enocean.ts. If we are able to change the ZGP device ID, the PTM216Z might be seen as a PTM215ZE in Z2M as they share the same `ieeeAddr`.

Z2M uses `Device ID` and `ieeeAddr` to do the difference.

> Hopefully EnOcean is replying to your inquiry. 🤞

Wait and see....

chris-1243, 2024-06-19:
> Not sure if the supported actions are exactly the same as with an actual PTM215ZE... 👀

According to the two user manuals, if configured as a ZGP device ID 0x02, commands seem to be the same.

samuele2723, 2024-06-20:
thank you! this post saved my time to understand how were unresponsive those switch

chris-1243, 2024-06-23:
So, I am still waiting an answer from EnOcean. 

By the time, I have found an usb NFC reader/writer. I hope I could use it with the PC based software and then be able to change the ZGP device ID configuration.

chris-1243, 2024-06-23:
@burmistrzak 

>@chris-1243 You'll have to register to download the NFC software... 🙄
https://www.enocean.com/en/product/enocean-nfc-configurator

Done already and the software is running. I read the docs and thought I may try with a more generic NFC reader

> 4.2.1 PC-based configuration
For PC-based configuration, EnOcean recommends TWN4 (order code T4BT-FB2BEL2-SIMPL)
from Elatec RFID Systems ( https://www.elatec-rfid.com/en/ ) as USB NFC reader.
This reader is shown in Figure 15 below

> AFAICT, it's designed to work with a specific model of USB NFC reader, but maybe a generic one will work as well?

I was completely aware while looking for a NFC reader... let's try a more generic one which was cheap anyway.

chris-1243, 2024-06-25:
@burmistrzak Some news for you... I received my NFC reader and unfortunatelly, the EnOcean software is only compatible with Elelabs Twin 4 family readers... At least, I had a try. This is the not funny part of the game.

I finally had an answer from EnOcean and things start to be really interesting from them. I am still waiting a final answer but it seems possible to change the device ID without their software😁.

chris-1243, 2024-06-25:
@burmistrzak 

the NFC memory Layout you are interested in changing is below (The values next to the line number represent the page address, ranging from 0x74 to 0x7f):
[image]

you will need to set the ZGP_PROTCOL_CONFIG to your desired value, and if needed, customize the ZGP_COMMAND_MAP.
The steps you need to follow:
1- authenticate with the NFC tag using the NFC password set by EnOcean, which is a device-specific 4 bytes value that could be extracted from the QR code on the label. 
for example if the QR code reading shows:
30PS3271-A216+2PDB05+30S015640A4+ZBD3B239A25D8FA6D1A42F9AFE58C59D2+3Z6401FAEDE90F69842BC440E8E3B707A7+31ZFD208781+S01640903168978
my password would be the 4 bytes value following +31Z which is 0xFD 0x20 0x87 0x81
2- The ZGP_PROTOCOL_CONFIG is the byte 2 on the page 0x74, its default value is 0x47 which represents the following:

[image]

you can change it to your desired value, for example, if you want to use device ID 0x01 you have to set the page 0x74 to 0x60 0x00 0x41 0x00

3- you have to confirm your change request by setting the page 0x61 to 0x00 0x0C 0x00 0x00
and the page 0x60 to 0x19 0x00 0x00 0x00

4- Click the button 5 times to apply your changes.

5- read the byte 2 on page 0x44 (represents the current configuration values for ZGP_PROTOCOL_CONFIG) to check if your configuration was successful.

Here is an example on how to set the device ID to 0x01 using NFC director:

[image]

chris-1243, 2024-06-25:
@burmistrzak my main problem remains the software used. 

PTM216Z is pin protected and you need to use it in order to read/write the nfc memory...i'll need some help🤫

chris-1243, 2024-06-25:
[image]

chris-1243, 2024-06-25:
@burmistrzak 

> The 4 bytes of the PIN can be found right before the last "+" in that string.

`EED9011A` (+31ZEED9011A)

It was already extracted😉

chris-1243, 2024-06-25:
@burmistrzak 

"The NFC memory Layout you are interested in changing is below (The values next to the line number represent the page address, ranging from 0x74 to 0x7f):

[image]

you will need to set the ZGP_PROTCOL_CONFIG to your desired value, and if needed, customize the ZGP_COMMAND_MAP.
The steps you need to follow:
1- authenticate with the NFC tag using the NFC password set by EnOcean, which is a device-specific 4 bytes value that could be extracted from the QR code on the label. 
for example if the QR code reading shows:
30PS3271-A216+2PDB05+30S015640A4+ZBD3B239A25D8FA6D1A42F9AFE58C59D2+3Z6401FAEDE90F69842BC440E8E3B707A7+31ZFD208781+S01640903168978
my password would be the 4 bytes value following +31Z which is 0xFD 0x20 0x87 0x81
2- The ZGP_PROTOCOL_CONFIG is the byte 2 on the page 0x74, its default value is 0x47 which represents the following:

[image]

you can change it to your desired value, for example, if you want to use device ID 0x01 you have to set the page 0x74 to 0x60 0x00 0x41 0x00

3- you have to confirm your change request by setting the page 0x61 to 0x00 0x0C 0x00 0x00
and the page 0x60 to 0x19 0x00 0x00 0x00

4- Click the button 5 times to apply your changes.

5- read the byte 2 on page 0x44 (represents the current configuration values for ZGP_PROTOCOL_CONFIG) to check if your configuration was successful.

Here is an example on how to set the device ID to 0x01 using NFC director:
[image]

"

Wrong manipulation. I do repost the answer from EnOcean...

chris-1243, 2024-06-25:
@burmistrzak 

Just to be sure.

I need to enter `1B:EE:D9:01:1A` in the command tab and send the command to the PTM ?

Then, I do the same with the other commands one by one ?

Sorry, brand new to do such things for me.

chris-1243, 2024-06-26:
@burmistrzak 

Here we go. The new output

[image]

chris-1243, 2024-06-26:
> Have you pressed the switch five times to finalize the configuration before reading the memory again?

Yes and I even confirm this with the support.

[image]

Definitely, there is something wrong

chris-1243, 2024-06-26:
> Oh, and do me a quick favor: Send just 60 to issue a GET_VERSION. This allows us to identify the NFC chip used.

```
>> 60
<< 00 04 04 05 02 02 13 03
``` 

My PTM216Z is quite recent. I have it since the beginning of the year and for testing only

Device Revision `DB-05`

KHV8, 2024-06-26:
Hi, I have 7 215Z enOcean switches and experience precisely what you describe.

Im wondering how you determine which router they use. The switches do not show up on the "map".

chris-1243, 2024-06-26:
@KHV8 
> Hi, I have 7 215Z enOcean switches and experience precisely what you describe.
> 
> Im wondering how you determine which router they use. The switches do not show up on the "map".

When you pair your PTM215Z, you should select a [compatible](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices) router and used the function `permit join via this router`. In you have used the `permit all`, your PTM215Z are paired in `broadcast` mode and there might starts your slow response.

chris-1243, 2024-06-26:
@burmistrzak 

> A correct PIN seems to be indicating by 00 00, so the issue must be somewhere else...

It looks like the configuration is not written in the device. Is an answer `00` the same as a true message ?

chris-1243, 2024-06-26:
@burmistrzak 

I had a try with an other pin (0000). The pin from the QR-Code is still the one to be used.

I got `00` as a response.

[image]

chris-1243, 2024-06-26:
I am unable to download your link. I still have the message connecting to the sender

chris-1243, 2024-06-26:
I am sorry...one more time for the link. Firefox badly crashed🤬 with my whole computer. I had to restart and I lost the connection.

A last new link plz???🫣

[36 more replies]

closed 2025-03-21 by github-actions[bot]