Koenkk/zigbee2mqtt #22898: Unable to start new installation of Z2M using SLZB-07P7 USB dongle

cassiusca, 2024-06-02:
### What happened?

When starting Z2M, I am getting Error: SRSP - SYS - ping after 6000ms. 

I'm setting up Zigbee for the first time on a new Synology.  have HA running in a VM on a Synology NAS DSM 7.2 with the USB port made available to the VM. The MQTT broker and Z2M are running in HA. I have a SLZB-07P7 USB dongle. I flashed the dongle with the recommended firmware linked on the Z2M website (https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P7_coordinator_20230507.zip).  

I tried all of the suggestions on the website that was linked in the log. Tried many variations on the config file and other tips around but just can't seem to get it to start. After asking for help on Discord, turns out there were two other people there right now with a similar problem and it was suggested to open an issue here.

Config file:
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: >-
    /dev/serial/by-id/usb-SMLIGHT_SMLIGHT_SLZB-07p7_be9faa0786e1ea11bd68dc2d9a583cc7-if00-port0
  adapter: zstack
advanced:
  log_level: debug

### What did you expect to happen?

I expected that Z2M would startup without an error.

### How to reproduce it (minimal and precise)

I followed the posted installation instructions closely.
Start MQTT broker in HA
Install Z2M in HA
Try to start Z2M 

### Zigbee2MQTT version

1.38.0-1

### Adapter firmware version

https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P7_coordinator_20230507.zip

### Adapter

SLZB-07P7

### Setup

Add-on in HA; HA running in VM on Synology NAS 723+ DSM 7.2

### Debug log

[08:56:43] INFO: Preparing to start...
[08:56:43] INFO: Socat not enabled
[08:56:44] INFO: Starting Zigbee2MQTT...
[2024-06-02 08:56:46] info: 	z2m: Logging to console, file (filename: log.log)
[2024-06-02 08:56:46] info: 	z2m: Starting Zigbee2MQTT version 1.38.0 (commit #unknown)
[2024-06-02 08:56:46] info: 	z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-02 08:56:46] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-SMLIGHT_SMLIGHT_SLZB-07p7_be9faa0786e1ea11bd68dc2d9a583cc7-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-06-02 08:56:46] info: 	zh:zstack:znp: Serialport opened
[2024-06-02 08:56:47] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-06-02 08:56:48] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-06-02 08:57:06] error: 	z2m: Error while starting zigbee-herdsman
[2024-06-02 08:57:06] error: 	z2m: Failed to start zigbee
[2024-06-02 08:57:06] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-06-02 08:57:06] error: 	z2m: Exiting...
[2024-06-02 08:57:07] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:101:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)

dinhchinh82, 2024-06-02:
have you tried to use an USB extent cable yet?

Marius-Ar, 2024-06-02:
Same issue here with the same dongle flashed with the recommended Z2M firmware.
With or without Docker (compose), extension cable, ... on a Raspberry Pi 5 running Debian bookworm. Also the same outcome on a Raspberry Pi b+.

I'm not dealing with the HA add-on though.

Testing the dongle directly on a MacOS laptop was successful and even allowed me to pair a device.

Not sure if that's of any help but `dmesg -w` output shows error on Zigbee2Mqtt startup :
```
[   70.847487] br-4c314505e575: port 1(vethf3b1010) entered blocking state
[   70.847495] br-4c314505e575: port 1(vethf3b1010) entered disabled state
[   70.847506] vethf3b1010: entered allmulticast mode
[   70.847561] vethf3b1010: entered promiscuous mode
[   70.875274] br-4c314505e575: port 2(vethdeeae5e) entered blocking state
[   70.875282] br-4c314505e575: port 2(vethdeeae5e) entered disabled state
[   70.875294] vethdeeae5e: entered allmulticast mode
[   70.875369] vethdeeae5e: entered promiscuous mode
[   70.875464] br-4c314505e575: port 2(vethdeeae5e) entered blocking state
[   70.875467] br-4c314505e575: port 2(vethdeeae5e) entered forwarding state
[   70.875540] br-4c314505e575: port 2(vethdeeae5e) entered disabled state
[   71.453925] eth0: renamed from veth3a1c8ef
[   71.502780] br-4c314505e575: port 2(vethdeeae5e) entered blocking state
[   71.502786] br-4c314505e575: port 2(vethdeeae5e) entered forwarding state
[   71.502811] eth0: renamed from vethb2100c0
[   71.526482] br-4c314505e575: port 1(vethf3b1010) entered blocking state
[   71.526488] br-4c314505e575: port 1(vethf3b1010) entered forwarding state
[   76.747711] cp210x ttyUSB0: failed set request 0x5 status: -32
[   76.899109] cp210x ttyUSB0: failed set request 0x5 status: -32
[   77.050507] cp210x ttyUSB0: failed set request 0x5 status: -32
[   96.206039] vethb2100c0: renamed from eth0
[   96.263165] br-4c314505e575: port 1(vethf3b1010) entered disabled state
[   96.281661] br-4c314505e575: port 1(vethf3b1010) entered disabled state
[   96.282351] vethf3b1010 (unregistering): left allmulticast mode
[   96.282357] vethf3b1010 (unregistering): left promiscuous mode
[   96.282359] br-4c314505e575: port 1(vethf3b1010) entered disabled state
[  100.496464] cp210x ttyUSB0: failed set request 0x12 status: -110
[  100.619612] veth3a1c8ef: renamed from eth0
[  100.659236] br-4c314505e575: port 2(vethdeeae5e) entered disabled state
[  100.678736] br-4c314505e575: port 2(vethdeeae5e) entered disabled state
[  100.679320] vethdeeae5e (unregistering): left allmulticast mode
[  100.679325] vethdeeae5e (unregistering): left promiscuous mode
[  100.679327] br-4c314505e575: port 2(vethdeeae5e) entered disabled state
```
Right after that the Z2M "SRSP - SYS - ping after 6000ms" error pops up.

Koenkk, 2024-06-02:
`SRSP - SYS - ping after 6000ms` means z2m cannot communicate with the device, so this a setup issue (nothing can be fixed from z2m here)

Lapeno89, 2024-06-16:
@cassiusca Were you able to figure out why the problem? I have the same problem with z2m, while if I use ZHA it works properly.

cassiusca, 2024-06-18:
> @cassiusca Were you able to figure out why the problem? I have the same problem with z2m, while if I use ZHA it works properly.

I ended up getting it working after switching to an ethernet coordinator dongle. My best guess as to why it wasn't working for me initially with the USB dongle was that there are issues with the USB port on Synology NAS.

closed 2024-07-07 by cassiusca