Koenkk/zigbee2mqtt #22903: Several Issues with Hue Bulbs After Upgrading to 1.38.0-1

fxfitz, 2024-06-02:
### What happened?

Several of my hue bulbs appeared to stop working with Z2A. I first noticed when my four "Lounge Ceiling" lights wouldn't turn on. They seem to be failing with this:

```
[2024-06-02 14:39:52] debug: 	z2m: Error: ZCL command 0x001788010d588288/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42382 - 11 - 243 - 6 - 11 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:485:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:767:28)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:587:23)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:48:17)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1155:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
```

Additionally, one of my lamp bulbs stopped responding too. I removed it from the network, however now it's won't repair. This appears to be the error:

```
[2024-06-02 14:38:19] debug: 	zh:ezsp:driv: zdoRequest error: Error: {"address":31314,"clusterId":32773,"sequence":181} after 10000ms Error: {"address":31314,"clusterId":32773,"sequence":181} after 10000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:727:43)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:404:31
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:626:35)
    at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:487:13)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:612:17)
[2024-06-02 14:38:19] debug: 	zh:controller:device: Interview - active endpoints request failed for '0x001788010277929f', attempt 2
[2024-06-02 14:38:19] debug: 	zh:controller:device: Interview - quirks check for 'LCT014'-'Signify Netherlands B.V.'-'Router'
[2024-06-02 14:38:19] debug: 	zh:controller:device: Interview - quirks did not match
[2024-06-02 14:38:19] debug: 	zh:controller:device: Interview - failed for device '0x001788010277929f' with error 'Error: Interview failed because can not get active endpoints ('0x001788010277929f')
    at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:633:19)
    at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:487:13)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:612:17)'
[2024-06-02 14:38:19] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2024-06-02 14:38:19] error: 	zh:controller: Interview failed for '0x001788010277929f with error 'Error: Interview failed because can not get active endpoints ('0x001788010277929f')'
[2024-06-02 14:38:19] error: 	z2m: Failed to interview '0x001788010277929f', device has not successfully been paired
```

### What did you expect to happen?

It should work. :-) 

### Zigbee2MQTT version

1.38.0-1

### Adapter firmware version

 7.3.1.0 build 176

### Adapter

I'm sorry, but I have no idea what to put here :-( 

### Setup

Home Assistant Yellow

### Debug log

[log.log](https://github.com/user-attachments/files/15526659/log.log)

schlaggi, 2024-06-03:
Same for me. Bulbs were lagging or not responding at all. Commands from Home Assistant were completely ignored despite HA saying the devices are online.
After troubleshooting this for half an hour I reverted back to 1.37.1 and everything worked immediately.

fxfitz, 2024-06-09:
@Koenkk Is there any more information you might need? How can I help? :-D

Koenkk, 2024-06-09:
You are using the old `ezsp` driver, it's recommended to use the new `ember` drivers ([info](https://github.com/Koenkk/zigbee2mqtt/discussions/21462))

fxfitz, 2024-06-09:
Ok, so, I _think_ I've flashed my Home Assistant Yellow's zigbee with v13 firmware. Switched to ember, but still having the same issue.
[image]

[log.log](https://github.com/user-attachments/files/15753140/log.log)

fxfitz, 2024-06-09:
Powering off the lights and powering them on seems to have resolved. Sorry for the false alarm!

I think I'm good now, so I'll close. Thanks for the hint on ember! :-D

closed 2024-06-09 by fxfitz