Koenkk/zigbee2mqtt #22904: Failed to remove group

RichardIstSauer, 2024-06-02:
### What happened?

I cannot delete the new created group.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.4.2 [GA]

### Adapter

 EmberZNet

### Setup

HA OS Addon

### Debug log

Info 2024-06-02 22:50:27 Removing 'Küche-Router' from 'Test'
Error 2024-06-02 22:50:38 Failed to remove from group (ZCL command 0x804b50fffe5e8e1d/1 genGroups.remove({"groupid":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":51513,"apsFrame":{"profileId":260,"clusterId":4,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":81},"zclSequence":96,"commandIdentifier":3} timed out after 10000ms))
Info 2024-06-02 22:54:52 Removing group 'Test' (force: false)
Error 2024-06-02 22:55:02 Request 'zigbee2mqtt/bridge/request/group/remove' failed with error: 'Failed to remove group 'Test' (force: false) (Error: ZCL command 0x804b50fffe5e8e1d/1 genGroups.remove({"groupid":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":51513,"apsFrame":{"profileId":260,"clusterId":4,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":113},"zclSequence":123,"commandIdentifier":3} timed out after 10000ms))'

irata-coder, 2024-06-08:
I'm seeing the same issue, but also I'm finding devices cannot be removed from any groups.  Also using EmberZNet, albeit adapter firmware 7.4.3.

> Request 'zigbee2mqtt/bridge/request/group/remove' failed with error: 'Failed to remove group 'Motion Test' (force: false) (Error: ZCL command 0x0ceff6fffe697835/1 genGroups.remove({"groupid":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":45097,"apsFrame":{"profileId":260,"clusterId":4,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":84},"zclSequence":38,"commandIdentifier":3} timed out after 10000ms))'

> Failed to remove from group (ZCL command 0x001788010916108e/2 genGroups.remove({"groupid":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":34274,"apsFrame":{"profileId":260,"clusterId":4,"sourceEndpoint":1,"destinationEndpoint":2,"options":4416,"groupId":0,"sequence":160},"zclSequence":44,"commandIdentifier":3} timed out after 10000ms))

Fred19732, 2024-08-05:
I don't know if you found how to remove but I did.
Iwent in the config.yaml of z2m.
I recreate a groupe with the name of the non possible erase group.
i had to put a device inside it (no matter which device)
I save the file
I restart HA, and after I could go in z2m and delete the group.

I hope this will help.
F.

RichardIstSauer, 2024-08-05:
It doesn't work for me.

Fred19732, 2024-08-05:
I'm sorry to hear that it doesn't work. and if you go into configuration.yml of z2m, normally you don't have faulty group. Is it well inside this configuraiton.yml that you rewrite the group.
To be clear, for me, after a crah, group I created before chnaged name for the ID number . my group was ID2 and rename (I don't know how and why) itself as 2.  i went in the configuration file and there was no more this group. I took the syntax for the first group and copy paste it, then rename with the faulty group "2". then copy paste a device I had in group 1 and save. afeter that , I restart HA, then go back in the configuration file, and what I wrote was saved. then, I went on z2m/goups. and I could acces to it, i renamed it, saved it, then restart again HA, and then I could go to z2M/group and delete it. I 'm sorry, i hope this is more clear, but I can't make print screnn as I'm not at home right now. i hope you will be able to do it.

retrography, 2024-09-26:
This is a bug and needs to be fixed. I am going through the same issue: Once the friendly name and the group id become the same for a group with no member, the group is impossible to remove.

stich86, 2024-09-26:
Got the same problem yesterdays
Trying to play with configuration.yaml, removing group form there break the front end (that doesn't start anymore)

Now I've a group that I cannot be deleted and also any action done on it doesn't do anything, device are not paired between themselves 

@Koenkk any hints or debug that we can run?

Currently I'm on the 1.40.1 with EzNet coordinator (7.4.2)

djurny, 2024-09-26:
Hi there,
In case removing devices from the group with zigbee2mqtt frontend does not work, you could try the following:
- Shut down any instance of zigbee2mqtt.
- **Make backups of your `configuration.yaml` and `database.db`**.
- Modify your `configuration.yaml`:
   - Remove all the `group:` definitions that match the group you want to remove by hand.
- Modify your `database.db`:
   - Remove any line that contains: `"groupID": <number>` by hand. Where `<number>` is the number that was assigned to the group.
- Restart your zigbee2mqtt instance.
- Re-pair the devices that were part of the group you just removed. 
Group membership is stored on the devices. Re-pairing the devices will remove any membership of any group on the device side.

This has worked for me on several occasions.
Groetjes,

hipp0o, 2024-10-16:
Thanks for your instructions!
Worked like a charme!

alexd-3d, 2024-11-20:
@djurny thanks for the great tip! Came here partially because my MOES smart knob failed to work in a group with the dimmer. I could delete stuff, but the newly created didn't work. Your key sentence is "Group membership is stored on the devices. Re-pairing the devices will remove any group membership on the device side." I didn't think about it. Now they are alive - thanks!

MichaelUray, 2024-12-27:
I can confirm this bug.

In my case I just had to stop the docker container and to remove the following line in the `database.db` with a text editor:
`{"id":57,"type":"Group","groupID":5,"members":[],"meta":{}}`

My used versions:
Zigbee2MQTT version:    [1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0) commit: [861cba6](https://github.com/Koenkk/zigbee2mqtt/commit/861cba6)
Coordinator type:    zStack3x0
Coordinator revision:    20240710
Frontend version:    0.7.6
zigbee-herdsman-converters version:    20.58.0
zigbee-herdsman version:    2.1.9

gvlekke, 2025-01-28:
Same problem but with different log:

```
error 2025-01-28 21:26:03z2m: Request 'zigbee2mqtt/bridge/request/group/remove' failed with error: 'Group '2' does not exist'
info 2025-01-28 21:26:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/remove', payload '{"data":{},"error":"Group '2' does not exist","status":"error","transaction":"qrt5s-2"}'
```

Zigbee2MQTT version:
2.0.0-2

Adapter firmware version
unknown, dont know where to find this

Adapter
ember

Setup
HA OS Addon

Fred19732, 2025-03-30:
This is ok all is working now.

xenadmin, 2025-04-06:
I'm having the same problem, just tried to create my first group:
```
[2025-04-06 17:33:25] error: 	z2m: Request 'zigbee2mqtt/bridge/request/group/remove' failed with error: 'Group '1' does not exist'
[2025-04-06 17:33:25] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/remove', payload '{"data":{},"error":"Group '1' does not exist","status":"error","transaction":"g7sp9-1"}'
```