Koenkk/zigbee2mqtt #22905: Difference between config and actually transmitted color_mode

strandbein, 2024-06-02:
### What happened?

Philips 4090331P9
Hue Ensis (white) and
Philips 4080248P9
Hue Signe floor light
Exposes	light (color_xy, color_hs)
but
homeassistant/light/0xxx/light/config: "supported_color_modes":["xy","color_temp"]
"xs" is missing, which then leads to problems in HA. 

### What did you expect to happen?

homeassistant/light/0xxx/light/config: "supported_color_modes":["xy","xs","color_temp"]

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Add-on Home Assistant, rpi4-64

sjorge, 2024-06-05:
xs is not a valid color mode according to the ZCL, do you mean hs ?

strandbein, 2024-06-05:
sorry, yes: hs

sjorge, 2024-06-05:
@Koenkk could with the refactors to modernExtend some of the devices that were slightly d ifferent have there config lost?

I know most newer stuff does XY, HS, CT or CT or nothing, but IIRC some older did not.  It seems the hue one now only has a generic color: true options.

Koenkk, 2024-06-08:
Z2M only exposes either `hs` or `xy` to Home Assistant since HA cannot handle both `hs` and `xy` at the same time (from what I remember)

sjorge, 2024-06-08:
> Z2M only exposes either `hs` or `xy` to Home Assistant since HA cannot handle both `hs` and `xy` at the same time (from what I remember)

That sounds vaguely familiar, I guess technically would could export the entity twice one as xy once as hs... but I don't use hass so that might just be impossible or make thing worse.

closed 2024-12-21 by github-actions[bot]