Koenkk/zigbee2mqtt #22906: [New device support]: For Tuya Model: RTCZ03Z Vendor: _TZE204_uxllnywp Description: Human presence sensor 24g

krikkoo, 2024-06-02:
### Link

https://a.aliexpress.com/_EHadEkr

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0xa4c13881f0f4bbe0","nwkAddr":48518,"manufId":4417,"manufName":"_TZE204_uxllnywp","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74,"zclVersion":3}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1717362619079}

### Comments

I tried a lot of external config and changed also to dev environment but nothing is working. the only external config that is able to recognize the sensor is the following but only this nothing works, I have changed only model.
I'm using z2m on Raspberry pi following this installation [link](https://www.zigbee2mqtt.io/guide/installation/01_linux.html). 

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_uxllnywp']),
    model: 'RTC ZCZ03Z,
    vendor: 'TuYa',
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: console.log, 
    // configure: tuya.configureMagicPacket,
   
    exposes: [
        e.presence(), 
           
        exposes.enum('movement_type', ea.STATE, ['none', 'small', 'large'])
            .withDescription('traffic of movement from the point of view of the radar'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(1).withUnit('s')
            .withDescription('detection delay'),
        exposes.numeric('detection_range', ea.STATE_SET).withValueMin(150).withValueMax(600).withValueStep(10)
            .withDescription('far sensitivity of the radar'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1)
            .withDescription('Sensitivity of the radar'),
        exposes.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1)
            .withDescription('Motionless sensitivity of the radar'),
        exposes.numeric('current_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
        exposes.numeric('battery_level', ea.STATE).withDescription('battery level').withUnit('%'),
           
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence',tuya.valueConverter.trueFalse1],
            [11, 'movement_type', tuya.valueConverterBasic.lookup({'none': 0, 'small': 1, 'large': 2})],
            [12, 'detection_delay', tuya.valueConverter.raw],
            [13, 'detection_range', tuya.valueConverter.raw],
            [15, 'radar_sensitivity', tuya.valueConverter.raw],
            [16, 'motion_sensitivity', tuya.valueConverter.raw],
            [19, 'current_distance', tuya.valueConverter.raw],
            [101, 'battery_level', tuya.valueConverter.raw],
            
         
        ],
    },
};

module.exports = definition;
```
Thanks in advance for your amazing job guys.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_uxllnywp',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

krikkoo, 2024-06-02:
after post also above config is stopped to works.

rcubed47, 2024-06-08:
I'm having issues with adding this device as well, adding support for this device would be great!

GuillaumeB-GitHub, 2024-06-20:
I’d love to buy this product so adding support would be great. Thanks

krikkoo, 2024-06-27:
I have done some step forward following this [procedure](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html) I have scanned data points from Tuya IoT Platform. 
This is the result: ```{"1":"State",
"101":"Target distance",
"102":"Light intensity",
"103":"Hold delay",
"104":"Indicator led",
"105":"None delay time(Min)",
"106":"None delay time(Sec)",
"107":"Detected distance max",
"108":"Dectected distance min",
"111":"sensitivity",
"112":"Hold sensitivity",
"113":"Restore factory settings",
"120":"debug"}```.

After that i started to integrate them in the external converter, almost all is now working, the devise is marked as supported.
I'm not able yet to find a solution for presence state field, i have tried several configuration but nothing seems to work. Also it is not clear the meaning and how to integrate the data points 112, 113 and 120, moreover if I disconnect the power to the sensor in z2m is marked as online.
Here below what i have done till now and the result in z2m.
```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_uxllnywp']),
    model: 'RTC ZCZ03Z',
    vendor: 'TuYa',
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: console.log,
    // configure: tuya.configureMagicPacket,

    exposes: [
                e.illuminance_lux(), e.presence(),
       // exposes.enum('presence_state', ea.STATE, ['none', 'someone']).withDescription('Presence'),
                exposes.numeric('detection_distance_max', ea.STATE_SET).withValueMin(0).withValueMax(840).withValueStep(1)
            .withDescription('Max detection distance').withUnit('cm'),
                exposes.numeric('detection_distance_min', ea.STATE_SET).withValueMin(0).withValueMax(840).withValueStep(1)
            .withDescription('Min detection distance').withUnit('cm'),
                e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(1).withValueMax(59).withValueStep(1)
            .withDescription('Delay time').withUnit('s'),
                exposes.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1)
            .withDescription('Presence sensitivity'),
                e.binary('indicator', ea.STATE_SET, 'ON', 'OFF').withDescription('LED Indicator'),
        //exposes.enum('man_state', ea.STATE, ['nobody','exist']).withDescription('Presence state'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence_state', tuya.valueConverter.trueFalse0],
            [101, 'target_distance', tuya.valueConverter.raw],
            [102, 'illuminance_lux', tuya.valueConverter.raw],
            [103, 'fading_time', tuya.valueConverter.raw],
            [104, 'indicator', tuya.valueConverter.onOff],
            [107, 'detection_distance_max', tuya.valueConverter.raw],
            [108, 'detection_distance_min', tuya.valueConverter.raw],
            [111, 'presence_sensitivity', tuya.valueConverter.raw],
        ],
    },
};
module.exports = definition;
```
Dashboard
[image]

Exposes
[image]

State
[image]

Some help will be very appreciated.
Thanks

krikkoo, 2024-07-01:
Hi folk, I made some step forward and i reached a compromise, all it seems to work as expected except  the doubts regarding the meaning and how to integrate the data points 112, 113 and 120, and if I disconnect the power to the sensor in z2m is marked as online.
Here below my final external configuration and the result in z2m.

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_uxllnywp']),
    model: 'RTC ZCZ03Z',
    vendor: 'TuYa',
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: console.log,

    exposes: [
        e.illuminance_lux(),
        exposes.enum('presence_state', ea.STATE, ["none","large_move"])
            .withDescription('Presence'),
        exposes.numeric('detection_distance_max', ea.STATE_SET).withValueMin(0).withValueMax(840).withValueStep(1)
            .withDescription('Max detection distance').withUnit('cm'),
        exposes.numeric('detection_distance_min', ea.STATE_SET).withValueMin(0).withValueMax(840).withValueStep(1)
            .withDescription('Min detection distance').withUnit('cm'),
                e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(1).withValueMax(59).withValueStep(1)
            .withDescription('Delay time').withUnit('s'),
        exposes.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1)
            .withDescription('Presence sensitivity'),
        e.binary('indicator', ea.STATE_SET, 'ON', 'OFF').withDescription('LED Indicator'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence_state', tuya.valueConverterBasic.lookup({'None': 0,'Someone': 4})],
            [101, 'target_distance', tuya.valueConverter.raw],
            [102, 'illuminance_lux', tuya.valueConverter.raw],
            [103, 'fading_time', tuya.valueConverter.raw],
            [104, 'indicator', tuya.valueConverter.onOff],
            [107, 'detection_distance_max', tuya.valueConverter.raw],
            [108, 'detection_distance_min', tuya.valueConverter.raw],
            [111, 'presence_sensitivity', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

Dashboard
[image]

Exposes
[image]

State
[image]

To make it to work I had re-pairing the sensor in z2m.

If someone could clarify my doubts will be very appreciated.
Thanks

Koenkk, 2024-07-01:
@krikkoo would you mind creating a pull request? You can do so by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)

krikkoo, 2024-07-02:
@Koenkk , i have opened a pull request but i received some errors in the code, I'm very newbie can you kindly check what is wrong.
Thanks

samumar82, 2024-07-08:
Hello, I have the same sensor both through Local Tuya and Z2M I'm not able to pair it up.
Can you please share the step by step guide to pair it via Z2M?
I've got a Sonoff usb dongle plus and it does not discover the sensor at all, many thanks!

rcubed47, 2024-07-12:
@krikkoo I tried using your converter pasted above but after restarting Home Assistant and re-pairing the device nothing changes, it exposes nothing and still shows unsupported. Any suggestions on what i could try? I have added the converter in Z2M configuration.yaml and created the converter file in the Z2M folder. Under settings of Z2M it shows my converter listed under External Converter tab. Logs show only this for that device

> Info 2024-07-12 14:32:31z2m: Successfully interviewed '0xa4c138f062b2515a', device has successfully been paired
Warning 2024-07-12 14:32:31z2m: Device '0xa4c138f062b2515a' with Zigbee model 'TS0601' and manufacturer name '_TZE204_uxllnywp' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2024-07-12 14:32:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138f062b2515a","ieee_address":"0xa4c138f062b2515a","status":"successful","supported":null},"type":"device_interview"}'

krikkoo, 2024-07-13:
You're Welcome.

closed 2024-12-29 by github-actions[bot]