Koenkk/zigbee2mqtt #22909: [New device support]: Moes ZTS-US_2gang Model: ZS-US1-WH-MS

KyleStilkey, 2024-06-03:
### Link

https://www.amazon.com/MOES-Requires-Neutral-Capacitor-Control/dp/B08X3K14FC/ref=sr_1_6?th=1

### Database entry

{"id":54,"type":"EndDevice","ieeeAddr":"0xa4c13842d5d68a72","nwkAddr":4865,"manufId":4417,"epList":[1,2],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65503":"���-\u0012���-\u0016","65506":55,"65508":0,"appVersion":80}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"interviewCompleted":false,"meta":{},"lastSeen":1717382314868}

### Comments

I attempted to try and create a external definition but sadly I am at a loss and not sure where I am going wrong, but I believe these US switches are based off the EU ones but I am at a loss on where to go from here as my latest attempt gives me an error in the logs.

error: 	z2m: Failed to load external converter file 'ZS-US1-WH-MS.js' (Cannot read properties of undefined (reading 'push'))
error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version

https://github.com/Koenkk/zigbee-herdsman-converters/blob/8a6cda40755e2899f7fdf875c54f0c3e9a6672af/src/devices/moes.ts#L187

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['ZTS-US_2gang'],
    model: 'ZS-US1-WH-MS',
    vendor: 'Moes',
    description: 'Wall touch light switch (2 gang)',
    extend: [],
    fromZigbee: [fz.ignore_basic_report, legacy.fz.tuya_switch, legacy.fz.moes_switch], // <-- added here all clusters reported from zigbee
    toZigbee: [legacy.tz.tuya_switch_state, legacy.tz.moes_switch], // <-- add here all clusters to send commands to zigbee
    exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
    e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
    e.enum('indicate_light', ea.STATE_SET, Object.values(legacy.moesSwitch.indicateLight))
        .withDescription('Indicator light status'),
    e.enum('power_on_behavior', ea.STATE_SET, Object.values(legacy.moesSwitch.powerOnBehavior))
        .withDescription('Controls the behavior when the device is powered on')],
    onEvent: tuya.onEventSetLocalTime,
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 1};
},
configure: async (device, coordinatorEndpoint) => {
    await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
    if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    // Reports itself as battery which is not correct: https://github.com/Koenkk/zigbee2mqtt/issues/6190
    device.powerSource = 'Mains (single phase)';
    device.save();
}
}
```

LaurentChardin, 2024-06-03:
Those definition will need a bit of loving and migration to `modernExtend`.
@Nerivec could you tag this as `converters` ?

KyleStilkey, 2024-09-07:
Looks like since then a update was pushed that provides support, not sure which version but decided to try and give this another go and see what I could do and it paired and works.

I'll close this out. Thanks!

closed 2024-09-07 by KyleStilkey