Koenkk/zigbee2mqtt #22910: Unabe to change Debounce or Optimistic settings since 1.38

DuncanRae, 2024-06-03:
### What happened?

Since the update to 1.38, I seem to be unable to change settings such as Optimistic or Debounce. Others may also not work, but these are the only ones I've tried. 

When selecting a new Debounce value or unchecking the Optimistic check box, it immediately reverts back to the previous setting. This definitely used to work, but now seems to be broken. 

I have 2 instances of Z2M running with different coordinators (one using Ember and one using EZSP) but they both exhibit the same behaviour. 

### What did you expect to happen?

Setting to be applied.

### How to reproduce it (minimal and precise)

Select the device (I've tried multiple types), go to Settings and try to change the Optimistic Check box setting or increment the Debounce value. It immediately reverts back to the previous setting. 

### Zigbee2MQTT version

1.38

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff Zigbee 3.0 USB E

### Setup

Home Assistant Add on

### Debug log

I can't see anything in the logs related to this at all.

drjjr2, 2024-08-16:
I am having this same problem. The UI won't allow me to set a debounce value.

starox, 2024-11-01:
This is the whole setting page not working, debuounce, retain, disable, etc ...
Update to zigbee2mqtt 1.41.0 doesn't change anything.
I am not a web guy, but if someone have a hint to start debugging it, I will try to do that.
I don't know what to search in firefox developer tools.

closed 2025-02-05 by github-actions[bot]