Koenkk/zigbee2mqtt #22911: Touchlink scan fails with errors

andrzejpolis, 2024-06-03:
### What happened?

When I scan for Touchlink devices I get errors and no devices are detected. Frontend logs contains a sequence of 
```
Info 2024-06-03 11:24:42`Set InterPAN channel to '20'`
Warning 2024-06-03 11:24:43`Scan request failed or was not answered: 'Error: {"target":null,"apsFrame":{"profileId":49246,"clusterId":4096,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":65535,"sequence":0},"zclSequence":0,"commandIdentifier":1} timed out after 500ms'`
```
with different channels. Same result for each.

I'm using new ember driver and Sonoff Plus V2 with firmware 7.4.3. Did not have a chance to check if it worked before with ezsp as did not have Touchlink remote back then.

### What did you expect to happen?

Device scan finished and Touchlink device successfully discovered.

### How to reproduce it (minimal and precise)

Just click Scan in Touchlink tab

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.4.3 [GA]

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231031205140-if00, ember ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl

### Setup

plain, LXE container, x86-64

### Debug log

[z2m.log](https://github.com/user-attachments/files/15532134/z2m.log)

knumskull, 2024-06-20:
The same issue applies to my environment.
I had a couple of devices integrated, used ZHA before. Some time is stopped working and I decided to switch to Zigbee2MQTT and ember `ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl` 

Zigbee is starting well, but no devices are found.

Melotron, 2024-07-04:
Im in the same boat, updated to the latest firmware and all my units are gone. I dont have a hue bridge.
All others can be re paired, but hue are a pain to get in pairing mode. If anyone have a easy way with out buying a hub please message me or should I count them as bricked?

Koenkk, 2024-07-06:
This error means the device (e.g. the bulb) didn't respond to the scan request, not something that can be solved from z2m. It may help to reconnect the bulb to power.

closed 2024-12-29 by github-actions[bot]