Koenkk/zigbee2mqtt #22913: Z2M Tuya radar pressence sensor not workig, gives Z2M error

jcffrt, 2024-06-03:
### What happened?

Supported Tuya radar oresence sensor
TuYa | ZY-M100-24G 
(tuya-zy-m100-24g) is recognised but does not work (no entity values appear, no change in values) and get continued error in Zigbee2 MQTT:
Exception while calling fromZigbee converter: Value '5' is not allowed, expected one of 0,1,2}

### What did you expect to happen?

The sensor to work normally, detect pressence and give distance values etc. No errors in Z2M.

### How to reproduce it (minimal and precise)

Just plug the device, nothing in particular

### Zigbee2MQTT version

 1.38.0 commit: fe048e6

### Adapter firmware version

20210708

### Adapter

Sonoff zStack3x0

### Setup

Containerized

### Debug log

[log.log](https://github.com/user-attachments/files/15535980/log.log)

uveral, 2024-06-11:
Same for me

eyeballic, 2024-06-13:
I also have the same problem

elterko, 2024-06-14:
Me too, updated to Z2M latest version which correctly recognizes the sensor but unfortunately settings are kaputt

Koenkk, 2024-06-15:
Could you provide the data/database.db entry of your device?

uveral, 2024-06-15:
> I also have the same problem

With z2m edge works for me

Koenkk, 2024-06-15:
Great, so a fix will be included in the 1 July release

closed 2024-06-15 by Koenkk