Koenkk/zigbee2mqtt #22914: Third Reality 3RTHS24BZ Temp and Humidity Shows as Mains Powered Instead of Battery

jmedemar, 2024-06-03:
### What happened?

The Third Reality 3RTHS24BZ Temp and Humidity Sensors show as Mains Powered Instead of Battery Powered. This may be causing the errors upon startup shown below, because zigbee2mqtt cannot connect to the sensors until they wake up. Once the sensors wake up, there are no more errors in the log.

[image]

`Z2M Server: Failed to configure 'Basement Bath Sensor', attempt 3 (Error: Bind 0x282c02bfffe3a768/1 genPowerCfg from '0xdc8e95fffe0b6126/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":102}) at EmberOneWaitress.deliveryFailedFor (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\ember\adapter\oneWaitress.ts:96:31) at EmberAdapter.onMessageSentDeliveryFailed (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\ember\adapter\emberAdapter.ts:540:30) at Ezsp.emit (node:events:513:28) at Ezsp.ezspMessageSentHandler (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\ember\ezsp\ezsp.ts:3960:18) at Ezsp.callbackDispatch (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\ember\ezsp\ezsp.ts:797:18) at Ezsp.tick (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\ember\ezsp\ezsp.ts:451:22) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))`

### What did you expect to happen?

The sensors should show as battery powered in the front end and that the log doesn't show errors connecting to these sensors before they wake up.

### How to reproduce it (minimal and precise)

Pair sensor. Reboot zigbee2mqtt, watch log for errors within a couple minutes.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.43

### Adapter

Sonoff Dongle-E

### Setup

Installed through Homeseer on Windows 11 Pro

Skirmisher, 2024-06-04:
Seeing this issue with the same device using ZBDongle-E with Z2M 1.38.0. It was previously reported correctly as battery-powered on 1.36.

marc-gist, 2024-06-04:
I'm also getting errors like:
```
Failed to configure 'thirdreality_temp_hum_sensor', attempt 1 (Error: Bind 0x282c02bfffe39964/1 genPowerCfg from '0x00124b0021cc4303/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:691:28 at processTicksAndRejections (node:internal/process/task_queues:95:5) at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Endpoint.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:464:13) at setupAttributes (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:75:13) at configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:112:17) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:172:21) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:121:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))
```
wonder if it is related.

jmedemar, 2024-12-02:
Still an issue. Anything else I can get to help someone take a look into this?

glassbase, 2024-12-10:
Just got 2 of these for BF. Seeing same issue.

triosniolin, 2024-12-13:
I have 5 of these and am having the same problem with them all...

LBTHR333, 2025-01-05:
I just bought 8 of these, same issue.

closed 2025-03-30 by github-actions[bot]

jmedemar, 2025-05-17:
This is still an issue.

jmedemar, 2025-05-17:
@chris-1243, I see you fixed this same issue with a different Third Reality sensor here #27210 . Any chance you could take a look at this one too?

chris-1243, 2025-05-17:
It should be fixed in the latest-dev or in the next official release when accepted and merged by the dev team

francoisp59, 2025-06-23:
On June 23rd I'm still seeing this issue in Home Assistant. Do we know if the solution has been pushed?

francoisp59, 2025-06-24:
@3reality-support As you suggested I've updated the firmware to v1.00.40 however it still showing as plugin power. In zigbee2mqtt I've tried to do a new interview but got no response. Do I need to remove it and re-add it?

[image]

3reality-support, 2025-06-24:
yes，remove it and re-add it

francoisp59, 2025-06-24:
> @francoisp59 <https://github.com/francoisp59> yes，remove it and re-add it

I've removed it, pushed the button on the side for 10 seconds and
zigbee2mqtt found it again but it's still showing as a plugin device.

francoisp59, 2025-06-25:
> @francoisp59 <https://github.com/francoisp59> First, delete the device,
> then restart z2m, and finally add the device(Please ensure that there are
> no external converters related to 3RTHS24BZ locally)

Thanks, that worked.