Koenkk/zigbee2mqtt #22916: Problem accessing https://www.zigbee2mqtt.io

viama, 2024-06-03:
### What happened?

For the last couple of days I've been unable to access the https://www.zigbee2mqtt.io website from my home broadband

### What did you expect to happen?

The website would load (and device pictures populate in my z2m instance)

### How to reproduce it (minimal and precise)

access the website in a bowser

### Zigbee2MQTT version

1.38

### Adapter firmware version

20230507

### Adapter

Sonoff P

### Setup

debian docker

wastez, 2024-06-03:
Dont know what you mean, it is working as it should.

viama, 2024-06-03:
> Dont know what you mean, it is working as it should. 

I agree. The site is up and I can access it fine on mobile or over a VPN. For some reason my ISP seems unable to access this site. It feels like a firewall block. I've changed IP addresses without luck too.

viama, 2024-06-03:
Looks like it's a broader BT wholesale issue with things hosted on GitHub.io reports from EE and Plusnet.

Closing this issue as clearly not a problem for here.

closed 2024-06-03 by viama

viama, 2024-06-04:
For the record BT were blocking all GitHub.io pages. Been fixed now