Koenkk/zigbee2mqtt #22923: 1.38.0 HA Addon - No Frontend

jerrm, 2024-06-03:
### What happened?

Using 1.38.0 HA Addon on HA 2024.5.5.

No devices, or any info for that matter is displayed in the front end.  All the menus are there, but no data.  Devices still work via HA, just Front End is missing.

Settings->Raw returns:
```
{
    "configSchema": {
        "properties": {},
        "required": []
    },
    "config": {
        "advanced": {}
    }
}
```

### What did you expect to happen?

Devices/Data to be there as in the past.

### How to reproduce it (minimal and precise)

Updated to 1.38.0 Addon via HA interface.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20211028

### Adapter

ZBDongle-P

### Setup

HA Addon

francisp2, 2024-06-04:
Try restarting the add-on

jerrm, 2024-06-04:
Already restarted multiple times, also have done a full system reboot.

niculw, 2024-06-09:
Experienced same behavior with a Dongle-E when updating to 1.38.0. Z2M frontend shows no devices, however devices are present in the "Raw" tab and available in HA

jerrm, 2024-06-24:
Found a fix.  

My devices.yaml had an `icon:` entry for each device (the data:image line below is truncated for brevity).:
```
'0xb0ce1814036baacf':
  friendly_name: Plug Utility Room
  icon: >-
    data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAgAA...
```

I removed all the `icon:` sections from devices.yaml with a quick grep command, and everything started working again.

closed 2024-12-29 by github-actions[bot]