Koenkk/zigbee2mqtt #22927: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyACM0'

mauhler, 2024-06-04:
### What happened?

If I try start the zigbee2mqtt via 

root@zigbee2mqtt:/opt/zigbee2mqtt# npm start

the following error message appears:

[2024-06-04 11:33:05] info:     z2m: Logging to console, file (filename: log.txt)
[2024-06-04 11:33:05] debug:    z2m: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
[2024-06-04 11:33:05] info:     z2m: Starting Zigbee2MQTT version 1.37.1 (commit #cda867a)
[2024-06-04 11:33:05] info:     z2m: Starting zigbee-herdsman (0.46.6)
[2024-06-04 11:33:05] debug:    z2m: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":true},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
[2024-06-04 11:33:05] debug:    zh:deconz:driver: Set delay to 0
[2024-06-04 11:33:05] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":true,"concurrent":null,"delay":null}}'
[2024-06-04 11:33:05] debug:    zh:deconz:driver: Opening with /dev/ttyACM0
[2024-06-04 11:33:05] error:    z2m: Error while starting zigbee-herdsman
[2024-06-04 11:33:05] error:    z2m: Failed to start zigbee
[2024-06-04 11:33:05] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-06-04 11:33:05] error:    z2m: Exiting...
[2024-06-04 11:33:05] error:    z2m: Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyACM0'
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:202:28)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/dist/index.js:82:22)
    at /opt/zigbee2mqtt/node_modules/@serialport/stream/dist/index.js:118:18

I have checked everything from the help Zigbee2MQTT fails to start and everything fits. 

Can anyone help me?

### What did you expect to happen?

Start of the Zigbee2MQTT and access to the frontend

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

26780700

### Adapter

Deconz Conebee II

### Setup

LXC on Proxmox on Intel NUC

LamerTex, 2024-06-04:
Did you try to run the command with root access?

I had a similar problem (not sure if exactly the same) with "npm start", but it was working correctly with "sudo npm start".

If this works then probably the correct solution is to give access to the serial port to the current user, but I'm not an expert in Linux permissions so I can't help you very much in that case.

mauhl0194, 2024-06-04:
> Did you try to run the command with root access?
>
> I had a similar problem (not sure if exactly the same) with "npm start", but it was working correctly with "sudo npm start".
>
> If this works then [...]

That was not the problem. My user has root access...

LamerTex, 2024-06-04:
And does the command ```ls -l /dev/ttyACM0``` return the right info [as displayed in the installation instruction](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#determine-location-of-the-adapter-and-checking-user-permissions)?

If that command works correctly then I don't know how to help you, let's see if someone else can.

mauhl0194, 2024-06-04:
> And does the command `ls -l /dev/ttyACM0` return the right info [as displayed in the installation [...]

Yes, I get the right info as seen below:

root@proxmox:~# ls -l /dev/ttyACM0
crw-rw-rw- 1 root dialout 166, 0 Jun  4 13:39 /dev/ttyACM0

closed 2024-12-21 by github-actions[bot]