Koenkk/zigbee2mqtt #22928: Device works correctly using "ezsp" driver but not with "ember" driver (EGLO 99099 with ZBDongle-E)

LamerTex, 2024-06-04:
### What happened?

I have an [EGLO 99099](https://www.zigbee2mqtt.io/devices/99099.html) remote controller which is correctly recognized and paired using both "ezsp" and "ember" driver, but after the interview the device works correctly with the "ezsp" driver but not with the "ember" driver. In the latter case the "State" remain empty and the "Exposes" remain N/A.
[image]

Even in the "broken" state when I press a button of the remote in the log the press is displayed:
```
[2024-06-04 11:20:52] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-04 11:20:52] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4]
[2024-06-04 11:20:52] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=7] Added to rxQueue
[2024-06-04 11:20:52] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0]
[2024-06-04 11:20:52] debug:    zh:ember:ezsp: <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=51 Len=32]
[2024-06-04 11:20:52] debug:    zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [type=BROADCAST], [indexOrDestination=65532], [apsFrame={"profileId":0,"clusterId":31,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":0}], [messageTag=255], [status=SUCCESS], [messageContents=810162ba2f5fb238c1a4]
``` 

### What did you expect to happen?

The device should work correctly with both or, at most, just with the newer driver.

### How to reproduce it (minimal and precise)

Just pair the device and see if it works correctly.

Even after pairing, without changing anything else, I can simply stop Z2M, change the "adapter:" setting from "ember" to "ezsp", restart Z2M and the remote will work correctly. 

If I do the opposite, stop Z2M, change the "adapter:" setting from "ezsp" to "ember", restart Z2M, the remote will go back to the "broken" state. 

### Zigbee2MQTT version

1.38.0 commit: 0e6da85d

### Adapter firmware version

7.4.2.0 build 0

### Adapter

ZBDongle-E

### Setup

Plain on x64 Windows laptop 

### Debug log

This is the log from an empty state, interview and pairing of the device and press of RED, GREEN, BLUE buttons one after the other after the pairing was successful. Done with "ember" and with "ezsp" driver.
[Ember_Interview+RGBButtons_log.log](https://github.com/user-attachments/files/15546296/Ember_Interview%2BRGBButtons_log.log)
[EZSP_Interview+RGBButtons_log.log](https://github.com/user-attachments/files/15546324/EZSP_Interview%2BRGBButtons_log.log)

olyckne, 2024-06-04:
Just noticed something similar. I've had it set to ember for quite some time but after updating to 1.38 suddenly all my non-battery devices stopped responding. Changed it to ezsp and everything works again.

LamerTex, 2024-06-05:
> Just noticed something similar. I've had it set to ember for quite some time but after updating to 1.38 suddenly all my non-battery devices stopped responding. Changed it to ezsp and everything works again.

This seems the opposite for me, the device that doesn't work with _ember_ ([EGLO 99099](https://www.zigbee2mqtt.io/devices/99099.html)) is the only battery-powered device I have, the others that work with _ember_ are not with battery, [Frient IOMZB-110](https://www.zigbee2mqtt.io/devices/IOMZB-110.html)  and [Tuya TS011F](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html)

adriangalbincea, 2024-06-06:
> Just noticed something similar. I've had it set to ember for quite some time but after updating to 1.38 suddenly all my non-battery devices stopped responding. I changed it to ezsp and everything works again.

I have similar issue. I changed the driver to ember and my IKEA VINDSTYRKA started to throw errors and not report anymore, also OTA check for updates does not work either. Once I change back to ezsp, everything is back to normal. I have a PIR sensor with a battery, and that one works fine with both drivers.

senna1992, 2024-06-09:
my sunricher / iluminize battery powered light switches also stopped working.
other sensors are working (aqara mostly).
smar tplugs also seem to do fine.

sbemmann, 2024-06-09:
I have also a similar problem. The Paulmann switch 501.34 has no action on ember (null), on ezsp it runs.

Ra72xx, 2024-06-10:
I'll link in my error report: https://github.com/Koenkk/zigbee2mqtt/issues/22249
For my it also looks like battery powered devices still report values, but mains powered devices don't work, or work rather slowly.

Nerivec, 2024-06-14:
For anyone where the problem appeared only after 1.38, there were no change in `ember` in 1.38 that would explain this (practically no change at all), so I'm pretty sure the `ezsp` switch that "fixes the problem", is more about the switch itself, than `ezsp` (possible partial network rebuild, reboot of the adapter, etc).

@LamerTex I've identified the issue with the EGLO (it's using multicast -group- for state updates...). I'll push a fix.
@sbemmann I'm assuming it's the same issue as the EGLO. Can you provide a debug log of the switch being triggered on `ember` or `ezsp`, so I can confirm?

LamerTex, 2024-06-18:
I confirm that this problem is fixed for me in the dev branch, I'll leave this open until @sbemmann can confirm if it's fixed also for him. 
If it is fixed also for him I will close the issue.

closed 2024-07-01 by Nerivec

adriangalbincea, 2024-07-17:
I can confirm that it works well.