Koenkk/zigbee2mqtt #22931: State only updates when manually refreshed on large network

mortenmoulder, 2024-06-04:
### What happened?

At my work we have a 140+ device network with 50/50 split between switches and power plugs. One plug is bound (via OnOff binding) to one power plug. The power plug is used as a state for various things in the office, so getting the proper state is actually important to us.

We have experienced a few of the plugs do not update their on/off state, unless we press the refresh button. When changed via Home Assistant or MQTT it updates the state perfectly. If I publish an MQTT message to `zigbee2mqtt/FRIENDLY_NAME/get` with payload `{"state": ""}` the state also updates (a hack we could implement, but would rather not).

We cannot reproduce the issue consistently. We have not updated to the latest Ember firmware - yet.

I realise this is a weird issue, seeing that 99% of the plugs report the state correctly, which also means debugging is hard. Let me know if you want me to test things out.

Plug: https://www.zigbee2mqtt.io/devices/SPLZB-141.html
Switch: https://www.zigbee2mqtt.io/devices/SBTZB-110.html

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle-E

### Setup

Docker on x86

LaurentChardin, 2024-06-04:
`One plug is bound (via OnOff binding) to one power plug`

Do you mean: One *switch* is bound (via OnOff binding) to one power plug ?

mortenmoulder, 2024-06-04:
> Do you mean: One _switch_ is bound (via OnOff binding) to one power plug ?

Yes, one switch is bound to one plug. Typo.

LaurentChardin, 2024-06-04:
`We have experienced a few of the plugs do not update their on/off state`

So what you mean is:
- the switch properly control the plug (on and off)
- But the state is not reflected properly in Z2M (therefore the UI is incorrect, as well no MQTT event was received)

Is that correct ?

mortenmoulder, 2024-06-04:
@LaurentChardin Yes, that is completely correct.

LaurentChardin, 2024-06-04:
Ok ... since you have defined a binding between each switch and plug, it means they can work directly together, aka without even the coordinator on the mesh network. So this is fine.

Now, it could be that the plug is failing to have its notification status event reaching the coordinator: did you check the status of your mesh if you have *holes* with low LQI ? there might be some routing issues that prevent the notification event to reach the coordinator. From your description, i would try to have an idea of the mesh network sanity. 

With that many routers, it should help to have a quite stable network, but you never know. it could explain the *unpredictability* of this behavior.

mortenmoulder, 2024-06-04:
@LaurentChardin Yes, I think the plug with the lowest LQI is about 150, so it should all be good. But that was my main concern too - that some other router in the path somehow won't report this plug's state correctly. Very hard to actually debug.

LaurentChardin, 2024-06-04:
@mortenmoulder It looks like your issue is really Zigbee-ish, meaning I don't think Z2M can actually help much here.

https://community.home-assistant.io/t/zigbee-networks-how-to-guide-for-avoiding-interference-optimize-using-zigbee-router-devices-repeaters-extenders-to-get-a-stable-network-with-best-possible-range-and-coverage/515752

There could be some actions you could check:
- try to identify and remember the plugs that are failing to report
- check if they are nearby a source of interference, which would make them break from the zigbee network for time to time
- Look for errors that could have been reported in the logs within a large timeframe (multiple days, weeks) : use debug level during that time 

You have a lot of routers (plugs) but in the end, this can be quite dependent of your physical setup.

mortenmoulder, 2024-06-04:
@LaurentChardin I'm not really sure either. It could be Zigbee but it could also be Zigbee2MQTT.

We save all logs and can view them through Grafana, so it's quite easy to filter out the regular updates and focus on the odd ones. Nothing looks suspicious, unfortunately.

Our physical setup is somewhat unique and funny. The plugs control the power going to each employee's desk. Each room/area has four desks, and in the middle underneath the floor, there are four outlets with one plug in each. The plug we are currently monitoring is relatively close to the coordinator, however, it is most likely routed through a bunch of other plugs - of which each could be the culprit. Or not?

I would hate to implement the `send MQTT message to /get with an empty state to refresh the state` hack, but if that's what it takes.. It just sucks we have to do that to 70+ plugs and "overload" the network even more.

Hopefully someone stumbles upon this issue and has a fix 🙏

LaurentChardin, 2024-06-04:
@mortenmoulder  Do your plugs implement as well `haElectricalMeasurement` based reportings ? like `ActivePower`, `RmsCurrent` etc ?

Just asking because, if you have a lot of them, and some devices are very *very* chatty : they could somehow create pressure on your network because we would get flooded with messages, bouncing around with your routers... trying to get a route for each of them.

mortenmoulder, 2024-06-04:
> Do your plugs implement as well `haElectricalMeasurement` based reportings ? like `ActivePower`, `RmsCurrent` etc ?

They sure do: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/develco.ts#L522-L545

> Just asking because, if you have a lot of them, and some devices are very _very_ chatty : they could somehow create pressure on your network because we would get flooded with messages, bouncing around with your routers... trying to get a route for each of them.

I would say we get around 10 reports per second - if not more. The log is unreadable and the Zigbee2MQTT UI is barely working.

> Since you are in a working area, i guess you have some strong Wifi needs on the same 2.4Ghz band. You might want to be sure your Zigbee channel is not too close from the usual Wifi channel.

We actually don't have a lot of 2.4 GHz WiFi coverage. I believe we have an IoT network but primarily 5 or 6 GHz.

mortenmoulder, 2024-06-06:
Update: EZSP v13 (Ember) update and Zigbee2MQTT 1.38.0 did not work.

Lmax12, 2024-09-26:
I'm having simular issues with a switch that can be operated with a manual button. The state is also not updated. Only when i press the refresh button it shows. Its also a fairly large network with 138 devices, of which 103 are routers.
Zigbee2mqtt version is 1.40.1. Coordinator verions is 20240316 on a UZG-01 (CC2652P7 chipset)

When i move the switch to another Z2M instance i have running, the state is updated on the fly.

It used to work fine, but stopped working a week or 3 ago. Did you find any solution?

Lmax12, 2024-09-26:
I found a solution/work-around on another site. Posting it here to help others:

I changed the "Min Rep Change" for the "OnOff" attribute in the device settings page from "0" to "1".

closed 2024-12-29 by github-actions[bot]

RezzZ, 2025-03-09:
> [@LaurentChardin](https://github.com/LaurentChardin) I'm not really sure either. It could be Zigbee but it could also be Zigbee2MQTT.
>
> We save all logs and can view them through Grafana, so it's [...]

I experience the same with most of my sonoff zbmini switches (01MINIZB). Often they don't register going off and I have to manually refresh the state in Z2M. I guess I'll create an automation in HomeAssitant to refresh the state when they are on for a specific time to refresh the state over mqtt. Can't figure out why specificly these devices fail to report the off state properly. They are directly linked to my sonoff controller and have a good signal quality. They also turn offline at some point but even then refreshing the state wakes them up again :(

palini762, 2025-08-22:
Got the same issue with a Aqara QBKG41LM smart switch. Very weird, as LQI is excellent and pushing the state from HA to the device is flawless and instantaneous.  Currently updating the firmware and will report back, if it worked.