Koenkk/zigbee2mqtt #22937: [New device support]: Support for Signify Netherlands B.V LWF004

mixedbreed, 2024-06-04:
### Link

https://www.philips-hue.com/en-gb/p/hue-white-a60---e27-smart-bulb---800/8719514329843#overview

### Database entry

{"id":38,"type":"Router","ieeeAddr":"0x001788010cd01fcf","nwkAddr":26754,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LWF004","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64515,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWF004","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20211210","swBuildId":"1.93.7"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00257ccc68","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20211210","swBuildId":"1.93.7","zclVersion":2,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1717521892607}

### Comments

Hello, would like to add device Signify Netherlands B.V LWF004 to Z2M.
Please see the attached external definition.

BR
Mixedbreed

### External definition

```shell
const {identify, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['LWF004'],
    model: 'LWF004',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [identify(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-06-05:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-06-05 by Koenkk

mixedbreed, 2024-06-08:
> Added!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Thanks for the quick reply and fix. 
Devices is now working with full functionality.
Have a nice weekend!