Koenkk/zigbee2mqtt #22939: New devices don't work after update

SaymonPhoenix, 2024-06-04:
### What happened?

Hello. My ceiling human presence sensor did not work. After the update it became supported, thank you for that. But:
1. It is displayed as a TuYa ZY-M100-24G model, although in fact it is ZY-M100-S_1 (also 24G)
2. In fact it doesn't work and I keep getting an error "Exception while calling fromZigbee converter: Value '5' is not allowed, expected one of 0,1,2}"

### Zigbee2MQTT version

1.38.0-1

### Adapter firmware version

20240501.1

### Adapter

EZSP

### Setup

Docker Container

uveral, 2024-06-12:
same for me

wwyyrrmm, 2024-06-29:
same to me

closed 2024-12-29 by github-actions[bot]