Koenkk/zigbee2mqtt #22941: New devices dont show up in entities in Home assistent

PSfreak, 2024-06-04:
### What happened?

After i upgraded to ember (firmware 7.4.2.0)
After that i can add new devices to Zigbee2Mqqt, but they are not showing up in entities in Home assistent

### What did you expect to happen?

New devices should show up in entities in Home assistent

### How to reproduce it (minimal and precise)

Adding a new divice after upgrade

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff ZigBee Gateway ZigBee 3.0 USB Dongle Plus-E

### Setup

VM on proxmox

wastez, 2024-06-04:
Sounds like problems with your mqtt server, that’s the only possibility if it is available in z2m and not in home assistant.

PSfreak, 2024-06-05:
Problem solved by downgrading to version 1,37

closed 2024-12-21 by github-actions[bot]