Koenkk/zigbee2mqtt #22945: Fails to start after upgrade to 1.38.0

nlindblo, 2024-06-05:
### What happened?

I upgraded to 1.38.0 from 1.37.0 by running update.sh and Z2M fails to start. I managed to revert back to 1.37.1 and im now back up, but if someone could take a look at the debug log and see if it gives some hints on what may be causing this, before I make another attempt

### What did you expect to happen?

Upgrade to be successful

### How to reproduce it (minimal and precise)

Seem to be an isolated problem as I've seen no other reports.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator

### Setup

PI 4 -> Ubuntu 22.04

### Debug log

[2024-06-04 17:54:00] info: 	z2m: Logging to console, file (filename: log.log)
[2024-06-04 17:54:00] info: 	z2m: Starting Zigbee2MQTT version 1.38.0 (commit #0e6da85d)
[2024-06-04 17:54:00] info: 	z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-04 17:54:03] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-06-04 17:54:03] info: 	zh:zstack:znp: Serialport opened
[2024-06-04 17:54:04] error: 	z2m: Error while starting zigbee-herdsman
[2024-06-04 17:54:04] error: 	z2m: Failed to start zigbee
[2024-06-04 17:54:04] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-06-04 17:54:04] error: 	z2m: Exiting...
[2024-06-04 17:54:04] error: 	z2m: Error: SREQ '--> ZDO - extAddGroup - {"endpoint":242,"groupid":2948,"namelen":0,"groupname":[]}' failed with status '(0xb2: APS_TABLE_FULL)' (expected '(0x00: SUCCESS)')
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:313:27
    at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at ZnpAdapterManager.addToGroup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:450:13)
    at ZnpAdapterManager.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:97:9)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:62:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:109:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)

gitaday, 2024-06-06:
@nlindblo , the issue was "only" that Z2M did not start anymore ? Or you also experienced a crash and a restart of the whole OS / Pi (as it happens in #22925) ?

nlindblo, 2024-06-06:
@gitaday  Only Z2M, OS intact, above is the log output I could find. I'm basically looking for a hint to what I can troubleshoot before I make another attempt

smint85, 2024-06-09:
Same issue. Any fix?

Koenkk, 2024-06-09:
Could you try updating your coordinator fw?

Koenkk, 2024-06-15:
Yes, it looks related to the coordinator at first glance

sbuchanYVR, 2024-06-18:
rpi 5, HA current prod build, 

SONOFF ZigBee 3.0 USB Dongle Plus, EFR32MG21 Coordinator, flashed with Zigbee coordinator firmware based on EZNet 6.10.3.  There is no later firmware available. 

- FAILS - current dev branch (edge) with "zigbee-herdsman-converters": "19.57.0", AND .. 
- FAILS same issue in stable build (1.38.0-1).
- FAILS while not apparently required when running on a rp5, Hass Os Built, also tried specificing DEVNAME: /dev/ttyACM0

What else can I do?   Really need this up and running. 
--
[17:56:08] INFO: Preparing to start...
[17:56:08] INFO: Socat not enabled
[17:56:09] INFO: Starting Zigbee2MQTT...
[2024-06-17 17:56:11] info: 	z2m: Logging to console, file (filename: log.log)
[2024-06-17 17:56:11] info: 	z2m: Starting Zigbee2MQTT version 1.38.0-dev (commit #a85d3e4)
[2024-06-17 17:56:11] info: 	z2m: Starting zigbee-herdsman (0.50.0)
[2024-06-17 17:56:11] **error: 	z2m: Error while starting zigbee-herdsma**n
[2024-06-17 17:56:11] **error: 	z2m: Failed to start zigbee**
[2024-06-17 17:56:11] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-06-17 17:56:11] error: 	z2m: Exiting...
[2024-06-17 17:56:11] error: 	z2m: Error: spawn udevadm ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:284:19)
    at onErrorNT (node:internal/child_process:477:16)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)
--

[image]

closed 2024-12-29 by github-actions[bot]