Koenkk/zigbee2mqtt #22948: HEIMAN HS2WD-E sometimes cannot stop siren

cecc83, 2024-06-05:
### What happened?

When i send mqtt message to stop alarm sometime i need to send many messages to effective stop the sound

### What did you expect to happen?

I expect that with only one message the siren stop sound.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff USB Dongle-E

### Setup

Docker container

### Debug log

[heiman-log.txt](https://github.com/user-attachments/files/15580293/heiman-log.txt)

Nerivec, 2024-06-09:
Looks like the delivery of the message failed on that first stop.
Can you provide a `debug` log for each scenario, on 220v and off 220v?

cecc83, 2024-06-11:
The day before yesterday I redid the pairing of the siren closer to the ZigBee dongle... and now everything seems to work!!

closed 2024-06-11 by cecc83