Koenkk/zigbee2mqtt #22950: [New device support]: no name Zigbee Water Valve Timer (model: 'TS0049', vendor: '_TZ3000_kz1anoi8')

tomj44300, 2024-06-05:
### Link

https://fr.aliexpress.com/item/1005006859734151.html?spm=a2g0o.order_list.order_list_main.51.448a5e5bATkN6j&gatewayAdapt=glo2fra

### Database entry

{"id":31,"type":"EndDevice","ieeeAddr":"0xa4c138a6269055ea","nwkAddr":35579,"manufId":4417,"manufName":"_TZ3000_kz1anoi8","powerSource":"Battery","modelId":"TS0049","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[3,4,5,1,6,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�+�-\u0012L`�-\u0012","65506":55,"65508":0,"stackVersion":0,"dateCode":"","appVersion":72}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":184,"batteryVoltage":29}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2c822","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2c822","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1460692585},"lastSeen":1717605244700}

### Comments

HI
it's a yellow copy of MOES water valve https://github.com/Koenkk/zigbee2mqtt/issues/21788 , I have already tried external_converters.
I m running zigbee2mqtt edge 1.38 

action on/off from valve is seen on z2m , from z2m action on works but i can't set off 
can someone help me please
Thomas
 

### External definition

```shell
const {identify, battery, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
zigbeeModel: ['TS0049'],
model: 'TS0049',
vendor: '_TZ3000_kz1anoi8',
description: 'Automatically generated definition',
extend: [identify(), battery(), onOff({"powerOnBehavior":false})],
meta: {},
};

module.exports = definition;
```

YellowMonster76, 2024-06-11:
I have the same device, I have the following in log:
`[2024-06-11 10:43:48] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":57345,"srcaddr":29907,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":174,"securityuse":0,"timestamp":10409573,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[8,101,10,64,208,72,6,11,0,0,2,88]}}
[2024-06-11 10:43:48] debug: 	zh:controller: Failed to parse frame: Error: Read for '6' not available
[2024-06-11 10:43:48] debug: 	zh:controller: Received payload: clusterID=57345, address=29907, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=174, frame=undefined
[2024-06-11 10:43:48] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-11 10:43:48] debug: 	z2m: Received Zigbee message from 'Water Plants', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,101,10,64,208,72,6,11,0,0,2,88],"type":"Buffer"}' from endpoint 1 with groupID 0
[2024-06-11 10:43:48] debug: 	z2m: No converter available for 'TS0049' with cluster 'manuSpecificTuya_3' and type 'raw' and data '{"data":[8,101,10,64,208,72,6,11,0,0,2,88],"type":"Buffer"}'
[2024-06-11 10:43:54] debug: 	zh:zstack:unpi:parser: <-- [254,29,68,129,0,0,1,224,211,116,1,1,0,182,0,117,248,164,0,0,9,8,102,10,64,208,72,3,12,1,211,116,29,0]
[2024-06-11 10:43:54] debug: 	zh:zstack:unpi:parser: --- parseNext [254,29,68,129,0,0,1,224,211,116,1,1,0,182,0,117,248,164,0,0,9,8,102,10,64,208,72,3,12,1,211,116,29,0]
[2024-06-11 10:43:54] debug: 	zh:zstack:unpi:parser: --> parsed 29 - 2 - 4 - 129 - [0,0,1,224,211,116,1,1,0,182,0,117,248,164,0,0,9,8,102,10,64,208,72,3,12,1,211,116,29] - 0
[2024-06-11 10:43:54] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":57345,"srcaddr":29907,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":182,"securityuse":0,"timestamp":10811509,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[8,102,10,64,208,72,3,12,1]}}
[2024-06-11 10:43:54] debug: 	zh:controller: Failed to parse frame: Error: Read for '3' not available
[2024-06-11 10:43:54] debug: 	zh:controller: Received payload: clusterID=57345, address=29907, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=182, frame=undefined
[2024-06-11 10:43:54] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-11 10:43:54] debug: 	z2m: Received Zigbee message from 'Water Plants', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,102,10,64,208,72,3,12,1],"type":"Buffer"}' from endpoint 1 with groupID 0
[2024-06-11 10:43:54] debug: 	z2m: No converter available for 'TS0049' with cluster 'manuSpecificTuya_3' and type 'raw' and data '{"data":[8,102,10,64,208,72,3,12,1],"type":"Buffer"}'`

PoloG12, 2024-06-24:
Same problem here, with this model. I bought 2 items. I tried a custom configuration but it still with an on/off issue. And it doesn't return the battery status too.
[image]
Any help is very welcome!

fabien53, 2024-06-26:
Hi, same problem here, seems work with zha, but not with z2m. :-(

PoloG12, 2024-06-30:
It works!
In the file
/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/moes.js
I replace the line 
fingerprint: tuya.fingerprint('TS0049', ['_TZ3000_cjfmu5he']),
by
fingerprint: tuya.fingerprint('TS0049', ['_TZ3000_kz1anoi8']),
and it's ok (on/off/battery)

Koenkk, 2024-07-01:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-01 by Koenkk

tomj44300, 2024-07-02:
@Koenkk @PoloG12  hi just come back at home from vacation, when I have written my first message last month,I have already tried this easy fix (just change fingerprint name).
Just retry modification fix in moes.js file from GitHub.
Remove device and re-add 
Now battery is back , but i'm still not able set off valve from z2m.
Sorry not fixed mean not closed 😞
It's sure sure everything works with Zha @fabien53 ?

fabien53, 2024-07-02:
> @Koenkk @PoloG12 hi just come back at home from vacation, when I have written my first message last month,I have already tried this easy fix (just change fingerprint name). Just retry modification [...]

Yes on off work with zha but not battery state.

tomj44300, 2024-07-03:
hi just test zha same prob can't set off from HA
[image]

YellowMonster76, 2024-07-04:
Working for me after deleting and pairing again using dev branch. On/off working aswell as battery reporting all good!