Koenkk/zigbee2mqtt #22955: external_converters must be array

Nailik, 2024-06-05:
### What happened?

When clicking "submit" button on the External Converters page the error:
`external_converters must be array`
and
`Request 'zigbee2mqtt/bridge/request/options' failed with error: 'external_converters must be array'`

### What did you expect to happen?

List to be safed - no error

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

-

### Adapter

-

### Setup

Home Assistant AddOn 1.38.0-1

Nailik, 2024-06-05:
Seems to be fixed in `EDGE` version

sandello-alkr, 2024-06-22:
@Nailik could you elaborate a bit? What is `EDGE` version and how did you switched to that?

Nailik, 2024-06-24:
@sandello-alkr It depends using home assistant when adding the z2mqtt repo you also see the edge version and can just use it instead of the stable one.
More information can be found here https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html