Koenkk/zigbee2mqtt #22957: Failed to add HUE Smart Button and Outdoor Motion sensor after upgrade to EMBER

esusxunil, 2024-06-06:
### What happened?

I upgraded to EMBER Firmware, my SLZB-06M is running on Firmware 20240510.

Unfortunately I had to re-pair all devices - also my Backup didn't worked. I managed to get almost all devices (HUE Lamps, Aqara Sensors) back online. Just the HUE Motion Sensors, Dimmer and Buttons made issues.

With a hit to delete all entries for this devices in the configuration.yaml, database.db and state.json I was able to bring the indoor Motion Sensors and the Dimmer Switch V2 back online.

Die HUE Smart Buttons and the HUE Outdoor Motion Sensors are not interviewed successfully.

```
Info 2024-06-06 11:34:06Starting interview of '0x001788010cdb896b'
Info 2024-06-06 11:34:06MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010cdb896b","ieee_address":"0x001788010cdb896b","status":"started"},"type":"device_interview"}'
Info 2024-06-06 11:34:18Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "26414".
Info 2024-06-06 11:34:35MQTT publish: topic 'zigbee2mqtt/0x001788010cdb896b', payload '{"last_seen":"2024-06-06T11:34:35+02:00"}'
Error 2024-06-06 11:34:35Interview failed for '0x001788010cdb896b with error 'Error: Interview failed because can not get node descriptor ('0x001788010cdb896b')'
Error 2024-06-06 11:34:35Failed to interview '0x001788010cdb896b', device has not successfully been paired
```

__

### What did you expect to happen?

I expect a successfull Interview, with EZSP the devices worked perfectly

### How to reproduce it (minimal and precise)

Try to interview a HUE Smart Button or HUE Outdoor Motion Sensor with Ember firmware

### Zigbee2MQTT version

1.38.0-dev

### Adapter firmware version

20240510

### Adapter

SLZB-06M

### Setup

Docker Container

### Debug log

Info 2024-06-06 11:57:00z2m: Zigbee: allowing new devices to join.
Info 2024-06-06 11:57:00zh:ember: [STACK STATUS] Network opened.
Info 2024-06-06 11:57:00z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"3qchs-1"}'
Info 2024-06-06 11:57:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x001788010cdcbc3f', payload '{"last_seen":"2024-06-06T11:57:10+02:00"}'
Info 2024-06-06 11:57:11zh:controller: Interview for '0x001788010cdcbc3f' started
Info 2024-06-06 11:57:11z2m: Starting interview of '0x001788010cdcbc3f'
Info 2024-06-06 11:57:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010cdcbc3f","ieee_address":"0x001788010cdcbc3f","status":"started"},"type":"device_interview"}'
Info 2024-06-06 11:57:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x001788010cdcbc3f', payload '{"last_seen":"2024-06-06T11:57:10+02:00"}'
Info 2024-06-06 11:57:23zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "3832".
Info 2024-06-06 11:57:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ Stecker DG Schlafzimmer', payload '{"auto_off":false,"consumer_connected":true,"consumption":0.012930933386087418,"current":0,"device_temperature":29,"energy":0.01,"last_seen":"2024-06-06T11:57:22+02:00","led_disabled_night":true,"linkquality":172,"power":0.73,"power_outage_count":1,"power_outage_memory":true,"state":"ON","voltage":225}'
Info 2024-06-06 11:57:24zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "3832".
Info 2024-06-06 11:57:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x001788010cdcbc3f', payload '{"last_seen":"2024-06-06T11:57:28+02:00"}'
Error 2024-06-06 11:57:40zh:controller: Interview failed for '0x001788010cdcbc3f with error 'Error: Interview failed because can not get node descriptor ('0x001788010cdcbc3f')'
Error 2024-06-06 11:57:40z2m: Failed to interview '0x001788010cdcbc3f', device has not successfully been paired
Info 2024-06-06 11:57:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010cdcbc3f","ieee_address":"0x001788010cdcbc3f","status":"failed"},"type":"device_interview"}'

Nerivec, 2024-06-09:
Did you try keeping the device awake during interview by pushing a button on the device at regular 3-4sec intervals until successful?

esusxunil, 2024-06-11:
I finally was able to get a successful interview for the Outdoor Motion Sensor.

But the two Smart Buttons still failed. I press the Reset Button at the back (only short as in the description), the light is blinking orange and I press frequently the button itself to keep him alive. The interview always starts but then I get the error message about the failed interview.

I also tried removing batteries first. With the old driver they worked. 

I read at the device page of other Hue Battery devices that it helped to pair it to another Zigbee bridge on another channel (like a HUE Bridge) - I will give it a try soon.

Nerivec, 2024-08-20:
Can you confirm this issue is still present with latest dev/edge (or after September release)?

closed 2024-12-29 by github-actions[bot]