Koenkk/zigbee2mqtt #22961: Error: ZCL command...

chipsa1982, 2024-06-06:
### What happened?

Paired a new zigbee bulb as usual, pairing works fine and finishes, but when I try to control the bulb (turn on/off, change brightness etc), there is no reaction, after some time I get this error

Publish 'set' 'state' to '0x30fb10fffedd6ec4' failed: 'Error: ZCL command 0x30fb10fffedd6ec4/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 5298 - 1 - 65 - 6 - 11 after 10000ms)'

All other zigbee devices on the network, that were paired previously, work long term as expected with no issues. This affects only newly added devices. Tried multiple devices and vendors (Ikea GU10 and E27 bulbs, Hornbach GU10, Miboxer LED controller...). Same error every time.

### What did you expect to happen?

I expected to be able to control the paired device

### How to reproduce it (minimal and precise)

Pair device
Wait for pairing to finish
Go to Exposes tab and try to control the device

### Zigbee2MQTT version

1.38.0-1

### Adapter firmware version

20220219

### Adapter

Zigstar LAN Gateway

### Setup

Addon in Home Assistant, HASS running in a proxmox VM

wesselah, 2024-06-08:
Having excactly the same errors also with ikea bulb, TRADFRI bulb GU10 WS 345lm
When having this error muliple times the docker containers freezes and needs a restart. This worked before flawless are running this image. 1.37.1 Since it was published and even before these bulbs worked good without issues

GCRVL, 2024-06-10:
Same with Home Assistant Yellow and Ikea bulbs.

thenoid, 2024-06-12:
Getting similar errors with, previously working, enbrighten outlets.

```zigbee2mqtt     | [2024-06-12 14:19:34] error:  z2m: Publish 'set' 'state' to 'Zigbee Near Fountain' failed: 'Error: ZCL command 0x84fd27fffebe97c1/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":123})'
zigbee2mqtt     | [2024-06-12 14:19:59] error:  z2m: Publish 'set' 'state' to 'Backyard Wifi Zigbee' failed: 'Error: ZCL command 0x84fd27fffeba278d/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerC
ode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint"
:1,"options":4416,"groupId":0,"sequence":124})'
```

nicklutt, 2024-06-23:
I had similar errors in my log, I could receive updates from my devices , but not publish any state. My fix: flash my coordinator again with the same firmware. (CC2531)

array81, 2024-06-28:
I have the same error.
I flashed the last firmware ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl  on my Sonoff ZBDongle-E but I get the same problem. When I try to set value I get "Error: ZCL command..."

Any solution?

ShammanIT, 2024-06-30:
I have the same problem. Adapters - Sonoff P and Zstar with last firmware both. Devices which have a problem it's lumi Aqara bulbs 12, tuya reley and perenio open/close.

Arakon, 2024-06-30:
I have the same issue on most of my devices. I updated to the latest z-stack firmware and it didn't make any difference. I tried to re-pair my ceiling light and it dropped off the network and now can't pair at all anymore. Ikea Tradfri router also won't connect. Oddly, various switches (also Tradfri) still connect and respond.

Also getting lots and lots of ping errors.

samuele2723, 2024-07-05:
Followuing

Koenkk, 2024-07-07:
`Timeout - 5298 - 1 - 65 - 6 - 11 after 10000ms` -> this error means that the device doesn't respond to z2m, it could have a lot of causes, usually it's a bad network, bad device or interference.

chipsa1982, 2024-07-29:
Thanks! 

The weird thing is that all older devices work with no problems.

Tried multiple vendors, multiple devices, including the same ones already used in the house (bulbs, smart plugs). 

Only the new show this error.

Also, as mentioned, pairing works as expected and finishes OK.

Koenkk, 2024-07-30:
You could also try the latest [coordinator firmware](https://github.com/Koenkk/Z-Stack-firmware/discussions/505) and see if that helps.

bartj12, 2024-09-07:
[image]

As a temporary fix I’m restarting Z2M twice a day. When no automations are running.

closed 2024-12-29 by github-actions[bot]